morrownr/8812au-20210629 #6: Building without dkms leads to Exec format error

ronchaine, 2021-11-04:
Trying to build this without dkms and it leads to Exec format error,
with dmesg telling me about diagreement about version of symbol
module_layout.

```
# make -C /lib/modules/5.15.0-corax/build M=`pwd` modules
# make
# make install
```
and
```
# make -j16 KVER=$(uname -r) KSRC=/lib/modules/$(uname -r)/build
# make install
```
produce the same binary (as expected)

```
# modprobe -vvvv 8812au
modprobe: INFO: custom logging function 0x56405dd2e650 registered
insmod /lib/modules/5.15.0-corax/kernel/drivers/net/wireless/8812au.ko rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_beamform_cap=10 rtw_dfs_region_domain=0 rtw_switch_usb_mode=2
modprobe: INFO: Failed to insert module '/lib/modules/5.15.0-corax/kernel/drivers/net/wireless/8812au.ko': Exec format error
modprobe: ERROR: could not insert '8812au': Exec format error
```
dmesg output:
```
[ 1058.824975] 8812au: disagrees about version of symbol module_layout
```

For the system building, kernel version is the same
```
# uname -r
5.15.0-corax
```

```
# modinfo 8812au.ko | grep version

version:        v5.13.6-15-gc40b977e2.20210629
srcversion:     9ABE6835FA0EE1F5E12AB9E
vermagic:       5.15.0-corax SMP mod_unload modversions
parm:           rtw_chip_version:int
```

And the compiler version is the same as the one that built the kernel
```
# cat /proc/version
Linux version 5.15.0-corax (root@corax) (gcc (Gentoo 11.2.0 p1) 11.2.0, GNU ld (Gentoo 2.37_p1 p0) 2.37) #4 SMP Tue Nov 2 21:31:32 EET 2021
```
```
# gcc --version
gcc (Gentoo 11.2.0 p1) 11.2.0
```

ronchaine, 2021-11-04:
Same thing happens with dkms installed and running the install script too, now that I checked

morrownr, 2021-11-04:
I can't think of what the problem might be off of the top of my head. I am extremely busy behind the scenes right now but will be glad to investigate as I have time.

What I need is for you to investigate as far as you can and report back.

Something to keep in mind: While this is a new version of this driver, the installation script is nearly unchanged and is the same script used on all of the Realtek drivers on this site. It is used thousands of times per week. Yes, this site gets a LOT of traffic. It gets tested by users running everything from  ARM to amd64 with OS's all over the spectrum. Is it possible there is a problem in this driver? Sure and if there is, I would like to address it as I have time but you might want to look for gentoo specific issues.

Regards

morrownr, 2021-11-19:
In the normal course of testing other things, I did some installs with dkms and I am not seeing a problem. Most of my testing is done on systems from the Debian world such as Kali, Linux Mint, Ubuntu and RasPiOS but there are some Fedora, Arch and Manjaro users that hang out here and I have heard anything out of them... which is a good sign. If you are able to find the problem, hopefully we can find a solution... if it is this driver that is causing the problem.

For now I am going to close the issue. We can reopen it if need be.

Regards

closed 2021-11-19 by morrownr