morrownr/8812au-20210629 #7: ap 5G(802.11n)  is unstable

wanglouxiaozi, 2021-11-04:
### /etc/modprobe.d/8812au.conf
options 8812au rtw_drv_log_level=4 rtw_vht_enable=1 rtw_power_mgnt=0 rtw_beamform_cap=1 rtw_switch_usb_mode=2

### hostapd.conf
interface=hotspot
driver=nl80211
logger_syslog=-1
logger_syslog_level=2
logger_stdout=-1
logger_stdout_level=2
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
country_code=US
max_num_sta=255
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
ap_max_inactivity=300
eapol_key_index_workaround=0
eap_server=0
own_ip_addr=127.0.0.1

ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40]
#vht_capab=[MAX-MPDU-11454][SHORT-GI-80][SU-BEAMFORMER][HTC-VHT]
#vht_oper_chwidth=1

# 80211n setting
ieee80211n=1
wmm_enabled=1

# Encryption stytle(wpa)
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=CCMP

hw_mode=a
channel=36
ssid=xxxx
wpa_passphrase=xxxxxxxx

### dmesg
[16080.360494] RTW: OnAction_back
[16080.360518] RTW: OnAction_back, action=0
[16080.361219] RTW: issue_addba_rsp_wait_ack(hotspot) ra=d4:6d:6d:fa:9b:eb status:=0 tid=0 size:64, acked, 1/3 in 0 ms
[16085.714850] RTW: OnAction_back
[16085.714870] RTW: OnAction_back, action=2
[16085.714880] RTW: OnAction_back(): DELBA: 0(0)
[16090.302208] RTW: OnAction_back
[16090.302227] RTW: OnAction_back, action=0
[16090.302889] RTW: issue_addba_rsp_wait_ack(hotspot) ra=d4:6d:6d:fa:9b:eb status:=0 tid=0 size:64, acked, 1/3 in 0 ms
[16095.784811] RTW: OnAction_back
[16095.784841] RTW: OnAction_back, action=2
[16095.784863] RTW: OnAction_back(): DELBA: 0(0)
[16100.275394] RTW: OnAction_back
[16100.275404] RTW: OnAction_back, action=0
[16100.276274] RTW: issue_addba_rsp_wait_ack(hotspot) ra=d4:6d:6d:fa:9b:eb status:=0 tid=0 size:64, acked, 1/3 in 0 ms
[16105.458923] RTW: OnAction_back
[16105.458957] RTW: OnAction_back, action=2
[16105.458974] RTW: OnAction_back(): DELBA: 0(0)
[16110.273053] RTW: OnAction_back
[16110.273072] RTW: OnAction_back, action=0
[16110.274130] RTW: issue_addba_rsp_wait_ack(hotspot) ra=d4:6d:6d:fa:9b:eb status:=0 tid=0 size:64, acked, 1/3 in 0 ms
[16116.477928] RTW: OnAction_back
[16116.477958] RTW: OnAction_back, action=2
[16116.477976] RTW: OnAction_back(): DELBA: 0(0)

[16118.583465] RTW: +OnAuth
[16118.583480] RTW: auth alg=0, seq=1
[16118.583517] RTW: issue_auth
[16118.585552] RTW: OnAssocReq
[... 136 lines cut ...]
[16147.422665] RTW: issue_nulldata(hotspot) to d4:6d:6d:fa:9b:eb, ch:36, 3/3 in 140 ms
[16147.422683] RTW: expire_timeout_chk(hotspot) asoc check, d4:6d:6d:fa:9b:eb keep_alive_trycnt=2
[16149.502705] RTW: issue_nulldata(hotspot) to d4:6d:6d:fa:9b:eb, ch:36, 3/3 in 140 ms
[16149.502725] RTW: expire_timeout_chk(hotspot) asoc check, d4:6d:6d:fa:9b:eb keep_alive_trycnt=3
[16151.582799] RTW: issue_nulldata(hotspot) to d4:6d:6d:fa:9b:eb, ch:36, 3/3 in 140 ms
[16151.582811] RTW: expire_timeout_chk(hotspot) asoc expire d4:6d:6d:fa:9b:eb, state=0x4001**
[16151.582823] RTW: rtw_cfg80211_indicate_sta_disassoc(hotspot)
[16151.582850] RTW: rtw_ht_operation_update current operation mode=0x15
[16151.582856] RTW: rtw_ht_operation_update new operation mode=0x15 changes=0
[16151.582861] RTW: bss_cap_update_on_sta_leave, updated=0
[16151.582870] RTW: rtw_stadel_event_callback(mac_id=3)=d4:6d:6d:fa:9b:eb
[16151.582876] RTW: rtw_sta_mstatus_disc_rpt(hotspot) - mac_id=3
[16151.583139] RTW: cfg80211_rtw_del_key(hotspot) key_index=0, addr=d4:6d:6d:fa:9b:eb
[16151.583216] RTW: cfg80211_rtw_del_key(hotspot) key_index=0, addr=d4:6d:6d:fa:9b:eb
[16151.583262] RTW: cfg80211_rtw_change_station(hotspot) mac:d4:6d:6d:fa:9b:eb
[16151.583323] RTW: +cfg80211_rtw_del_station(hotspot) mac=d4:6d:6d:fa:9b:eb
[16151.583329] RTW: free sta macaddr =d4:6d:6d:fa:9b:eb
[16151.583337] RTW: -cfg80211_rtw_del_station(hotspot)
[16151.583933] RTW: start_addr=1026, data_H:00010500, data_L:80005010, macid=3, txrpt_h4b=10500
[16151.585556] RTW: hotspot wakeup macid=3, ori reg_0x4d4=0x00000000
[16151.585610] RTW: ### Clean STA_(3) info ###
[16151.585618] RTW: rtw_release_macid(hotspot) if1, mac_addr:d4:6d:6d:fa:9b:eb macid:3
[16151.585628] RTW: rtw_set_rts_bw connect_to_8812=0,enable=1
[16151.585930] RTW: report_del_sta_event(hotspot) d4:6d:6d:fa:9b:eb mac_id=3, enqueue:0, res:1
[16151.585938] RTW: clear key for addr:d4:6d:6d:fa:9b:eb, camid:2

morrownr, 2021-11-04:
Can define unstable for me? A description of what you are trying to do and what you are seeing would help.

Regards

wanglouxiaozi, 2021-11-05:
Hi morrownr,

Thanks for your reply!
After the log in dmesg appeared, the AP cut off the connection with my laptop. There are not many APs on channel36 on my side, so the cause of interference and poor signal strength can be eliminated.

morrownr, 2021-11-05:
Please call me Nick.

Hi @wanglouxiaozi 

I'm still trying to form an overall picture of the details of your setup.

A quick glace of hostapd.conf tells me you probably need to add the following line:

rsn_pairwise=CCMP

I see some other things that might need to be changed but I want to get a better idea of your setup first. Please post the results of:

$ sudo uname -a && mokutil --sb-state && lsusb && rfkill list all && dkms status && iw dev

wanglouxiaozi, 2021-11-05:
Hi Nick，

root@raspberrypi:~# sudo uname -a && lsusb && rfkill list all && iw dev
Linux raspberrypi 5.10.32-v7l+ #4 SMP Thu Aug 19 16:38:13 CST 2021 armv7l GNU/Linux
Bus 003 Device 004: ID 0bda:881a Realtek Semiconductor Corp.
Bus 003 Device 003: ID 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet
Bus 003 Device 002: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 004: ID 2c7c:0800 Quectel Wireless Solutions Co., Ltd.
Bus 002 Device 003: ID 2c7c:0800 Quectel Wireless Solutions Co., Ltd.
Bus 002 Device 002: ID 2c7c:0800 Quectel Wireless Solutions Co., Ltd.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
1: phy1: Wireless LAN
        Soft blocked: no
        Hard blocked: no
2: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
phy#1
        Interface hotspot
                ifindex 7
                wdev 0x100000001
                addr b4:c9:b9:c9:5e:63
                ssid TVUPACK_4872
                type AP
                channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz
                txpower 17.00 dBm
phy#0
        Unnamed/non-netdev interface
                wdev 0x2
                addr de:a6:32:fe:d2:3c
                type P2P-device
                txpower 31.00 dBm
        Interface wlan0
                ifindex 6
                wdev 0x1
                addr dc:a6:32:fe:d2:3c
                ssid HUAWEI-AQF5SV_HiLink_5G
                type managed
                channel 153 (5765 MHz), width: 80 MHz, center1: 5775 MHz
                txpower 31.00 dBm

morrownr, 2021-11-05:
Okay. We are making progress.

I now know that the computer you are using as an access point is a RasPi. I need you to confirm which RasPi it is? RasPi4B?

I know that the OS on the RasPi is the standard RasPiOS 32 bit. It is running kernel 5.10. Did you run ./raspi32.sh during the installation process?

Did the driver ./install-driver.sh script report that the driver was successfully installed?

I know that you have a USB WiFi adapter based on a rtl8812au chipset because ID 0bda:881a tells me this is so which means you are installing the right driver.

Are you plugging the usb wifi adapter directly into a usb port on the RasPi or into an external powered hub? I am concerned about power usage and some other issues. 8812au based adapters can use up to around 800 mA of power and the limit of the RasPi 3B, 3B+ and 4B is only 1200 mA total. This is one reason that I generally recommend USB WiFi adapters based on mt7612u chipsets instead of Realtek chips. The 7612u based adapters use only around 380 mA of power.

What else do you have plugged into usb ports on the RasPi? 

You are trying to use the USB WiFi adapter in AP mode but how is the internet access getting to the RasPi?

What setup guide did you use? I am seeing things that are making me scratch my head. How did the interface for the USB WiFi adapter come to be called "hotspot?"

Cheers

wanglouxiaozi, 2021-11-05:
> I now know that the computer you are using as an access point is a RasPi. I need you to confirm which RasPi it is? RasPi4B?

I used the CM4 core board with the carrier board we made ourselves. The 8812AU chipset is on the carrier board by SMT.

> I know that the OS on the RasPi is the standard RasPiOS 32 bit. It is running kernel 5.10. Did you run ./raspi32.sh during the installation process?
> Did the driver ./install-driver.sh script report that the driver was successfully installed?

The driver is installed and loaded normally, and the 8812AU chipset works. The problem is that it is unstable when used as a 
AP, and the station will be disconnected after a few hours, which I think is the problem of AP at present.

> Are you plugging the usb wifi adapter directly into a usb port on the RasPi or into an external powered hub? I am concerned about power usage and some other issues. 8812au based adapters can use up to around 800 mA of power and the limit of the RasPi 3B, 3B+ and 4B is only 1200 mA total. This is one reason that I generally recommend USB WiFi adapters based on mt7612u chipsets instead of Realtek chips when using a RasPi. The 7612u based adapters use only around 380 mA of power.

The power supply of 8812 is sufficient and has been filtered, so we don't think it is the power supply problem at present.

> What else do you have plugged into usb ports on the RasPi?

As you can see from the **lsusb**, the USB3.0 hub has three  modems, and the USB2.0 hub has one USB-to-ETH nic chip.

> You are trying to use the USB WiFi adapter in AP mode but how is the internet access getting to the RasPi?

 3 modems and USB2ETH

> What setup guide did you use? I am seeing things that are making me scratch my head. How did the interface for the USB WiFi adapter come to be called "hotspot?"

haha, to distinguish it from the wifi on the CM4 core board, we renamed WLAN1 (8812AU) to hotspot.

morrownr, 2021-11-05:
> I used the CM4 core board with the carrier board we made ourselves. The 8812AU chipset is on the carrier board by SMT.

Could you provide links to the exact hw you are using. I like to see what I am dealing with if possible.

> The driver is installed and loaded normally, and the 8812AU chipset works.

Good.

> The problem is that it is unstable when used as a AP, and the station will be disconnected after a few hours, which I think is the problem of AP at present.

Okay. I'm beginning to get enough information to form a theory about the problem you are seeing.

> The power supply of 8812 is sufficient and has been filtered, so we don't think it is the power supply problem at present.

I understand.
 
> As you can see from the **lsusb**, the USB3.0 hub has three modems, and the USB2.0 hub has one USB-to-ETH nic chip.

You said some magic words in that sentence: "USB3.0 hub"

It is a powered USB3 hub, right? If so, can you tell me what hub you are using and how it is plugged into the system you have built? There us a USB3 port for the hub to plug into?

> > You are trying to use the USB WiFi adapter in AP mode but how is the internet access getting to the RasPi?
> 
> 3 modems and USB2ETH

I understand.

> > What setup guide did you use? I am seeing things that are making me scratch my head. How did the interface for the USB WiFi adapter come to be called "hotspot?"
> 
> haha, to distinguish it from the wifi on the CM4 core board, we renamed WLAN1 (8812AU) to hotspot.

Oh. Cool. You can shut down the onboard wifi if you want.

Let me spell out my theory on this issue. Long story but maybe it will help.

The most recent episode of this problem came up over the last few days as some helpers and I brought a new driver online:

https://github.com/morrownr/88x2bu-20210702

After the repo was loaded but still private we set about testing. Reports of AP dropping offline came in. I even saw it on my RasPi4B that I use as an AP. To help see what is happening you can set up an iperf3 server on your CM4:

$ sudo apt install iperf3

Near the end of the document Bridged Wireless Access Point, which is here in the repo, there is a section about installing and setting iperf3 to autostart. Once the server is running on the CM4, you can use iperf3 on a client and test like crazy. I usually use the -t 60 parameter to get it to run for 60 seconds. You can adjust the time as needed. I was seeing AP drop offline after 20-30 seconds generally. We tried all available drivers parameters and I even changed some parts of the code but that did not help so we went a different direction. We saw better results when we used switched to

rtw_switch_usb_mode=2

so as to stay in USB2 mode. So USB3 mode seemed to be part of the problem... or was it a symtom? We then eliminated the powered hubs which also meant that some of us had to remove various other things so as to get power usage down to what the Pi usb subsystem can handle. Like I mentioned earlier, These Realtek based adapters in the AC1200 class use a lot of more power than the ones based on Mediatek chipsets (this problem just does not happen with mt7612u based adapters and they are available.)

https://github.com/morrownr/USB-WiFi

Back to topic: The solution was to document the issue. There is a well known issue with the RasPi4B usb subsystem where it will allow backfeeding of power from powered hubs. The hub I use is known to not be bad at that and it only seems to happen with faster Realtek based adapters but then they use more power which probably contributes to the problem. Things to try:

rtw_switch_usb_mode=2 (keep it in USB2 mode)
rtw_power_mgnt=1 (keep some power saving going)
rtw_beamform_cap=0 (trash the beamforming)

Try plugging the usb wifi adapter directly into the Pi, not the hub.
Get an adapter based on the mt7612u chipset and try it.

The weight of the evidence seems to point to some kind of problem in the engineering of the RasPi4B/400 usb susbsystem. Could it be a problem in the CM4 as well? Maybe so.

Hopefully I've given you some things to try and if you need me to run some tests, let me know.

Nick

morrownr, 2021-11-05:
I probably did not emphasize this point: The only reports of this problem has come from RasPi4B users. x86/amd64 users said they did not see the problem. This problem seems to be specific to the RasPi's with USB3 ports. That would include the RasPi4B/400 and maybe the CM4 but I still need to spin up to speed there.

This problem happens far less with 8812au based adapters that it does with 8812bu or 8814au adapters.

wanglouxiaozi, 2021-11-06:
Hi  Nick，

Thank you for your replies and suggestions. I will try and test them later.

zaphod24, 2021-11-08:
I have a very similar setup, a pi4 with an AirLink 101 USB3 device:

0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter

I recently updated from a pi3 to pi4 to take full advantage of the AC speeds. For me, my devices would all suddenly disconnect but hostapd was still running. The logs showed a bunch of associate and immediate dissociate messages. Restarting hostpapd got my devices connected again. After changing rtw_switch_usb_mode from 1 back to 2, the random disconnects have stopped. Unfortunately the speed also dropped from 300-400 mbps to around 250mbps.

I've got 2 AWUS036ACM devices on the way as soon as Rokland gets them back in stock.

wanglouxiaozi, 2021-11-09:
1. with the below module parameters, after runing 16 hours, AP cut the connection with station :
`options 8812au rtw_drv_log_level=4 rtw_vht_enable=2 rtw_power_mgnt=0 rtw_beamform_cap=10 rtw_switch_usb_mode=2`

`interface=hotspot
driver=nl80211
logger_syslog=-1
logger_syslog_level=2
logger_stdout=-1
logger_stdout_level=2
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
country_code=US
max_num_sta=255
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
ap_max_inactivity=300
eapol_key_index_workaround=0
eap_server=0
own_ip_addr=127.0.0.1

# 80211n setting
ieee80211n=1
ieee80211ac=1
wmm_enabled=1

ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40]

# Encryption stytle(wpa)
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=CCMP

hw_mode=a
channel=36
ssid= my_ap
`

2. if set  rtw_vht_enable=2 to enable 80211ac , rtw_beamform_cap must be 10, othervise the ap stability would be worse.

3. at my side, the best test result with rtw_power_mgnt is rtw_power_mgnt=0.

dmesg of describle 1:
[59816.107170] RTW: issue_nulldata(hotspot) to d4:6d:6d:fa:9b:eb, ch:36, 3/3 in 150 ms
[59816.107194] RTW: expire_timeout_chk(hotspot) asoc check, d4:6d:6d:fa:9b:eb keep_alive_trycnt=1
[59818.845133] RTW: OnAction_back
[59818.845172] RTW: OnAction_back, action=0
[59818.845974] RTW: issue_addba_rsp_wait_ack(hotspot) ra=d4:6d:6d:fa:9b:eb status:=0 tid=0 size:64, acked, 1/3 in 0 ms
[59823.963061] RTW: OnAction_back
[59823.963071] RTW: OnAction_back, action=2
[59823.963077] RTW: OnAction_back(): DELBA: 0(0)
[59829.093039] RTW: +OnAuth
[59829.093058] RTW: auth alg=0, seq=1
[59829.093078] RTW: issue_auth
[59829.095743] RTW: OnAssocReq
[59829.095769] RTW: OnAssocReq(hotspot) old AID=2
[59829.095780] RTW: HT: STA d4:6d:6d:fa:9b:eb HT Capabilities Info: 0x09e7
[59829.095789] RTW: bss_cap_update_on_sta_join STA d4:6d:6d:fa:9b:eb - no greenfield, num of non-gf stations 2
[59829.095797] RTW: rtw_ht_operation_update current operation mode=0x15
[59829.095804] RTW: rtw_ht_operation_update new operation mode=0x15 changes=0
[59829.095811] RTW: bss_cap_update_on_sta_join, updated=0
[59829.095819] RTW: update_sta_info_apmode
[59829.095844] RTW: issue_del_ba(hotspot) ra=d4:6d:6d:fa:9b:eb reason=37, tid=0, initiator=1
[59829.095852] RTW: Current STA(2) VHT STBC = 0A
[59829.095859] RTW: Current STA(2) VHT Beamforming Setting = 1001
[59829.095868] RTW: ### Set STA_(3) info ###
[59829.095881] RTW: issue_asocrsp
[59829.095906] RTW: report_add_sta_event: add STA
[59829.097205] RTW: rtw_get_sta_tx_nss: 2 SS
[59829.097212] RTW: rtw_get_sta_rx_nss: 2 ss
[... 213 lines cut ...]
[59877.922539] RTW: rtw_ht_operation_update new operation mode=0x15 changes=0
[59877.922550] RTW: bss_cap_update_on_sta_leave, updated=0
[59877.922573] RTW: report_del_sta_event(hotspot) d4:6d:6d:fa:9b:eb mac_id=3, enqueue:1, res:1
[59877.922591] RTW: -cfg80211_rtw_del_station(hotspot)
[59877.922776] RTW: cfg80211_rtw_del_key(hotspot) key_index=0, addr=d4:6d:6d:fa:9b:eb
[59877.922886] RTW: cfg80211_rtw_del_key(hotspot) key_index=0, addr=d4:6d:6d:fa:9b:eb
[59877.922951] RTW: cfg80211_rtw_change_station(hotspot) mac:d4:6d:6d:fa:9b:eb
[59877.923044] RTW: +cfg80211_rtw_del_station(hotspot) mac=d4:6d:6d:fa:9b:eb
[59877.923057] RTW: free sta macaddr =d4:6d:6d:fa:9b:eb
[59877.923071] RTW: -cfg80211_rtw_del_station(hotspot)
[59877.923223] RTW: cfg80211_rtw_del_key(hotspot) key_index=0, addr=d4:6d:6d:fa:9b:eb
[59877.923469] RTW: cfg80211_rtw_del_key(hotspot) key_index=0, addr=d4:6d:6d:fa:9b:eb
[59877.923596] RTW: cfg80211_rtw_del_key(hotspot) key_index=0, addr=d4:6d:6d:fa:9b:eb
[59877.923670] RTW: cfg80211_rtw_change_station(hotspot) mac:d4:6d:6d:fa:9b:eb
[59877.923748] RTW: +cfg80211_rtw_del_station(hotspot) mac=d4:6d:6d:fa:9b:eb
[59877.923765] RTW: free sta macaddr =d4:6d:6d:fa:9b:eb
[59877.923780] RTW: -cfg80211_rtw_del_station(hotspot)
[59877.929757] RTW: rtw_stadel_event_callback(mac_id=3)=d4:6d:6d:fa:9b:eb
[59877.929773] RTW: rtw_sta_mstatus_disc_rpt(hotspot) - mac_id=3
[59877.933414] RTW: start_addr=1026, data_H:00010600, data_L:8000503b, macid=3, txrpt_h4b=10600
[59877.935001] RTW: hotspot wakeup macid=3, ori reg_0x4d4=0x00000000
[59877.935117] RTW: ### Clean STA_(3) info ###
[59877.935139] RTW: rtw_release_macid(hotspot) if1, mac_addr:d4:6d:6d:fa:9b:eb macid:3
[59877.935165] RTW: rtw_set_rts_bw connect_to_8812=0,enable=1
[59877.935505] RTW: clear key for addr:d4:6d:6d:fa:9b:eb, camid:2

morrownr, 2021-11-09:
> I recently updated from a pi3 to pi4 to take full advantage of the AC speeds. For me, my devices would all suddenly disconnect but hostapd was still running.

Sounds familiar. You can a good look at it with iperf3. I set iperf3 to run for at least 60 seconds because it usually takes 15 to 30 seconds to drop offline. This happens most frequently with the 88x2bu driver.

> The logs showed a bunch of associate and immediate dissociate messages.

I've worked on this and my best working theory at this point is that we may have an EMI issue. This does not happen on x86/amd64 hardware. In fact, the only place I have seen this or had others report it with RasPi4B's with USB3 mode on.

> Restarting hostpapd got my devices connected again. After changing rtw_switch_usb_mode from 1 back to 2, the random disconnects have stopped.

That is consistent with what I have seen. Take the driver back down in to USB2 mode and it all goes away.

 > Unfortunately the speed also dropped from 300-400 mbps to around 250mbps.

Well, yes. It hurts to shift your race car down into low gear.

> I've got 2 AWUS036ACM devices on the way as soon as Rokland gets them back in stock.

My RasPi4B is primarily used as a bridged wireless access point. The 5GHz radio is handled by an Alfa ACM. I call it my 24/7/365 setup. Iperf3 shows that the ACM can consistently pump out 320-400 Mb/s in AP mode. One big advantage it has over 8812au based adapters when used with Pi's is that it only uses 360 mA of power at full speed where as some 8812au based adapters can take up to around 800 mA of power at full speed. The usb subsystem in the Pi 3b, 3b+ and 4b only supplies up to 1200 mA making the mt7612u adapters the go to adapter on Pi's. I'm not saying the contest between the 8812au and 7612u is totally one sided. The Realtek based adapter is probably around 10 faster overall, which is an estimate on my part. I've tested many adapters but a lot of things can vary according to a specific design so that is just what I consider a good estimate. But then, if the 7612u adapter can run stable in usb3 mode and it takes usb2 for the 8812au to run stable, we know who will win that race. For what it is worth, I have never seen this problem with the Alfa ACM.

morrownr, 2021-11-09:
> 1. with the below module parameters, after runing 16 hours, AP cut the connection with station :
>    `options 8812au rtw_drv_log_level=4 rtw_vht_enable=2 rtw_power_mgnt=0 rtw_beamform_cap=10 rtw_switch_usb_mode=2`
> 
> ```
> 
> interface=hotspot
> driver=nl80211
> logger_syslog=-1
> logger_syslog_level=2
> logger_stdout=-1
> logger_stdout_level=2
> ctrl_interface=/var/run/hostapd
> ctrl_interface_group=0
> country_code=US
> max_num_sta=255
> macaddr_acl=0
> auth_algs=1
> ignore_broadcast_ssid=0
> ap_max_inactivity=300
> eapol_key_index_workaround=0
> eap_server=0
> own_ip_addr=127.0.0.1
> 
> ieee80211n=1
> ieee80211ac=1
> wmm_enabled=1
> ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40]
> wpa=2
> wpa_key_mgmt=WPA-PSK
> wpa_pairwise=CCMP
> hw_mode=a
> channel=36
> ssid= my_ap
> ```

Short answer. Getting late here.

It is not clear to me why you are turning vht on in the driver and in hostapd but yet you do not provide hostapd any capabilities for it to use. Here is what I use for my 8812au adapter:

```
# hw ht capab: 0x862
ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
#
# hw vht capab: 0x03c001a2
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
```

There are some settings in hostapd.conf I would like to discuss and your exact hw setup is still some what of a mystery to me but I'm done for today. Have a good one.

closed 2021-11-11 by wanglouxiaozi