morrownr/8812au-20210629 #10: network device isn't created

marillat, 2021-11-11:
I was using the previous driver from https://github.com/morrownr/8812au-20210629 since months without problem, but this one doesn't work.

Linux amd64 debian unstable, kernel 5.14.17

Debug entries (debug set to 6)
```
$ grep RTW /var/log/syslog
Nov 11 10:02:28 www kernel: [ 8408.448888] RTW: module init start
Nov 11 10:02:28 www kernel: [ 8408.448892] RTW: rtl8812au v5.13.6-15-gc40b977e2.20210629
Nov 11 10:02:28 www kernel: [ 8408.448902] RTW: rtw_inetaddr_notifier_register
Nov 11 10:02:28 www kernel: [ 8408.448933] RTW: module init ret=0
```
modprobe output
```
$ sudo modprobe -v -v -v  8812au
modprobe: INFO: ../libkmod/libkmod.c:367 kmod_set_log_fn() custom logging function 0x55a06f801700 registered
modprobe: DEBUG: ../libkmod/libkmod-index.c:757 index_mm_open() file=/lib/modules/5.14.17/modules.dep.bin
modprobe: DEBUG: ../libkmod/libkmod-index.c:757 index_mm_open() file=/lib/modules/5.14.17/modules.alias.bin
modprobe: DEBUG: ../libkmod/libkmod-index.c:757 index_mm_open() file=/lib/modules/5.14.17/modules.symbols.bin
modprobe: DEBUG: ../libkmod/libkmod-index.c:757 index_mm_open() file=/lib/modules/5.14.17/modules.builtin.alias.bin
modprobe: DEBUG: ../libkmod/libkmod-index.c:757 index_mm_open() file=/lib/modules/5.14.17/modules.builtin.bin
modprobe: DEBUG: ../libkmod/libkmod-module.c:556 kmod_module_new_from_lookup() input alias=8812au, normalized=8812au
modprobe: DEBUG: ../libkmod/libkmod-module.c:562 kmod_module_new_from_lookup() lookup modules.dep 8812au
modprobe: DEBUG: ../libkmod/libkmod.c:597 kmod_search_moddep() use mmaped index 'modules.dep' modname=8812au
modprobe: DEBUG: ../libkmod/libkmod.c:405 kmod_pool_get_module() get module name='8812au' found=(nil)
modprobe: DEBUG: ../libkmod/libkmod.c:413 kmod_pool_add_module() add 0x55a0705dcd90 key='8812au'
modprobe: DEBUG: ../libkmod/libkmod-module.c:202 kmod_module_parse_depline() 0 dependencies for 8812au
modprobe: DEBUG: ../libkmod/libkmod-module.c:589 kmod_module_new_from_lookup() lookup 8812au=0, list=0x55a0705dcea0
modprobe: DEBUG: ../libkmod/libkmod.c:502 lookup_builtin_file() use mmaped index 'modules.builtin' modname=8812au
modprobe: DEBUG: ../libkmod/libkmod-module.c:1760 kmod_module_get_initstate() could not open '/sys/module/8812au/initstate': No such file or directory
modprobe: DEBUG: ../libkmod/libkmod-module.c:1770 kmod_module_get_initstate() could not open '/sys/module/8812au': No such file or directory
modprobe: DEBUG: ../libkmod/libkmod-module.c:1404 kmod_module_get_options() modname=8812au mod->name=8812au mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1409 kmod_module_get_options() passed = modname=8812au mod->name=8812au mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1404 kmod_module_get_options() modname=88XXau mod->name=8812au mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1404 kmod_module_get_options() modname=88XXau mod->name=8812au mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1404 kmod_module_get_options() modname=snd_pcsp mod->name=8812au mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1404 kmod_module_get_options() modname=snd_usb_audio mod->name=8812au mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1404 kmod_module_get_options() modname=cx88_alsa mod->name=8812au mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1404 kmod_module_get_options() modname=snd_atiixp_modem mod->name=8812au mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1404 kmod_module_get_options() modname=snd_intel8x0m mod->name=8812au mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1404 kmod_module_get_options() modname=snd_via82xx_modem mod->name=8812au mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1404 kmod_module_get_options() modname=bonding mod->name=8812au mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1404 kmod_module_get_options() modname=dummy mod->name=8812au mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1760 kmod_module_get_initstate() could not open '/sys/module/8812au/initstate': No such file or directory
modprobe: DEBUG: ../libkmod/libkmod-module.c:1770 kmod_module_get_initstate() could not open '/sys/module/8812au': No such file or directory
modprobe: DEBUG: ../libkmod/libkmod-module.c:750 kmod_module_get_path() name='8812au' path='/lib/modules/5.14.17/updates/dkms/8812au.ko'
modprobe: DEBUG: ../libkmod/libkmod-module.c:750 kmod_module_get_path() name='8812au' path='/lib/modules/5.14.17/updates/dkms/8812au.ko'
insmod /lib/modules/5.14.17/updates/dkms/8812au.ko rtw_drv_log_level=6 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1  
modprobe: DEBUG: ../libkmod/libkmod-module.c:750 kmod_module_get_path() name='8812au' path='/lib/modules/5.14.17/updates/dkms/8812au.ko'
modprobe: DEBUG: ../libkmod/libkmod-module.c:468 kmod_module_unref() kmod_module 0x55a0705dcd90 released
modprobe: DEBUG: ../libkmod/libkmod.c:421 kmod_pool_del_module() del 0x55a0705dcd90 key='8812au'
modprobe: INFO: ../libkmod/libkmod.c:334 kmod_unref() context 0x55a0705dc480 released
```

morrownr, 2021-11-11:
Good morning @marillat 

> Bus 005 Device 002: ID 2001:3318 D-Link Corp. 11ac adapter

That device ID tells me your adapter has a 8811au chipset. This driver is for adapters with 8812au chipsets. You need to run the ```remove-driver.sh``` script for this driver.

The driver you need is located...

https://github.com/morrownr/8821au-20210708

Cheers

marillat, 2021-11-12:
Thanks and sorry for the noise; 8821au driver is working well.

closed 2021-11-12 by marillat