morrownr/8812au-20210629 #9: Status and Performance Report

morrownr, 2021-11-11:
Greetings

It has been a couple of weeks since this new driver was opened to the public. The numbers of visitors has steadily increased with 98 hits in one day earlier this week. Testing continues...

-----

Managed mode (aka: client) thoughts

This driver is faster in managed mode than the previous version. Here is the results of an extended session with iperf3:

Bitrate :    455 Mbits/sec           Retr : 0

I never saw results like this with the old version. Generally I was lucky to get 400 Mb/s in nearly perfect conditions. I have tested everything I can think of and all I can say at this point is there are no known bugs in managed mode. This driver is fast and very stable.

-----

Master mode (aka: AP) thoughts

DFS channel support is working well. I have been testing with hostapd and a RasPi4B. I have tested many things. This driver has been extremely stable for me. There is a document called Bridged_Wireless_Access_Point.md in the repo and if you are interested in AP mode, it should provide a lot of information. It is possible, with minor modifications to also support Wireless Router and Repeater setups. 

-----

Monitor mode thoughts

I have added a document called Monitor_Mode.md that provides help with monitor mode and I have provided a script to start and test monitor mode. The script is called ```start-mon.sh```. This script helps with entering monitor mode and it allows you to change settings in an interactive manner. It even allows you to scan and test injection functionality among other things. This script is needed because I had numerous questions about how to do various things with the last version of this driver. There are also some workarounds for issues in Aircrack-ng. The real issue is that Realtek does not follow the "book" in how everything works in monitor mode like the drivers for Mediatek/Ralink and Atheros based adapters do. Anyone that wants to help improve the script should contact me. Help is welcome. Overall, scanning, capture, injection, deauth and everything I thought to test is working well and is stable.

-----

Documentation and installation

With help from others, the documentation and installation has been improved. There is always room for improvement so if anyone wants to help in this area, let me know. There is one thing I need to make clear and that is that no everything can be supported. This driver takes a LOT of work and many OS's are tested and supported, not all of the MANY OS's and various configurations can be supported.

One big NOTE for those that used to use old versions of this driver that had support for 8811au chipset... this driver ONLY supports the 8812au chipset. Not to worry as there is also a new driver for the 8811au (8821au) chipset... but this is not it.

Thanks for using the driver.

Regards,

Nick

usama7628674, 2021-11-19:
@morrownr 
[image]

By saying interface combination, do you mean Virtual interface (VIF) which evil-twin scripts need to split the interface in more than two interfaces that can be used to perfom deauth and AP on one adaprer?

usama7628674, 2021-11-19:
To my knowledge Realtek didn't write driver for `mac80211` but `cfg80211`. That is why `VIF` is unsupported in pretty much all realtek wifi usb adapters.

morrownr, 2021-11-19:
> By saying interface combination, do you mean Virtual interface (VIF) which evil-twin scripts need to split the interface in more than two interfaces that can be used to perfom deauth and AP on one adapter?

I'm not ready to say exactly what I mean at this point because I have only had time to do limited testing. I can say that it will support one interface in AP mode and one in managed mode which is what you need for a repeater. If you are interested in testing and helping document this capability, I would welcome it.

To turn the capability on, do a `$ git pull` to refresh your driver, and then edit the Makefile.

Search for the following line if using I386_PC: (if using something else like ARM_RPI or ARM64_RPI, tell me)

#I386_PC - EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE

I have ensured that is a unique line as it is part of my strategy to support this mode. Change the line to read:

EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE

Then:

$ sudo ./remove-driver.sh
$ sudo ./install-driver.sh
reboot
$ iw dev

...and tell me what you see.

morrownr, 2021-11-19:
> To my knowledge Realtek didn't write driver for `mac80211` but `cfg80211`.

Which is why I recommend adapters using mt7612u and mt7610u chipsets to most people. Depending on how you get to this repo, you may not be aware of this repo that I also maintain:

https://github.com/morrownr/USB-WiFi

> That is why `VIF` is unsupported in pretty much all realtek wifi usb adapters.

As you can probably tell by reading the FAQ and by reading the USB-WiFi repo, I am not a fan of Realtek. I am maintaining these Realtek drivers for Linux users that already have the Realtek based adapters. I would like to see Realtek head in the direction of mac80211 and in-kernel maintenance but I have to admit that this driver is not bad.

usama7628674, 2021-11-19:
I'll test this tomorrow. I need your help to sort `txpower` issue of `mt7610u` chipset. I just received `alfa awus036achm`. It is indeed a great adapter, but txpower is too low. `sudo iw dev` sometimes shows `2 dbm` or `3 dbm` .Since, you have exactly this adapter, can you post your `iw dev` log?

morrownr, 2021-11-19:
> I'll test this tomorrow. I need your help to sort `txpower` issue of `mt7610u` chipset. I just received `alfa awus036achm`. It is indeed a great adapter, but txpower is too low. `sudo iw dev` sometimes shows `2 dbm` or `3 dbm` .Since, you have exactly this adapter, can you post your `iw dev` log?

This is a known issue. The problem is not with the actual performance of the adapter, it is with the reporting. I think I saw a fix go in over the last week over at linux-wireless so it will eventually work its way to us. In the mean time, disregard what iw is telling you about txpower.

Here is the mt76 repo over at OpenWRT:

https://github.com/openwrt/mt76

It is a downstream repo of the mt76 in the Linux kernel but things that are merged and tested here generally work their way up into the Linux kernel.

The ACHM is an adapter that really grew on me over time. It is rock solid as far as AP, monitor and managed modes are concerned but I did not fully appreciate its range until I had some time to do some head to head testing. The Alfa ACH has a reputation for having long range and when I did some head to head testing that included both, I was surprised as the ACHM beat the ACH. For the last 2 months, my ACHM has been pulling AP duty. It is plugged into the USB2 port on my ZyXEL NBG6817 wifi router. The ZyXEL runs OpenWRT. OpenWRT has driver packages in their repo for the mt7612u and mt7610u based adapters so I thought I would see how it worked. It was easy and I can't get the adapter back. It is a 24/7 workhorse performing a mission critical job. It is on channel 149 and shows better range than the built-in 5 GHz radio. I actually needed two 5 GHz radios on the router for different use cases.

usama7628674, 2021-11-20:
[image]

I was amazed when I saw this that ACHM was beating ACH and using low power. `iw dev` does indeed dump wrong txpower. It's a cosmetic bug. To check this, I spawned an AP and went 15 meters apart from adapter and I was able to connect to AP with excellent signal strength even though there were couple of thick walls in-between adapter and my smartphone. If adapter was transmitting at `2dBm`, I wouldn't be able to connect to AP.

usama7628674, 2021-11-20:
Another area where `ACHM` shines is injection rate. I was sitting 1.5 meters apart from my router and `8812au` adapter was injecting at ~43%, but with `ACHM` the injection rate went up to 100%. Last adapter, that outputs 100% injection rate was Atheros AR9271 which has shows strong injection rate. Realtek adapters lag in pretty much all functionalities compared to adapters with mediatek, railink and atheros chipset.

usama7628674, 2021-11-20:
I'll test concurrent mode and report back to you.

usama7628674, 2021-11-20:
> I'm not ready to say exactly what I mean at this point because I have only had time to do limited testing. I can say that it will support one interface in AP mode and one in managed mode which is [...]

[image]

I did see two interfaces, but wasn't successful to conduct evil-twin attack. As [fluxion](https://github.com/usama7628674/fluxion-1) launched evil-twin attack I got `Stack Trace`.

[image]

usama7628674, 2021-11-20:
VIF is not coming unless Realtek writes driver for `mac80211`.

morrownr, 2021-11-20:
> Another area where `ACHM` shines is injection rate. I was sitting 1.5 meters apart from my router and `8812au` adapter was injecting at ~43%, but with `ACHM` the injection rate went up to 100%.

I have seen this also. I have folks that are learning wifi security analysis skills ask me sometimes what adapter I would recommend and it is an easy call. I recommend the Alfa ACHM.

> Last adapter, that outputs 100% injection rate was Atheros AR9271 which has shown strong injection rate. Realtek adapters lag in pretty much all functionalities compared to adapters with mediatek, railink and atheros chipset.

A lot of that has to do with their decision not to switch to mac80211 and support their driver in-kernel.

morrownr, 2021-11-20:
> VIF is not coming unless Realtek writes driver for `mac80211`.

I appreciate the report. Now to figure out if the concurrent mode is good for anything.