morrownr/8812au-20210629 #11: guide for wpa3 would be great

ABerTSC, 2021-11-12:
hi for the linux beginners a guide for the wpa3 would be super!

morrownr, 2021-11-12:
Hi @ABerTSC 

Are you willing to test it?

morrownr, 2021-11-12:
I have completed a guide. I will post it after I have tested it a few more times.

It will come with the following disclaimer:

Disclaimer: This guide attempts to help you install software that has not been
released. This may break your system. If it does break your system, you own it.

ABerTSC, 2021-11-18:
I would test it if you like yes.

morrownr, 2021-11-19:
/home/morrow/AppDev/wpa-supplicant/update_wpa_supplicant

morrownr, 2021-11-19:
[update_wpa_supplicant.txt](https://github.com/morrownr/8812au-20210629/files/7567872/update_wpa_supplicant.txt)

Please report as you are able.

usama7628674, 2021-11-20:
I can test this.

usama7628674, 2021-11-20:
old wpa_supplicant
[image]

New wpa_supplicant
[image]

usama7628674, 2021-11-20:
Now to spawn wpa3-SAE with hostapd and connect with latest version of wpa_supplicant

usama7628674, 2021-11-20:
I have a question. I only have only `rtl8812au` based adapter. Should I spawn AP with `rtl8812au` using hostapd and connect with `AWUS036ACHM` (supports WPA3-SAE) using latest wpa_supplicant or vice versa or Should I need to `rtl8812au` adapters to test WPA3-SAE?

usama7628674, 2021-11-20:
Okay, so I have used `Atheros ar9565` for hostapd and used `rtl8812au` for wpa_supplicant to connect to AP and I can conform SAE is supported.
[image]

usama7628674, 2021-11-20:
hostpad.conf 
```
interface=wlan0
ssid=WPA3-Network

hw_mode=g
channel=1

wpa=2
wpa_passphrase=abcdefgh
wpa_key_mgmt=SAE
rsn_pairwise=CCMP
ieee80211w=2
```

wpa_supplicant.conf
```
network={
    ssid="WPA3-Network"
    psk="abcdefgh"
    key_mgmt=SAE
    ieee80211w=2
}
```

morrownr, 2021-11-20:
No, you do not need 2 8812au adapters to test if WPA3 works. How about testing it first in client mode. In this case, you simply need an access point that supports WPA3. That access point can be from a regular wifi router or from an access point that you made. I can tell you that your ALFA ACHM and a RasPi support WPA3 because I have tested it many times and no wpa_supplicant upgrade us required for adapters that use in-kernel drivers like the ACHM. If you use a RasPi with your ACHM as an AP for testing WAP3, here are the vital settings to activate WPA3 for hostapd.conf:

auth_algs=3
wpa=2
rsn_pairwise=CCMP
wpa_key_mgmt=SAE
ieee80211w=2

morrownr, 2021-11-20:
> Okay, so I have used `Atheros ar9565` for hostapd and `rtl8812au` for wpa_supplicant to connect to AP and I can confirm SAE is supported. [image]

That is a very positive result. I appreciate you for testing. I am going to edit the guide now to make it more simple. Once it is complete, I will post it here and we need to come up with a plan for wider testing before adding information to the README. Thoughts?

usama7628674, 2021-11-21:
When you post guide to enable wpa3 support, I'll retest based on a simple guide you post here.

morrownr, 2021-11-22:
> When you post guide to enable wpa3 support, I'll retest based on a simple guide you post here.

New guide is posted. See #17

closed 2021-12-06 by morrownr

ABerTSC, 2022-02-18:
hi i cant find #17 why its is gone?