morrownr/8812au-20210629 #12: dkms report Deprecated feature: REMAKE_INITRD

heku, 2021-11-17:
Hi,
I'm a Archlinux user, after a recent upgrade, I see a warning message from dkms.
```bash
$ dkms status
Deprecated feature: REMAKE_INITRD
rtl8812au/5.13.6, 5.15.2-arch1-1, x86_64: installedDeprecated feature: REMAKE_INITRD
```
This is not a problem so far, just a FYI in case there's any thing needs to catch up.

Versions:
```
$ uname -r
5.15.2-arch1-1
$ dkms --version
dkms-3.0.1
```

morrownr, 2021-11-18:
Hi @heku 

Thanks for the heads up. I don't have a clue what this is about and I am neck deep working on other things right now so if you find additional information or it becomes a problem, let me know.

Regards

heku, 2021-11-19:
sure.

morrownr, 2021-11-19:
I'm going to close this issue for now. We can reopen it if need be,

Regards

closed 2021-11-19 by morrownr

gardotd426, 2021-11-21:
This is an issue for all DKMS modules, not just this one. So @morrownr I don't think you need to worry about it.

morrownr, 2021-11-22:
@gardotd426 

Thanks for the hint. It is removed.

Regards

Locutus64, 2021-12-16:
I updated Garuda KDE today and received the messages in the bottom code box. I just copied over the entire post I made on Garud's forum here.

I don't know if the errors in the below update output are something that can be ignored or are an actual problem. I'd like to know one way or the other before doing a restart. Thanks
~~~
╰─λ inxi -Faz
System:
Kernel: 5.15.7-zen1-1-zen x86_64 bits: 64 compiler: gcc v: 11.1.0
parameters: BOOT_IMAGE=/@/boot/vmlinuz-linux-zen
root=UUID=eca0fa90-c671-48bf-8727-e2f8928f8012 rw rootflags=subvol=@
quiet splash rd.udev.log_priority=3 vt.global_cursor_default=0
systemd.unified_cgroup_hierarchy=1 loglevel=3
Desktop: KDE Plasma 5.23.4 tk: Qt 5.15.2 wm: kwin_x11 vt: 1 dm: SDDM
Distro: Garuda Linux base: Arch Linux
Machine:
Type: Desktop System: ASUS product: N/A v: N/A serial: <superuser required>
Mobo: ASUSTeK model: ROG STRIX B550-F GAMING v: Rev X.0x
serial: <superuser required> UEFI: American Megatrends v: 2423
date: 08/10/2021
Battery:
Device-1: hidpp_battery_0 model: Logitech Wireless Mouse M325
Use of uninitialized value $val2 in string eq at /usr/bin/inxi line 6943.
Use of uninitialized value $val2 in split at /usr/bin/inxi line 6948.
Use of uninitialized value $val2 in concatenation (.) or string at /usr/bin/inxi line 6950.
Use of uninitialized value $val2 in concatenation (.) or string at /usr/bin/inxi line 6953.
Use of uninitialized value $val2 in concatenation (.) or string at /usr/bin/inxi line 6954.
serial: <filter> charge: 55% (should be ignored) rechargeable: yes status:
Device-2: hidpp_battery_1 model: Logitech Wireless Keyboard K360
Use of uninitialized value $val2 in string eq at /usr/bin/inxi line 6943.
Use of uninitialized value $val2 in split at /usr/bin/inxi line 6948.
Use of uninitialized value $val2 in concatenation (.) or string at /usr/bin/inxi line 6950.
Use of uninitialized value $val2 in concatenation (.) or string at /usr/bin/inxi line 6953.
Use of uninitialized value $val2 in concatenation (.) or string at /usr/bin/inxi line 6954.
serial: <filter> charge: 100% (should be ignored) rechargeable: yes status:
CPU:
Info: model: AMD Ryzen 5 3600 bits: 64 type: MT MCP arch: Zen 2
family: 0x17 (23) model-id: 0x71 (113) stepping: 0 microcode: 0x8701021
Topology: cpus: 1x cores: 6 tpc: 2 threads: 12 smt: enabled cache:
L1: 384 KiB desc: d-6x32 KiB; i-6x32 KiB L2: 3 MiB desc: 6x512 KiB
L3: 32 MiB desc: 2x16 MiB
Speed (MHz): avg: 2439 high: 3595 min/max: 2200/4208 boost: enabled
scaling: driver: acpi-cpufreq governor: schedutil cores: 1: 2057 2: 2795
3: 2150 4: 2193 5: 2144 6: 2016 7: 3116 8: 2112 9: 2043 10: 3595 11: 2094
12: 2956 bogomips: 86243
Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
Vulnerabilities:
Type: itlb_multihit status: Not affected
Type: l1tf status: Not affected
Type: mds status: Not affected
Type: meltdown status: Not affected
Type: spec_store_bypass
mitigation: Speculative Store Bypass disabled via prctl
Type: spectre_v1
mitigation: usercopy/swapgs barriers and __user pointer sanitization
Type: spectre_v2 mitigation: Full AMD retpoline, IBPB: conditional,
STIBP: conditional, RSB filling
Type: srbds status: Not affected
Type: tsx_async_abort status: Not affected
Graphics:
Device-1: NVIDIA TU116 [GeForce GTX 1650 SUPER] vendor: eVga.com.
driver: nvidia v: 495.46 alternate: nouveau,nvidia_drm bus-ID: 08:00.0
chip-ID: 10de:2187 class-ID: 0300
Display: x11 server: X.Org 1.21.1.1 compositor: kwin_x11 driver:
loaded: nvidia unloaded: modesetting alternate: fbdev,nouveau,nv,vesa
display-ID: :0 screens: 1
Screen-1: 0 s-res: 2560x1080 s-dpi: 97 s-size: 670x283mm (26.4x11.1")
s-diag: 727mm (28.6")
Monitor-1: HDMI-0 res: 2560x1080 hz: 60 dpi: 97
size: 673x284mm (26.5x11.2") diag: 730mm (28.8")
Message: Unable to show advanced data. Required tool glxinfo missing.
Audio:
[... 80 lines cut ...]
Error! Bad return status for module build on kernel: 5.15.8-zen1-1-zen (x86_64)
Consult /var/lib/dkms/rtl88x2ce/5.7.3_35403/build/make.log for more information.
==> WARNING: `dkms install --no-depmod rtl88x2ce/5.7.3_35403 -k 5.15.8-zen1-1-zen' exited 10
==> dkms install --no-depmod rtl88xxau/r1174.3a6402e -k 5.15.8-zen1-1-zen
Deprecated feature: REMAKE_INITRD
==> dkms install --no-depmod nvidia/495.46 -k 5.15.8-zen1-1-zen
==> dkms install --no-depmod 8188eu/5.3.9.r145.6146193 -k 5.15.8-zen1-1-zen
Deprecated feature: REMAKE_INITRD
Error! Bad return status for module build on kernel: 5.15.8-zen1-1-zen (x86_64)
Consult /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/make.log for more information.
==> WARNING: `dkms install --no-depmod 8188eu/5.3.9.r145.6146193 -k 5.15.8-zen1-1-zen' exited 10
==> dkms install --no-depmod rtl88x2bu/5.13.1.r149.7f96b19 -k 5.15.8-zen1-1-zen
==> dkms install --no-depmod 8192eu/128.b7faffd -k 5.15.8-zen1-1-zen
Deprecated feature: REMAKE_INITRD
Error! Bad return status for module build on kernel: 5.15.8-zen1-1-zen (x86_64)
Consult /var/lib/dkms/8192eu/128.b7faffd/build/make.log for more information.
==> WARNING: `dkms install --no-depmod 8192eu/128.b7faffd -k 5.15.8-zen1-1-zen' exited 10
==> dkms install --no-depmod rtl8192cu-fixes-git/r117.0f12fc0 -k 5.15.8-zen1-1-zen
Deprecated feature: REMAKE_INITRD
==> dkms install --no-depmod v4l2loopback-dc/1.8.0 -k 5.15.8-zen1-1-zen
==> depmod 5.15.8-zen1-1-zen
( 8/18) Updating linux initcpios...
==> Building image from preset: /etc/mkinitcpio.d/linux-zen.preset: 'default'
-> -k /boot/vmlinuz-linux-zen -c /etc/mkinitcpio.conf -g /boot/initramfs-linux-zen.img
~~~

morrownr, 2021-12-16:
Hi @Locutus64 

What you have is an almost perfect example of why I created the following repo:

https://github.com/morrownr/USB-WiFi

That shows and explains why most Linux would happier using USB WiFi adapters that use in-kernel drivers. It also gives links to many adapters that use in-kernel drivers.

What you have on your hands is commonly referred to as a mess. If you read the README here in this repo or in the other repos I maintain, I recommend removing any Realtek drivers before doing an upgrade. The reason? You just provided the reason in your post.

You are looking at lot of work to fix this problem. If it were me with this problem, I would save any important files on backup storage and do a format c:, fresh installation as it would be far quicker.

Of note:  The following is telling us what the problem is:

Deprecated feature: REMAKE_INITRD

You need to ensure that any drivers that you install to your fresh OS installation does not use the above. I have already corrected this problem in all of the drivers that I maintain:

https://github.com/morrownr?tab=repositories

Regards

Locutus64, 2021-12-16:
What does USB WiFi have to do with this? I'm not running any thing wireless and as for C drive Not on Windows as I stated. On another note saw a thread where supposedly the enduser can removed the remake_initrd cause it's not needed.

Locutus64, 2021-12-16:
> This is an issue for all DKMS modules that set `RMAKE_INITRD=` in their dkms.conf, not just this one. So @morrownr I don't think you need to worry about it. I would just remove it from dkms.conf as it's no longer needed.

@gardotd426 can you take a look at the terminal output I posted? Thanks

morrownr, 2021-12-16:
> What does USB WiFi have to do with this? I'm not running any thing wireless and as for C drive Not on Windows as I stated. On another note saw a thread where supposedly the enduser can removed the remake_initrd cause it's not need.

This site is about nothing but USB Wifi so it has everything to do with what is going on here. Why are you posting here? You just shut down any here from here!