morrownr/8812au-20210629 #15: Cannot connect to networks.

ppaslan, 2021-11-18:
**Hello first of all thank you for all your efforts messing around with realteks messy drivers :package: 
I'm having some issues with my AWUS036ACH (who doesn't) :P

Sometimes i've been able to put it into monitor mode other times i have not been able to. But the biggest issue for me right now is that i cannot connect to any networks with it? I can see networks so its obviously able to scan but i've never been able to connect with it.**

Linux kali 5.14.0-kali4-amd64 #1 SMP Debian 5.14.16-1kali1 (2021-11-05) x86_64 GNU/Linux
Bus 001 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
rtl8812au/5.13.6, 5.14.0-kali4-amd64, x86_64: installed
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 00:c0:ca:aa:59:fa
                type managed
                txpower -100.00 dBm

### Here is a snippet from rtw.log when i try to connect.

 RTW: ERROR HalPwrSeqCmdParsing: Fail to polling Offset[0x6]=00
 RTW: ERROR _InitPowerOn_8812AU: run power on flow fail
 RTW: ERROR rtw_hal_init: fail
 RTW: nolinked power save leave
 RTW: ERROR HalPwrSeqCmdParsing: Fail to polling Offset[0x6]=00
 RTW: ERROR _InitPowerOn_8812AU: run power on flow fail
 RTW: ERROR rtw_hal_init: fail
 RTW: nolinked power save leave
 RTW: ERROR HalPwrSeqCmdParsing: Fail to polling Offset[0x6]=00
 RTW: ERROR _InitPowerOn_8812AU: run power on flow fail
 RTW: ERROR rtw_hal_init: fail
 RTW: nolinked power save leave
 RTW: ERROR HalPwrSeqCmdParsing: Fail to polling Offset[0x6]=00
 RTW: ERROR _InitPowerOn_8812AU: run power on flow fail
 RTW: ERROR rtw_hal_init: fail
 RTW: nolinked power save leave
 RTW: ERROR HalPwrSeqCmdParsing: Fail to polling Offset[0x6]=00
 RTW: ERROR _InitPowerOn_8812AU: run power on flow fail
 RTW: ERROR rtw_hal_init: fail
 RTW: nolinked power save leave

Any idea if its possible to fix this?
Btw, i'm running the adapter on a virtualbox guest, so i guess that might be why im having these issues.
I mean just look at the txpower, -100 lmfao.

Thanks!

morrownr, 2021-11-19:
Hi @ppaslan 

> **Hello first of all thank you for all your efforts messing around with realteks messy drivers package

Don't thank me until things are working. <g>

> I'm having some issues with my AWUS036ACH (who doesn't) :P

I have one of those Alfa ACH adapters myself. I tested it a few days ago on the same Kali you are using. 
 
> Sometimes i've been able to put it into monitor mode other times i have not been able to.

I have a document, Monitor_Mode.md, and a script, start-mon.sh, here in the repo that should make this easy.

> But the biggest issue for me right now is that i cannot connect to any networks with it? I can see networks so its obviously able to scan but i've never been able to connect with it.**

What happens when you select a network and put your password in?

> Linux kali 5.14.0-kali4-amd64 #1 SMP Debian 5.14.16-1kali1 (2021-11-05) x86_64 GNU/Linux Bus 001 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter [...]

Can I get you to run `edit-options.sh` in the driver directory and post the contents of the options line:

$ sudo ./edit-options.sh
 
> Any idea if its possible to fix this?

I'm still looking for something to fix.

>  Btw, i'm running the adapter on a virtualbox guest, so i guess that might be why im having these issues.

Ouch. I have nothing against Virtualbox, I just don't need it and I don't test with it. Virtualbox is likely your problem. Maybe...

https://forums.virtualbox.org/viewtopic.php?f=35&t=82639

> I mean just look at the txpower, -100 lmfao.

You should use my script - start-mon.sh

$ sudo ./start-mon.sh (from the driver directory)

Note: I updated start-mon.sh today so you may want to do a `$ git pull` to update your driver directory.

Regards

morrownr, 2021-11-22:
https://null-byte.wonderhowto.com/forum/wifi-hacking-attach-usb-wireless-adapter-with-virtual-box-0324433/

morrownr, 2021-12-04:
Closed due to inactivity.

closed 2021-12-04 by morrownr