morrownr/8812au-20210629 #17: How to Enable WPA3 support

morrownr, 2021-11-21:
A guide showing how to enable WPA3 support has been requested. 

[update_wpa_supplicant_v2.txt](https://github.com/morrownr/8812au-20210629/files/7575751/update_wpa_supplicant_v2.txt)

The guide is not a general purpose guide. It is very specific and only for the drivers as noted in the guide. Please post your results and suggestions in this issue.

Regards,

Nick

usama7628674, 2021-11-22:
This guide is same as you've posted [here](https://github.com/morrownr/8812au-20210629/issues/11#issuecomment-973757563)

morrownr, 2021-11-22:
> This guide is same as you've posted [here](https://github.com/morrownr/8812au-20210629/issues/11#issuecomment-973757563)

They are similar but I made some changes that hopefully improve the use. I am now using the default defconfig for .config as it appears to work fine and I dropped using checkinstall. Hopefully there are no problems but I made enough changes that testing is needed.

Regards,

Nick

morrownr, 2021-11-29:
Here is the latest guide to upgrading wpa_supplicant:

[update_wpa_supplicant_v3-WPA3.txt](https://github.com/morrownr/8812au-20210629/files/7620298/update_wpa_supplicant_v3-WPA3.txt)

Hopefully I did not make as many mistakes this time.

I need reports of success or failure for both the installation of wpa_supplicant and if WPA3 is working.

Please include distro and kernel version in reports.

Thanks

Nick

morrownr, 2021-12-06:
Update: The following test setup is working well with WPA3. I have run this setup for just over a week now with no problems noted:

The driver from here, of course.
RasPi4B (internal wifi turned off)
Alfa - AWUS036ACH (rtl8812au chipset) (5 GHz AP mode)
RasPiOS - 2021-10-30 (32 bit)
hostapd (the version installed with $ `sudo apt install hostapd`) (no special version required)
Setup via Bridged Wireless Access Point as found in `/docs`

There was no need to upgrade wpa_supplicant or use a specific version of hostapd. The settings to turn on WPA3 are included in the guide. Basically, WPA3 this JUST WORKS in this case. If you have a RasPi and want to test WPA3, go for it.

The client adapter that connected to the RasPi4B AP is an Alfa - AWUS036ACM (mt7612u). We are excited about WPA3 starting to work on our Realtek based cards...finally, but it has been working on Mediatek based adapters for some time.

Enjoy.