morrownr/8812au-20210629 #18: PLEASE ADD MORE FREQUNCY

kaselix, 2021-11-29:
5060, 5900 ,5885,5400,5935.......................

morrownr, 2021-11-29:
Good morning @kaselix 

Can I get you to elaborate on this issue? My mind reading skills have deteriorated with age.

What interface mode are we talking about and how do I reproduce the issue?

kaselix, 2021-11-29:
Good morning @morrownr, Thanks for replying, no it's not a problem with your driver but it would be useful to have more frequencies available.Until recently I used the 5.6.4.2 aircrack-ng / rtl8812au driver where several frequencies have been added that from 5075Mhz up to 5885mhz, it would be very nice to have the full frequency range even beyond 5885 on your driver.

kaselix, 2021-11-29:
Thanks @morrownr, of course I will gladly do it
Greetings and good work.

morrownr, 2021-12-04:
Hi @kaselix 

I did not forget. I am very busy. I did some digging into the frequencies you are requesting. I am going to have to decline this request because the requested frequencies are not authorized frequencies. I  really don't need the FCC knocking on my door. 

I'll be glad to entertain other requests.

Regards

kaselix, 2021-12-04:
Hi morrown, I understand, thank you equally good job
Greetings.

closed 2021-12-04 by kaselix