morrownr/8812au-20210629 #20: Error! Bad return status for module build on kernel: 5.10.63-v7l+ (armv7l)

nikscha, 2021-11-29:
Newbie here, got the following error while installing the driver on a Raspberry Pi 4 running Raspbian.

> Running install-driver.sh version 20211002
  Starting installation...
  Copying source files to: /usr/src/rtl8812au-5.13.6
  Copying 8812au.conf to: /etc/modprobe.d
  Creating symlink /var/lib/dkms/rtl8812au/5.13.6/source ->
                   /usr/src/rtl8812au-5.13.6
  DKMS: add completed.
  Kernel preparation unnecessary for this kernel.  Skipping...
  Building module:
  cleaning build area...
  'make' -j4 KVER=5.10.63-v7l+ KSRC=/lib/modules/5.10.63-v7l+/build...(bad exit status: 2)
  Error! Bad return status for module build on kernel: 5.10.63-v7l+ (armv7l)
  Consult /var/lib/dkms/rtl8812au/5.13.6/build/make.log for more information.
  An error occurred. dkms build error = 10
  Please report this error.

-------------------------------------------------------------------
This is the content of the mentioned log file:

DKMS make.log for rtl8812au-5.13.6 for kernel 5.10.63-v7l+ (armv7l)
Mon 29 Nov 2021 02:23:50 PM GMT
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.10.63-v7l+/build M=/var/lib/dkms/rtl8812au/5.13$make[1]: Entering directory '/usr/src/linux-headers-5.10.63-v7l+'
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_io.o
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/Makefile.build:280: /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_cmd.o] E$make[2]: *** Waiting for unfinished jobs....
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/Makefile.build:280: /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_io.o] Er$gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/Makefile.build:280: /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_debug.o]$gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/Makefile.build:280: /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_security$make[1]: *** [Makefile:1825: /var/lib/dkms/rtl8812au/5.13.6/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.63-v7l+'
make: *** [Makefile:2499: modules] Error 2

morrownr, 2021-11-29:
Good morning @nikscha 

Hmmm... that result is not what we were looking for. 

Can I get you to post the results of the following command?

sudo uname -a && lsusb && rfkill list all && dkms status && iw dev

Did you run `./raspi32.sh` before the installation script?

nikscha, 2021-11-30:
Hey there, thanks for the quick reply!

This is the result:
pi@arges:~ $ sudo uname -a && lsusb && rfkill list all && dkms status && iw dev
Linux arges 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 32e4:0317
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
1: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
phy#0
        Unnamed/non-netdev interface
                wdev 0x2
                addr de:a6:32:78:3c:c2
                type P2P-device
                txpower 31.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr dc:a6:32:78:3c:c2
                ssid FRITZ!Repeater 6000
                type managed
                channel 108 (5540 MHz), width: 80 MHz, center1: 5530 MHz
                txpower 31.00 `dBm`

----------------
I ran ./raspi64.sh, as I am using a Pi 4 B

morrownr, 2021-11-30:
> Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 001 Device 003: ID 32e4:0317 Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub 0

I am not seeing where you have a USB WiFi adapter plugged in. I would have expected to see that if it is plugged in.
 
Exactly what usb wifi adapter do you have?

> I ran ./raspi64.sh, as I am using a Pi 4 B

It is not the Pi4B that determines whether you run raspi32.sh or raspi64.sh, it the OS. If you are using Raspberry Pi OS 32 bit version then you need to use raspi32.sh. Actually things will be a mess at this point so it is best that you run remove-driver.sh per the removal instructions and then delete the driver directory. After a reboot, you need to start installation over.

nikscha, 2021-12-01:
> It is not the Pi4B that determines whether you run raspi32.sh or raspi64.sh, it the OS

Well this explains a lot...
I removed the driver and reinstalled using the 32bit script and voila, everything works just fine.

Thanks a lot, I really appreciate you helping me!

buicongnguyen80, 2023-12-12:
how to fix this error: Error! Bad return status for module build on kernel: 4.15.0-213-generic (aarch64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.1/build/make.log for more information.