morrownr/8812au-20210629 #21: Include New Device

FilipeSantana, 2021-12-02:
Dear all,

I would like to request the following device to be included:

2357:0120 
TP-Link 802.11ac WLAN Adapter

[TP Link Archer T2U Plus](https://www.tp-link.com/us/home-networking/usb-adapter/archer-t2u-plus/)

Thanks in advance.

Best regards

morrownr, 2021-12-02:
Hi @FilipeSantana 

Filipe,

This driver is for adapters based on the 8812au chipset. Your T2U Plus appears to have a 8811au chipset. Please go to the following repo and let me know how it goes:

https://github.com/morrownr/8821au-20210708

FilipeSantana, 2021-12-02:
Funny fact: until recently, I was using the aircrack-ng 8812au. It worked like a charm.

However, I received a kernel update recently that broke everything (ubuntu 20.04, kernel 5.11.0-41).

After installing the 8821au from the repo you shared, it came to life.

Thank you.

closed 2021-12-02 by FilipeSantana

morrownr, 2021-12-02:
> Funny fact: until recently, I was using the aircrack-ng 8812au. It worked like a charm.
> 
> However, I received a kernel update recently that broke everything (ubuntu 20.04, kernel 5.11.0-41).

The aircrack-ng 8812au driver is based on source that is somewhat dated. Back when that source became available Realtek maintained support for the 8812au and 8811au in the driver. I guess Realtek deemed that to be a bad idea and broke things into 2 drivers and it has been that way for the last few years.
 
> After installing the 8821au from the repo you shared, it came to life.

As it should have. Enjoy.