morrownr/8812au-20210629 #22: AP stops working: rtl8812_sreset_xmit_status_check tx hang

frsarieg, 2021-12-04:
Hi,

I've watching the following message in the system logs randomly:

`"rtl8812_sreset_xmit_status_check tx hang"`

When it appears, all stations connected to the AP configured with hostapd can't communicate anymore with it. The only solution is to unload the 8812au driver, load again and restart the hostapd daemon.

I've tested all variations with this hardware:
- Raspberry Pi 2 Model B: in all USB ports and with an external powered USB hub.
- Raspberry Pi 4 Model B 8 GB: in all USB ports (2 and 3) and with an external powered USB hub.
- Alfa Network AWUS036ACH (USB C version)
- BrosTrend AC3L

Currently using the Raspberry Pi 2, the AWUS036ACH and the USB powered hub. The config and status is shown below:

```
###############################################################################
# SOFTWARE
###############################################################################

cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

#######################################

cat /etc/debian_version
10.11

#######################################

uname -a
Linux rpiap 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux

###############################################################################
# HARDWARE
###############################################################################

lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 3: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 4: Dev 5, If 0, Class=Vendor Specific Class, Driver=rtl8812au, 480M

###############################################################################
# DRIVER
###############################################################################

dkms status
8812au, 5.6.4.2_35491.20191025, 5.10.63+, armv7l: installed
8812au, 5.6.4.2_35491.20191025, 5.10.63-v7+, armv7l: installed
8812au, 5.6.4.2_35491.20191025, 5.10.63-v7l+, armv7l: installed
rtl8812au, 5.13.6, 5.10.63-v7+, armv7l: installed

#######################################

modinfo 8812au
filename:       /lib/modules/5.10.63-v7+/kernel/drivers/net/wireless/8812au.ko
version:        v5.13.6-15-gc40b977e2.20210629
author:         Realtek Semiconductor Corp.
[... 617 lines cut ...]
	Device supports SAE with AUTHENTICATE command
	Device supports scan flush.
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Supported TX frame types:
		 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0xd0
		 * managed: 0x40 0xb0 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
	Supported extended features:
```

To sort this problem I've created a systemctl service which watchs for changes in /var/log/syslog and checks the last minute in journalctl to reload the driver and restart the hostapd :-) (and it works... but...)

NIck, thank you so much in advance :-)

morrownr, 2021-12-06:
> dkms status
> 8812au, 5.6.4.2_35491.20191025, 5.10.63+, armv7l: installed
> 8812au, 5.6.4.2_35491.20191025, 5.10.63-v7+, armv7l: installed
> 8812au, 5.6.4.2_35491.20191025, 5.10.63-v7l+, armv7l: installed
> rtl8812au, 5.13.6, 5.10.63-v7+, armv7l: installed

Hi @frsarieg 

The place to start is by removing previously installed drivers. The only one that is from this repo is the last one. Having multiple drivers for the same chipset installed can lead to problems.

frsarieg, 2021-12-06:
Hi Nick,

Thanks for the response :-)

Done and still happening:

```
# dkms remove 8812au/5.6.4.2_35491.20191025 --all

-------- Uninstall Beginning --------
Module:  8812au
Version: 5.6.4.2_35491.20191025
Kernel:  5.10.63+ (armv7l)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

88XXau.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.10.63+/updates/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod....

DKMS: uninstall completed.

-------- Uninstall Beginning --------
Module:  8812au
Version: 5.6.4.2_35491.20191025
Kernel:  5.10.63-v7+ (armv7l)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

88XXau.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.10.63-v7+/updates/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod....

DKMS: uninstall completed.

-------- Uninstall Beginning --------
Module:  8812au
Version: 5.6.4.2_35491.20191025
Kernel:  5.10.63-v7l+ (armv7l)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

88XXau.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.10.63-v7l+/updates/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod....

DKMS: uninstall completed.

------------------------------
Deleting module version: 5.6.4.2_35491.20191025
completely from the DKMS tree.
------------------------------
Done.

# reboot

# dkms status
rtl8812au, 5.13.6, 5.10.63-v7+, armv7l: installed
```

```
# grep -i "rtl8812_sreset_xmit_status_check" /var/log/syslog
Dec  6 11:03:10 rpiap kernel: [11742.409775] RTW: rtl8812_sreset_xmit_status_check tx hang 
Dec  6 11:03:16 rpiap kernel: [11748.649780] RTW: rtl8812_sreset_xmit_status_check tx hang
```

morrownr, 2021-12-06:
Hi @frsarieg 

Good day to you. I see you are down to one driver installed but you are still seeing the problem.

The best way for me to be able to work this issue is if I can duplicate the problem here and it appears that I can get close to the same setup as you with available hardware.

You are using:

Raspberry Pi 4 Model B 8 GB
Tested in all USB ports (2 and 3) and with an external powered USB hub.
Alfa Network AWUS036ACH (USB C version)

I have the following:

Raspberry Pi 4 Model B 4 GB
Alfa Network AWUS036ACH (not the USB C version)

It just so happens that I am already testing the above as I am testing this driver for WPA3 capability in AP mode. It is working well but back to the subject at hand. My log is set to level 4. I ran $ `./save-log.sh` and did a search for the line in question and it is not there. I also have not noticed any drops or problems and this system has been in use 24/7 for about a week.

I realize that it will be work on your end but can I talk you into re-configuring into a near common setup?

- Raspberry Pi 4 Model B
- Alfa Network AWUS036ACH
- No external powered usb hub
- No external ssd or hd
- Boot from sd card with latest RasPiOS. Specifically:

Raspberry Pi OS with desktop
Release date: October 30th 2021
Kernel version: 5.10

- No other usb devices plugged in

We need to talk about what software you are using and how you are configuring it. I use the Bridged Wireless Access Point guide that is located in the /docs folder.

Nick
-

frsarieg, 2021-12-07:
Hi Nick,

thanks a lot for your interest in the issue. It's something that I've suffering from a long ago... And I don't be able to sort it. 

> The best way for me to be able to work this issue is if I can duplicate the problem here and it appears that I can get close to the same setup as you with available hardware.
>
> You are using: [...]

I've have an small trouble with that... All the Pis that I own are in use by several services (LibreELEC, Callattendant...). Now I am in my second house and I bring my old Pi2 to get WiFi between my devices using the AP mode (I don't get ANY WiFi network here...). So for a few days I can't do changes in the Rpi4...

What is really strange is that I'm suffering the same issue despite the changes in all variables:
- Pi2 vs Pi4
- Brostrend ACL3 vs Alfa Network AWUS036ACH
- Some different devices between locations
- USB hub vs direct connection (at least with the Alfa I can see the USB 3 connection)
 
> Raspberry Pi OS with desktop Release date: October 30th 2021 Kernel version: 5.10
> 
>     * No other usb devices plugged in
> 
> 
> We need to talk about what software you are using and how you are configuring it. I use the Bridged Wireless Access Point guide that is located in the /docs folder.
> 

I am following the guide... the only thing "strange" is the use of a country code the get the txpower to 31 dBm... that is a thing I must use in this house... I can test without this setting or decrease it a bit...

Thanks

morrownr, 2021-12-08:
31 dBm! Ouch.

The Alfa ACH uses 800 mA of power during heavy usage (measured) when at normal power settings. I do not know what it uses when at 31 dBm. The USB subsystem of the RasPi 3B, 3B+ and 4B can only supply a maximum of 1200 mA of power. You very well may be in a situation where you are overtaxing the power available. If you have anything else plugged into the USB subsystem on the Pi, you may be asking for more power than can be delivered.

Is there anyway I can talk you out of that configuration? The Alfa ACH is a nice adapter and overclocking it like this is not good for the life span of the device and it is likely not much help. The Alfa ACH and ACHM are built for maximum range. Remember that it takes 2 to Tango. The devices that are connecting to your Alfa ACH are likely the problem and increasing the output of the ACH only marginally helps because this is a 2 way thing. Have you considered alternative solutions?

frsarieg, 2021-12-08:
> 31 dBm! Ouch.

;-)

> The Alfa ACH uses 800 mA of power during heavy usage (measured) when at normal power settings. I do not know what it uses when at 31 dBm. The USB subsystem of the RasPi 3B, 3B+ and 4B can only [...]

Yep... that's why I use an external powered USB hub. :-)

> Is there anyway I can talk you out of that configuration? The Alfa ACH is a nice adapter and overclocking it like this is not good for the life span of the device and it is likely not much help. The [...]

I decreased to 30 dBm... no issues in the last hours... time will say. Probably I could decrease a bit more.

Thanks!

morrownr, 2021-12-11:
> Yep... that's why I use an external powered USB hub. :-)

Powered USB3 hubs are notorious for backfeeding into the RasPi usb subsystem. Are you familiar with the issue?

> I decreased to 30 dBm... no issues in the last hours... time will say. Probably I could decrease a bit more.

Not giving up easy are you.

:)

frsarieg, 2021-12-11:
> Powered USB3 hubs are notorious for backfeeding into the RasPi usb subsystem. Are you familiar with the issue?

Nope... I don't know about it... I can test to disconnect it again... but the issue happened with and without the hub. 

> Not giving up easy are you.

Never giving up :-) I am decreasing 1 by 1 dBm. Now set at 27 dBm. Signal is strong enough. More than 24 hours without hanging.

Thanks!

morrownr, 2021-12-22:
Status?

frsarieg, 2021-12-24:
> Status?

Hi Nick,

Still happening.

Previously I was going down the value of the txpower one by one dBm. I reach the 26 dBm and still happening. I didn't test lower values because holidays ended and I am now at my main house.

Now the setup is:
- Rpi4 with latest Raspberry Pi OS.
- No powered USB hub.
- Latest version of the driver.
- Recompiled and configured CRDA
- txpower set to 29 dBm

And still happening.

Thanks!

morrownr, 2021-12-24:
Hi @frsarieg 

I had put a note in my test schedule when you reported this and then I changed out the adapter I was using as a 5 GHz AP on my RasPi4B. It was changed to an Alfa ACH similar to yours but it is not the USB-C version. I have been watching and checking the logs for the last 3 weeks and I have nothing. I needed to change out to test another adapter yesterday which triggered me to get back with you.

So, what could be different between our systems? Mine is a 4 gb RasPi while yours is 8 gb but that should not have anything to do with it. I'm running the lastest 32 bit RasPiOS as are you. Thoughts:

- Do you have another power supply that you could try on the Pi?
- Any possibility the cable between the adapter and Pi is bad?
- Antennas are secure?

I've looked around for that line you are seeing in the log and haven't come up with anything meaningful.

How about trying this:

Run:

sudo ./remove-driver.sh

Edit the Makefile:

sudo nano Makefile

Change this line

CONFIG_POWER_SAVING = y

To

CONFIG_POWER_SAVING = n

Save and...

sudo ./install-driver.sh

When it asks to edit the options file, select y and remove the following option (and config the other settings you need)

rtw_power_mgnt=1

Here is the deal. While it might seems that rtw_power_mgnt=0 and CONFIG_POWER_SAVING = n do the same thing, they do not. The power saving code in this driver is complex. We need an EE to really sort it all out. In all, there are about 7 module parameters that can be used regarding power saving but I keep it simple and the default is what seems to work the most reliably. CONFIG_POWER_SAVING = n really will shut power savings down. IF you keep the txpower high, you need to monitor it to make sure there is no overheating.

Let's see if that makes a difference.

frsarieg, 2021-12-24:
> Hi @frsarieg
>
> I had put a note in my test schedule when you reported this and then I changed out the adapter I was using as a 5 GHz AP on my RasPi4B. It was changed to an Alfa ACH similar to yours [...]

Yes. I've tested it with two different official Raspberry Pi 4 power supplies. Plus the Rpi2 with another official power supply.
> 
>     * Any possibility the cable between the adapter and Pi is bad?

I don't think so... Tested with 2 different devices and several cables.

> 
>     * Antennas are secure?
Yes, they are.
> I've looked around for that line you are seeing in the log and haven't come up with anything meaningful.
>
> How about trying this:
>
> Run:
>
> sudo ./remove-driver.sh
>
> Edit the Makefile:
>
> sudo nano [...]

Done :-) Test running now :-) I will inform you.

Thanks a lot.

frsarieg, 2021-12-25:
Hi Nick,

Same situation:

```
Dec 25 10:06:22 raspberrypi kernel: [56198.347005] RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 25 10:06:24 raspberrypi kernel: [56200.427024] RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 25 10:06:26 raspberrypi kernel: [56202.507054] RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 25 10:06:28 raspberrypi kernel: [56204.587076] RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 25 10:06:30 raspberrypi kernel: [56206.667027] RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 25 10:06:32 raspberrypi kernel: [56208.747178] RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 25 10:06:34 raspberrypi kernel: [56210.827162] RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 25 10:06:36 raspberrypi kernel: [56212.907140] RTW: rtl8812_sreset_xmit_status_check tx hang
```

Merry Hang (not hang over) Christmas :-)

Thanks!

frsarieg, 2021-12-25:
Hi Nick,

I did more tests with the same results:

- Change antennas: I've tried the included ones, the Alfa directional and a pair of omni antennas that stated 9 dBi.
- Change the region to default Spain (ES) and the default txpower (with the included antennas):

```
# iw reg get
global
country ES: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

# iw dev
phy#0
	Interface wlan0
		ifindex 15
		wdev 0x1
		addr 00:c0:ca:af:f4:97
		ssid U1AT3OTPX-RPI4
		type AP
		channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
		txpower 15.00 dBm
```

Next I will do is set again the power to 28 dBm and test the clients one by one. Just one client connected.

Thanks!

morrownr, 2021-12-25:
Hola @frsarieg 

Feliz Navidad y Prospero Ano Nuevo (yeah, English keyboard)

This is NOT going to defeat us! I went wandering around in the source for a few minutes yesterday. It will take longer to get more of an idea of what is going on there.

I see you are on channel 36. Have you tried other channels? This driver will support DFS channels in AP mode so there is that option but there are a couple of other configuration items. I realize not all devices can work on DFS channels so maybe you can send them to 2.4 GHz while working on this. The cool thing about DFS, if your location is like mine, is that most people have no idea what a DFS channel is so most DFS channels are clear air.

I am still pondering how, with nearly the same equipment, I cannot duplicate this problem. You are using hostapd? If do, please post the contents of your hostapd.conf.

The contents of your `options` line in /etc/modprobe.d/8812au.conf might help as well.

Saludos

frsarieg, 2021-12-26:
> Hola @frsarieg
> 
> Feliz Navidad y Prospero Ano Nuevo (yeah, English keyboard)
> 

Hahaha! Thanks a lot... and Happy New Year too! Be careful with your keyboard: "año" != "ano" (yes I know it... :D)

> This is NOT going to defeat us! I went wandering around in the source for a few minutes yesterday. It will take longer to get more of an idea of what is going on there.
>
> I see you are on channel 36. [...]

Yep... I can test the DFS... I've enabled the `ieee80211h=1` in the hostapd.conf . Do we need to change more parameters? With the Rpi2 I've tested It in an space free from any other WiFi networks. 100% free air :-)

> 
> I am still pondering how, with nearly the same equipment, I cannot duplicate this problem. You are using hostapd? If do, please post the contents of your hostapd.conf.
> 

No problem:

```
accept_mac_file=/etc/hostapd/hostapd.accept
auth_algs=1
beacon_int=100
bridge=br0
channel=36
country_code=ZY
ctrl_interface_group=0
ctrl_interface=/var/run/hostapd
driver=nl80211
dtim_period=2
fragm_threshold=2346
ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
hw_mode=a
ieee80211ac=1
ieee80211d=1
ieee80211h=1
ieee80211n=1
ignore_broadcast_ssid=0
interface=wlan0
logger_stdout=-1
logger_stdout_level=2
logger_syslog=-1
logger_syslog_level=2
macaddr_acl=1
max_num_sta=32
rsn_pairwise=CCMP
rts_threshold=2347
ssid=U1AT3OTPX-RPI4
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
vht_oper_centr_freq_seg0_idx=42
vht_oper_chwidth=1
wmm_enabled=1
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_passphrase=<protected>
```

The last thing I've tried is to change the `vht_capab` from `[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][SU-BEAMFORMER][SOUNDING-DIMENSION-2][HTC-VHT][MAX-A-MPDU-EXP7]` . I've seen another recommendation in the AP guide.

> The contents of your `options` line in /etc/modprobe.d/8812au.conf might help as well.
> 

`options 8812au rtw_drv_log_level=4 rtw_led_ctrl=0 rtw_vht_enable=2 rtw_power_mgnt=0 rtw_beamform_cap=1 rtw_dfs_region_domain=0 rtw_sel_p2p_iface=0 rtw_switch_usb_mode=1`

I've tested two Android tablets (Samsung and Huawei) in "one-at-a-time" mode and still happening.

Greets!

morrownr, 2021-12-26:
> > Feliz Navidad y Prospero Ano Nuevo (yeah, English keyboard)
> 
> Hahaha! Thanks a lot... and Happy New Year too! Be careful with your keyboard: "año" != "ano" (yes I know it... :D)

Gracias. Necesito practicar el español. (I was too lazy to go copy a ñ earlier.

> Yep... I can test the DFS... I've enabled the `ieee80211h=1` in the hostapd.conf . Do we need to change more parameters? With the Rpi2 I've tested It in an space free from any other WiFi networks. 100% free air :-)

Yes, additional changes are required:

You will need to add the following to 8812au.conf

rtw_dfs_region_domain=

The docs are in 8812au.conf:

DFS Options ( rtw_dfs_region_domain )

 0 = NONE (default)
 1 = FCC
 2 = MKK
 3 = ETSI

 Notes:
 - Activates DFS channels in AP mode.
 - DFS channels automatically work in managed mode.
 - DFS FCC 80 MHz channels for hostapd: 52(58), 100(106), 116(122) and 132(138)
 - For more information: https://en.wikipedia.org/wiki/List_of_WLAN_channels

 
> ```
> accept_mac_file=/etc/hostapd/hostapd.accept
> auth_algs=1
> beacon_int=100
> bridge=br0
> channel=36
> country_code=ZY
> ctrl_interface_group=0
> ctrl_interface=/var/run/hostapd
> driver=nl80211
> dtim_period=2 [...]

This line will need to change depending on the channel you use:

vht_oper_centr_freq_seg0_idx=42

What is this? country_code=ZY

What is this? accept_mac_file=/etc/hostapd/hostapd.accept

> The last thing I've tried is to change the `vht_capab` from `[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][SU-BEAMFORMER][SOUNDING-DIMENSION-2][HTC-VHT][MAX-A-MPDU-EXP7]` . I've seen another recommendation in the AP guide.

For now, you might just want to keep it simple until we find the problem:

`vht_capab=[MAX-MPDU-11454][SHORT-GI-80][HTC-VHT]`
 
> > The contents of your `options` line in /etc/modprobe.d/8812au.conf might help as well.
> 
> `options 8812au rtw_drv_log_level=4 rtw_led_ctrl=0 rtw_vht_enable=2 rtw_power_mgnt=0 rtw_beamform_cap=1 rtw_dfs_region_domain=0 rtw_sel_p2p_iface=0 rtw_switch_usb_mode=1`

You might want to stay away from concurrent mode and leave this out while we are looking for the problem:

 rtw_sel_p2p_iface=0

Regards

frsarieg, 2021-12-26:
Hi @morrownr ,

Done :-) Report below.

> 
> Gracias. Necesito practicar el español. (I was too lazy to go copy a ñ earlier.

Thanks for the effort :D

> You will need to add the following to 8812au.conf
> 
> rtw_dfs_region_domain=
> 
> The docs are in 8812au.conf:
> 
> DFS Options ( rtw_dfs_region_domain )
> 
> 0 = NONE (default) 1 = FCC 2 = MKK 3 = ETSI
> 

`rtw_dfs_region_domain=3`

> Notes:
> 
>     * Activates DFS channels in AP mode.
> 
>     * DFS channels automatically work in managed mode.
> 
>     * DFS FCC 80 MHz channels for hostapd: 52(58), 100(106), 116(122) and 132(138)
> 
>     * For more information: https://en.wikipedia.org/wiki/List_of_WLAN_channels
> 
> This line will need to change depending on the channel you use:
> 
> vht_oper_centr_freq_seg0_idx=42

Configured at 52(58) channel:

```
# iw dev phy#0
	Interface wlan0
		ifindex 33
		wdev 0x1
		addr 00:c0:ca:af:f4:97
		ssid U1AT3OTPX-RPI4
		type AP
		channel **52** (5260 MHz), width: 80 MHz, center1: **5290 MHz**
		txpower 28.00 dBm
```

> 
> What is this? country_code=ZY
> 

It's a trick to successfully configure txpower above 20 dBm:

```
# iw reg get
global
country ZY: DFS-UNSET
	(2400 - 2483 @ 40), (N/A, 32), (N/A)
	(5150 - 5250 @ 80), (N/A, 32), (N/A)
	(5250 - 5350 @ 80), (N/A, 32), (N/A)
	(5470 - 5725 @ 160), (N/A, 32), (N/A)
	(5725 - 5875 @ 80), (N/A, 32), (N/A)
	(57000 - 66000 @ 2160), (N/A, 32), (N/A)
```
I've tried with the default-compliant ES region without success with the issue.

> What is this? accept_mac_file=/etc/hostapd/hostapd.accept

The list of accepted MAC addresses for the client stations. I've tried to disable/enable. I was suffering this problem before this config.

> For now, you might just want to keep it simple until we find the problem:
>
> `vht_capab=[MAX-MPDU-11454][SHORT-GI-80][HTC-VHT]`

I will change it tomorrow if I see the problem.

> > `options 8812au rtw_drv_log_level=4 rtw_led_ctrl=0 rtw_vht_enable=2 rtw_power_mgnt=0 rtw_beamform_cap=1 rtw_dfs_region_domain=0 rtw_sel_p2p_iface=0 rtw_switch_usb_mode=1`
> 
> You might want to stay away from concurrent mode and leave this out while we are looking for the problem:
> 
> rtw_sel_p2p_iface=0

Done :-)

Thanks a lot.

frsarieg, 2021-12-27:
Hi @morrownr,

Still happening without and with changing the `vht_capab`:
```
Dec 27 09:17:44 raspberrypi rtwhang[599]: dic 27 09:17:12 raspberrypi kernel: RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 27 09:17:44 raspberrypi rtwhang[599]: dic 27 09:17:15 raspberrypi kernel: RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 27 09:17:44 raspberrypi rtwhang[599]: dic 27 09:17:17 raspberrypi kernel: RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 27 09:17:44 raspberrypi rtwhang[599]: dic 27 09:17:19 raspberrypi kernel: RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 27 09:17:44 raspberrypi rtwhang[599]: dic 27 09:17:21 raspberrypi kernel: RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 27 09:17:44 raspberrypi rtwhang[599]: dic 27 09:17:23 raspberrypi kernel: RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 27 09:17:44 raspberrypi rtwhang[599]: dic 27 09:17:25 raspberrypi kernel: RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 27 09:17:44 raspberrypi rtwhang[599]: dic 27 09:17:31 raspberrypi kernel: RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 27 09:17:44 raspberrypi rtwhang[599]: dic 27 09:17:33 raspberrypi kernel: RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 27 09:17:44 raspberrypi rtwhang[599]: dic 27 09:17:35 raspberrypi kernel: RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 27 09:17:44 raspberrypi rtwhang[599]: dic 27 09:17:37 raspberrypi kernel: RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 27 09:17:44 raspberrypi rtwhang[599]: dic 27 09:17:40 raspberrypi kernel: RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 27 09:17:44 raspberrypi rtwhang[599]: dic 27 09:17:42 raspberrypi kernel: RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 27 09:17:44 raspberrypi rtwhang[599]: dic 27 09:17:44 raspberrypi kernel: RTW: rtl8812_sreset_xmit_status_check tx hang
Dec 27 09:47:02 raspberrypi rtwhang[599]: dic 27 09:46:44 raspberrypi kernel: RTW: rtl8812_sreset_xmit_status_check tx hang
```

Thanks!

frsarieg, 2021-12-27:
Hi @morrownr,

Preparing a new, fresh and from scratch install for my Rpi 2 Model B. I can't use any of my Rpi 4 for the moment (they are really busy :-)

Cheers!

frsarieg, 2021-12-28:
Hi @morrownr,

The new Rpi2 setup is working like a charm and flawlessly for more than 24 hours.

I've just followed the AP bridge guide, configure the DFS and connect the Alfa device. No problems.

I didn't touch any of the CRDA config... and I been able to set the txpower to the max wich is 31,5 dBm. The current setting is 28 dBm.

In the Rpi4 I've reinstalled the packages crda and wireless-regdb and set to 28 dBm without any problem. The setup is running now in the Rpi4 waiting for any hang.

Thanks and greets!

morrownr, 2022-01-03:
Status?

frsarieg, 2022-01-04:
> Status?

Hi @morrownr,

The Rpi4 fails after 1 day... I've copied all the config from the "new" Rpi2 setup... but it didn't work.

I've switched back the Alfa device to the Rpi2 and is working flawlessly. I've decided to configure all the stuff is now running in the Rpi4. I need to configure Pihole, pivpn, wireguard, cacti, callattendant, borgmatic, syncthing, ansible... a lot of stuff. I'm doing it slowly.

If the problem replicates, I will tell you.

Thanks!

frsarieg, 2022-01-07:
Hi @morrownr ,

Everything was working fine... and suddenly it happened again.

But this time I've got more logs... Sorry because it's too long, but in a short period of time

[syslog.txt](https://github.com/morrownr/8812au-20210629/files/7831492/syslog.txt)

Thanks!

frsarieg, 2022-01-13:
Hi,

still happening

```
Jan 11 16:03:45 raspberrypi kernel: [162803.519269] RTW: SOUNDING_FW_VHT_TIMER
Jan 11 16:03:46 raspberrypi kernel: [162804.968481] RTW: [C2H], C2H_TXBF!!
Jan 11 16:03:46 raspberrypi kernel: [162804.968507] RTW: SOUNDING_FW_VHT_TIMER
Jan 11 16:03:46 raspberrypi kernel: [162805.258974] RTW: rtl8812_sreset_xmit_status_check tx hang
Jan 11 16:03:47 raspberrypi kernel: [162806.415452] RTW: [C2H], C2H_TXBF!!
Jan 11 16:03:47 raspberrypi kernel: [162806.415475] RTW: SOUNDING_FW_VHT_TIMER
Jan 11 16:03:48 raspberrypi kernel: [162807.338999] RTW: rtl8812_sreset_xmit_status_check tx hang
Jan 11 16:03:49 raspberrypi kernel: [162807.874914] RTW: [C2H], C2H_TXBF!!
Jan 11 16:03:49 raspberrypi kernel: [162807.874939] RTW: SOUNDING_FW_VHT_TIMER
Jan 11 16:03:50 raspberrypi kernel: [162809.310914] RTW: [C2H], C2H_TXBF!!
Jan 11 16:03:50 raspberrypi kernel: [162809.310939] RTW: SOUNDING_FW_VHT_TIMER
Jan 11 16:03:50 raspberrypi kernel: [162809.419045] RTW: rtl8812_sreset_xmit_status_check tx hang
Jan 11 16:03:52 raspberrypi kernel: [162810.758096] RTW: [C2H], C2H_TXBF!!
Jan 11 16:03:52 raspberrypi kernel: [162810.758119] RTW: SOUNDING_FW_VHT_TIMER
Jan 11 16:03:53 raspberrypi kernel: [162812.206219] RTW: [C2H], C2H_TXBF!!
Jan 11 16:03:53 raspberrypi kernel: [162812.206242] RTW: SOUNDING_FW_VHT_TIMER
Jan 11 16:03:55 raspberrypi kernel: [162813.653541] RTW: [C2H], C2H_TXBF!!
Jan 11 16:03:55 raspberrypi kernel: [162813.653564] RTW: SOUNDING_FW_VHT_TIMER
Jan 11 16:03:56 raspberrypi kernel: [162815.101521] RTW: [C2H], C2H_TXBF!!
Jan 11 16:03:56 raspberrypi kernel: [162815.101543] RTW: SOUNDING_FW_VHT_TIMER
Jan 11 16:03:57 raspberrypi kernel: [162815.659054] RTW: rtl8812_sreset_xmit_status_check tx hang
Jan 11 16:03:58 raspberrypi kernel: [162816.549282] RTW: [C2H], C2H_TXBF!!
Jan 11 16:03:58 raspberrypi kernel: [162816.549304] RTW: SOUNDING_FW_VHT_TIMER
Jan 11 16:03:59 raspberrypi kernel: [162817.997348] RTW: [C2H], C2H_TXBF!!
Jan 11 16:03:59 raspberrypi kernel: [162817.997376] RTW: SOUNDING_FW_VHT_TIMER
Jan 11 16:04:00 raspberrypi kernel: [162819.445114] RTW: [C2H], C2H_TXBF!!
Jan 11 16:04:00 raspberrypi kernel: [162819.445134] RTW: SOUNDING_FW_VHT_TIMER
Jan 11 16:04:01 raspberrypi CRON[31279]: (root) CMD (/bin/echo $(/bin/date) >> /var/log/hostapd.log)
Jan 11 16:04:02 raspberrypi kernel: [162820.892497] RTW: [C2H], C2H_TXBF!!
Jan 11 16:04:02 raspberrypi kernel: [162820.892520] RTW: SOUNDING_FW_VHT_TIMER
Jan 11 16:04:02 raspberrypi kernel: [162821.262097] RTW: OnDisassoc(wlan0) - Start to Disconnect
Jan 11 16:04:02 raspberrypi kernel: [162821.262186] RTW: OnDisassoc(wlan0) reason=3, ta=a4:93:3f:8e:4a:58
Jan 11 16:04:02 raspberrypi kernel: [162821.262250] RTW: rtw_cfg80211_indicate_sta_disassoc(wlan0)
Jan 11 16:04:02 raspberrypi kernel: [162821.262345] RTW: rtw_ht_operation_update current operation mode=0x4
Jan 11 16:04:02 raspberrypi kernel: [162821.262377] RTW: rtw_ht_operation_update new operation mode=0x0 changes=1
Jan 11 16:04:02 raspberrypi kernel: [162821.262406] RTW: update_bcn_htcap_ie
Jan 11 16:04:02 raspberrypi kernel: [162821.262435] RTW: update_bcn_htinfo_ie current operation mode=0x0
Jan 11 16:04:02 raspberrypi kernel: [162821.262511] RTW: num_sta_40mhz_intolerant(0), 20mhz_width_req(0), intolerant_ch_rpt(0), olbc(0)
Jan 11 16:04:02 raspberrypi kernel: [162821.262539] RTW: update_bcn_htinfo_ie:switching back to 40Mhz
Jan 11 16:04:02 raspberrypi kernel: [162821.262622] RTW: bss_cap_update_on_sta_leave, updated=1
Jan 11 16:04:02 raspberrypi kernel: [162821.262682] RTW: report_del_sta_event(wlan0) a4:93:3f:8e:4a:58 mac_id=2, enqueue:1, res:1
Jan 11 16:04:02 raspberrypi kernel: [162821.263945] RTW: cfg80211_rtw_del_key(wlan0) key_index=0, addr=a4:93:3f:8e:4a:58
Jan 11 16:04:02 raspberrypi kernel: [162821.264664] RTW: cfg80211_rtw_del_key(wlan0) key_index=0, addr=a4:93:3f:8e:4a:58
Jan 11 16:04:02 raspberrypi kernel: [162821.265013] RTW: cfg80211_rtw_change_station(wlan0) mac:a4:93:3f:8e:4a:58
Jan 11 16:04:02 raspberrypi kernel: [162821.265311] RTW: +cfg80211_rtw_del_station(wlan0) mac=a4:93:3f:8e:4a:58
Jan 11 16:04:02 raspberrypi kernel: [162821.265340] RTW: free sta macaddr =a4:93:3f:8e:4a:58
Jan 11 16:04:02 raspberrypi kernel: [162821.265367] RTW: -cfg80211_rtw_del_station(wlan0)
Jan 11 16:04:02 raspberrypi hostapd: wlan0: STA a4:93:3f:8e:4a:58 IEEE 802.11: disassociated
Jan 11 16:04:02 raspberrypi kernel: [162821.280471] RTW: rtw_stadel_event_callback(mac_id=2)=a4:93:3f:8e:4a:58
Jan 11 16:04:02 raspberrypi kernel: [162821.280504] RTW: rtw_sta_mstatus_disc_rpt(wlan0) - mac_id=2
Jan 11 16:04:02 raspberrypi kernel: [162821.281902] RTW: start_addr=1024, data_H:006c0600, data_L:8000c031, macid=2, txrpt_h4b=6c0600
Jan 11 16:04:02 raspberrypi kernel: [162821.284018] RTW: wlan0 wakeup macid=2, ori reg_0x4d4=0x00000000
Jan 11 16:04:02 raspberrypi kernel: [162821.284607] RTW: ### Clean STA_(2) info ###
Jan 11 16:04:02 raspberrypi kernel: [162821.284640] RTW: rtw_release_macid(wlan0) if1, mac_addr:a4:93:3f:8e:4a:58 macid:2
Jan 11 16:04:02 raspberrypi kernel: [162821.284673] RTW: rtw_set_rts_bw connect_to_8812=0,enable=1
Jan 11 16:04:02 raspberrypi kernel: [162821.285089] RTW: clear key for addr:a4:93:3f:8e:4a:58, camid:1
Jan 11 16:04:11 raspberrypi kernel: [162830.219055] RTW: rtl8812_sreset_xmit_status_check tx hang
Jan 11 16:04:13 raspberrypi kernel: [162832.299108] RTW: rtl8812_sreset_xmit_status_check tx hang
[... cut ...]
Jan 11 16:04:17 raspberrypi kernel: [162836.459141] RTW: rtl8812_sreset_xmit_status_check tx hang
Jan 11 16:04:24 raspberrypi kernel: [162842.699176] RTW: rtl8812_sreset_xmit_status_check tx hang
Jan 11 16:04:26 raspberrypi kernel: [162844.779141] RTW: rtl8812_sreset_xmit_status_check tx hang
Jan 11 16:04:28 raspberrypi kernel: [162846.859134] RTW: rtl8812_sreset_xmit_status_check tx hang
Jan 11 16:04:30 raspberrypi kernel: [162848.939180] RTW: rtl8812_sreset_xmit_status_check tx hang
```

I've leaved it in this state for more than 2 days and it didn't recovered by itself.

Thanks :-) !

morrownr, 2022-01-13:
Hi,

I'm sure I have ask for this before but this issue has been ongoing for a while so I want to check. Can you run and post the results of:

sudo dkms status

lsmod

Something is fishy.

frsarieg, 2022-01-14:
Hi @morrownr ,

```
# dkms status
rtl8812au, 5.13.6, 5.10.63-v7+, armv7l: installed
# lsmod
Module                  Size  Used by
8812au               2154496  0
cfg80211              761856  1 8812au
ip6t_REJECT            16384  1
nf_reject_ipv6         16384  1 ip6t_REJECT
nf_log_ipv6            16384  4
xt_recent              24576  4
xt_hl                  16384  22
ip6_tables             28672  52
ip6t_rt                16384  3
nf_log_ipv4            16384  4
nf_log_common          16384  2 nf_log_ipv6,nf_log_ipv4
ipt_REJECT             16384  1
nf_reject_ipv4         16384  1 ipt_REJECT
xt_LOG                 16384  8
nft_limit              16384  11
xt_limit               16384  0
xt_addrtype            16384  4
xt_tcpudp              16384  56
xt_conntrack           16384  24
tcp_diag               16384  0
inet_diag              24576  1 tcp_diag
wireguard              69632  0
curve25519_neon        28672  1 wireguard
libcurve25519_generic    24576  2 curve25519_neon,wireguard
libchacha20poly1305    16384  1 wireguard
chacha_neon            32768  1 libchacha20poly1305
poly1305_arm           24576  1 libchacha20poly1305
ip6_udp_tunnel         16384  1 wireguard
udp_tunnel             24576  1 wireguard
libblake2s             16384  1 wireguard
libblake2s_generic     20480  1 libblake2s
tun                    53248  2
bridge                163840  0
8021q                  32768  0
garp                   16384  1 8021q
stp                    16384  2 garp,bridge
llc                    16384  3 garp,bridge,stp
nft_counter            16384  298
nft_chain_nat          16384  4
xt_MASQUERADE          16384  85
nf_nat                 49152  2 xt_MASQUERADE,nft_chain_nat
nf_conntrack          139264  3 xt_MASQUERADE,xt_conntrack,nf_nat
nf_defrag_ipv6         20480  1 nf_conntrack
nf_defrag_ipv4         16384  1 nf_conntrack
xt_comment             16384  99
nft_compat             20480  367
nf_tables             196608  931 nft_limit,nft_compat,nft_chain_nat,nft_counter
nfnetlink              16384  2 nft_compat,nf_tables
snd_soc_hdmi_codec     20480  1
sha256_generic         16384  0
rfkill                 32768  3 cfg80211
vc4                   270336  2
cec                    49152  1 vc4
drm_kms_helper        241664  3 vc4
snd_soc_core          225280  2 vc4,snd_soc_hdmi_codec
snd_compress           20480  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
syscopyarea            16384  1 drm_kms_helper
sysfillrect            16384  1 drm_kms_helper
sysimgblt              16384  1 drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
raspberrypi_hwmon      16384  0
i2c_bcm2835            16384  0
snd_bcm2835            24576  0
bcm2835_codec          40960  0
snd_pcm               110592  5 snd_compress,snd_pcm_dmaengine,snd_soc_hdmi_codec,snd_bcm2835,snd_soc_core
bcm2835_v4l2           40960  0
bcm2835_isp            32768  0
v4l2_mem2mem           36864  1 bcm2835_codec
snd_timer              32768  1 snd_pcm
bcm2835_mmal_vchiq     32768  3 bcm2835_isp,bcm2835_codec,bcm2835_v4l2
videobuf2_dma_contig    20480  2 bcm2835_isp,bcm2835_codec
videobuf2_vmalloc      16384  1 bcm2835_v4l2
videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_vmalloc
videobuf2_v4l2         32768  4 bcm2835_isp,bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem
snd                    77824  6 snd_compress,snd_soc_hdmi_codec,snd_timer,snd_bcm2835,snd_soc_core,snd_pcm
videobuf2_common       61440  5 bcm2835_isp,bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
videodev              249856  6 bcm2835_isp,bcm2835_codec,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
mc                     45056  6 bcm2835_isp,bcm2835_codec,videobuf2_common,videodev,v4l2_mem2mem,videobuf2_v4l2
vc_sm_cma              32768  2 bcm2835_isp,bcm2835_mmal_vchiq
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
fixed                  16384  0
drm                   520192  3 vc4,drm_kms_helper
fuse                  126976  1
drm_panel_orientation_quirks    16384  1 drm
backlight              20480  1 drm
ip_tables              28672  8
x_tables               32768  15 xt_comment,xt_LOG,ipt_REJECT,ip_tables,nft_compat,xt_tcpudp,xt_MASQUERADE,xt_limit,ip6t_REJECT,xt_recent,xt_addrtype,ip6t_rt,xt_conntrack,ip6_tables,xt_hl
ipv6                  495616  54 nf_reject_ipv6,wireguard,bridge
```
Thanks.

morrownr, 2022-01-14:
Hmmm...

The output of your `lsmod` is a lot bigger than mine and I see a lot of extra items that look to be networking related. It very well could be that one of the networking apps you installed is not interacting well with this driver. It may be that the only practical way to track down what is causing the problem is to flash a new sd with RasPiOS and start fresh. Set up the AP first and see how it does. If it is stable, then we know it was one of the additional apps. You could then slowly install one at a time until you figure out which one is guilty.

frsarieg, 2022-01-15:
Hi @morrownr,

That's OK. I'll try to do it and checking what's going on.

If there is a interaction, my bet is with Wireguard.

Thanks :-)

frsarieg, 2022-01-16:
Hi @morrownr ,

It happened before I think...

```
Jan 16 05:37:59 raspberrypi kernel: [26189.368954] RTW: SOUNDING_FW_VHT_TIMER
Jan 16 05:38:01 raspberrypi kernel: [26190.814888] RTW: [C2H], C2H_TXBF!!
Jan 16 05:38:01 raspberrypi kernel: [26190.814909] RTW: SOUNDING_FW_VHT_TIMER
Jan 16 05:38:02 raspberrypi kernel: [26191.804755] RTW: rtl8812_sreset_xmit_status_check tx hang
Jan 16 05:38:02 raspberrypi kernel: [26192.261194] RTW: [C2H], C2H_TXBF!!
Jan 16 05:38:02 raspberrypi kernel: [26192.261218] RTW: SOUNDING_FW_VHT_TIMER
Jan 16 05:38:04 raspberrypi kernel: [26193.707696] RTW: [C2H], C2H_TXBF!!
Jan 16 05:38:04 raspberrypi kernel: [26193.707716] RTW: SOUNDING_FW_VHT_TIMER
Jan 16 05:38:04 raspberrypi kernel: [26193.884760] RTW: rtl8812_sreset_xmit_status_check tx hang
Jan 16 05:38:05 raspberrypi kernel: [26195.154125] RTW: [C2H], C2H_TXBF!!
Jan 16 05:38:05 raspberrypi kernel: [26195.154146] RTW: SOUNDING_FW_VHT_TIMER
Jan 16 05:38:06 raspberrypi kernel: [26195.964752] RTW: rtl8812_sreset_xmit_status_check tx hang
Jan 16 05:38:07 raspberrypi kernel: [26196.610689] RTW: [C2H], C2H_TXBF!!
Jan 16 05:38:07 raspberrypi kernel: [26196.610711] RTW: SOUNDING_FW_VHT_TIMER
Jan 16 05:38:08 raspberrypi kernel: [26198.048172] RTW: issue_nulldata(wlan0) to ac:d6:18:aa:8c:e3, ch:52, 1/1 in 0 ms
Jan 16 05:38:08 raspberrypi kernel: [26198.048202] RTW: expire_timeout_chk(wlan0) asoc check, ac:d6:18:aa:8c:e3 keep_alive_trycnt=1
Jan 16 05:38:08 raspberrypi kernel: [26198.068523] RTW: [C2H], C2H_TXBF!!
Jan 16 05:38:08 raspberrypi kernel: [26198.068547] RTW: SOUNDING_FW_VHT_TIMER
Jan 16 05:38:09 raspberrypi kernel: [26199.493253] RTW: [C2H], C2H_TXBF!!
Jan 16 05:38:09 raspberrypi kernel: [26199.493275] RTW: SOUNDING_FW_VHT_TIMER
Jan 16 05:38:10 raspberrypi kernel: [26200.128127] RTW: issue_nulldata(wlan0) to ac:d6:18:aa:8c:e3, ch:52, 1/1 in 0 ms
Jan 16 05:38:10 raspberrypi kernel: [26200.128157] RTW: expire_timeout_chk(wlan0) asoc check, ac:d6:18:aa:8c:e3 keep_alive_trycnt=2
Jan 16 05:38:11 raspberrypi kernel: [26200.939459] RTW: [C2H], C2H_TXBF!!
Jan 16 05:38:11 raspberrypi kernel: [26200.939479] RTW: SOUNDING_FW_VHT_TIMER
Jan 16 05:38:12 raspberrypi kernel: [26202.208073] RTW: issue_nulldata(wlan0) to ac:d6:18:aa:8c:e3, ch:52, 1/1 in 0 ms
Jan 16 05:38:12 raspberrypi kernel: [26202.208103] RTW: expire_timeout_chk(wlan0) asoc check, ac:d6:18:aa:8c:e3 keep_alive_trycnt=3
Jan 16 05:38:12 raspberrypi kernel: [26202.386027] RTW: [C2H], C2H_TXBF!!
Jan 16 05:38:12 raspberrypi kernel: [26202.386047] RTW: SOUNDING_FW_VHT_TIMER
Jan 16 05:38:14 raspberrypi kernel: [26203.832473] RTW: [C2H], C2H_TXBF!!
Jan 16 05:38:14 raspberrypi kernel: [26203.832494] RTW: SOUNDING_FW_VHT_TIMER
Jan 16 05:38:14 raspberrypi kernel: [26204.292305] RTW: issue_nulldata(wlan0) to ac:d6:18:aa:8c:e3, ch:52, 1/1 in 0 ms
Jan 16 05:38:14 raspberrypi kernel: [26204.292339] RTW: expire_timeout_chk(wlan0) asoc expire ac:d6:18:aa:8c:e3, state=0x4405
Jan 16 05:38:14 raspberrypi kernel: [26204.292375] RTW: rtw_cfg80211_indicate_sta_disassoc(wlan0)
Jan 16 05:38:14 raspberrypi kernel: [26204.292464] RTW: rtw_ht_operation_update current operation mode=0x4
Jan 16 05:38:14 raspberrypi kernel: [26204.292485] RTW: rtw_ht_operation_update new operation mode=0x0 changes=1
Jan 16 05:38:14 raspberrypi kernel: [26204.292503] RTW: update_bcn_htcap_ie
Jan 16 05:38:14 raspberrypi kernel: [26204.292523] RTW: update_bcn_htinfo_ie current operation mode=0x0
Jan 16 05:38:14 raspberrypi kernel: [26204.292542] RTW: num_sta_40mhz_intolerant(0), 20mhz_width_req(0), intolerant_ch_rpt(0), olbc(0)
Jan 16 05:38:14 raspberrypi kernel: [26204.292560] RTW: update_bcn_htinfo_ie:switching back to 40Mhz
Jan 16 05:38:14 raspberrypi kernel: [26204.292596] RTW: bss_cap_update_on_sta_leave, updated=1
Jan 16 05:38:14 raspberrypi kernel: [26204.292625] RTW: rtw_stadel_event_callback(mac_id=2)=ac:d6:18:aa:8c:e3
Jan 16 05:38:14 raspberrypi kernel: [26204.292644] RTW: rtw_sta_mstatus_disc_rpt(wlan0) - mac_id=2
Jan 16 05:38:14 raspberrypi kernel: [26204.293842] RTW: cfg80211_rtw_del_key(wlan0) key_index=0, addr=ac:d6:18:aa:8c:e3
Jan 16 05:38:14 raspberrypi kernel: [26204.294229] RTW: start_addr=1024, data_H:00000600, data_L:8000382f, macid=2, txrpt_h4b=600
Jan 16 05:38:14 raspberrypi kernel: [26204.294566] RTW: cfg80211_rtw_del_key(wlan0) key_index=0, addr=ac:d6:18:aa:8c:e3
Jan 16 05:38:14 raspberrypi kernel: [26204.294922] RTW: cfg80211_rtw_change_station(wlan0) mac:ac:d6:18:aa:8c:e3
Jan 16 05:38:14 raspberrypi kernel: [26204.295222] RTW: +cfg80211_rtw_del_station(wlan0) mac=ac:d6:18:aa:8c:e3
Jan 16 05:38:14 raspberrypi kernel: [26204.295250] RTW: free sta macaddr =ac:d6:18:aa:8c:e3
Jan 16 05:38:14 raspberrypi kernel: [26204.295279] RTW: -cfg80211_rtw_del_station(wlan0)
Jan 16 05:38:14 raspberrypi hostapd: wlan0: STA ac:d6:18:aa:8c:e3 IEEE 802.11: disassociated
Jan 16 05:38:14 raspberrypi kernel: [26204.296948] RTW: wlan0 wakeup macid=2, ori reg_0x4d4=0x00000000
Jan 16 05:38:14 raspberrypi kernel: [26204.299125] RTW: ### Clean STA_(2) info ###
Jan 16 05:38:14 raspberrypi kernel: [26204.299164] RTW: rtw_release_macid(wlan0) if1, mac_addr:ac:d6:18:aa:8c:e3 macid:2
Jan 16 05:38:14 raspberrypi kernel: [26204.299197] RTW: rtw_set_rts_bw connect_to_8812=0,enable=1
Jan 16 05:38:14 raspberrypi kernel: [26204.299665] RTW: report_del_sta_event(wlan0) ac:d6:18:aa:8c:e3 mac_id=2, enqueue:0, res:1
Jan 16 05:38:14 raspberrypi kernel: [26204.299700] RTW: clear key for addr:ac:d6:18:aa:8c:e3, camid:1
[... cut ...]
fixed                  16384  0
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
bridge                163840  0
stp                    16384  2 garp,bridge
llc                    16384  3 garp,bridge,stp
drm                   520192  3 vc4,drm_kms_helper
fuse                  126976  1
drm_panel_orientation_quirks    16384  1 drm
backlight              20480  1 drm
ip_tables              28672  0
x_tables               32768  1 ip_tables
ipv6                  495616  25 bridge
```

Thanks!

frsarieg, 2022-01-17:
Hi @morrownr ,

That's another thing strange in my setup. Sometimes it loses a lot of ping probes to the address configured in the Pi.

The network config follows the AP bridge rules and it is configured with a static IP address. Other devices connected to the same switch didn't loose any packet.

This morning while it is hanged:

```
100 packets transmitted, 56 received, 44% packet loss, time 101263ms
rtt min/avg/max/mdev = 0.526/0.639/2.108/0.204 ms
```

After a restart of the hostapd and reload the 8812au module I saw ~33 % packet loss. The same occurs after a reboot. This isn't always happen.

Thanks!

frsarieg, 2022-01-17:
Hi again :-)

This is a tcpdump view of a probe of ten pings from my desktop machine (192.168.1.2) to the Rpi2 with the AP (192.168.1.52)

# tcpdump -n -i any icmp and host 192.168.1.2
tcpdump: data link type LINUX_SLL2
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on any, link-type LINUX_SLL2 (Linux cooked v2), snapshot length 262144 bytes
09:26:33.970673 eth0  P   IP 192.168.1.2 > 192.168.1.52: ICMP echo request, id 4140, seq 1, length 64
09:26:33.970673 br0   In  IP 192.168.1.2 > 192.168.1.52: ICMP echo request, id 4140, seq 1, length 64
09:26:33.970798 br0   Out IP 192.168.1.52 > 192.168.1.2: ICMP echo reply, id 4140, seq 1, length 64
09:26:33.970821 eth0  Out IP 192.168.1.52 > 192.168.1.2: ICMP echo reply, id 4140, seq 1, length 64
09:26:34.970745 eth0  P   IP 192.168.1.2 > 192.168.1.52: ICMP echo request, id 4140, seq 2, length 64
09:26:34.970745 br0   In  IP 192.168.1.2 > 192.168.1.52: ICMP echo request, id 4140, seq 2, length 64
09:26:34.970899 br0   Out IP 192.168.1.52 > 192.168.1.2: ICMP echo reply, id 4140, seq 2, length 64
09:26:34.970934 eth0  Out IP 192.168.1.52 > 192.168.1.2: ICMP echo reply, id 4140, seq 2, length 64
09:26:35.989787 eth0  P   IP 192.168.1.2 > 192.168.1.52: ICMP echo request, id 4140, seq 3, length 64
09:26:35.989787 br0   In  IP 192.168.1.2 > 192.168.1.52: ICMP echo request, id 4140, seq 3, length 64
09:26:35.989915 br0   Out IP 192.168.1.52 > 192.168.1.2: ICMP echo reply, id 4140, seq 3, length 64
09:26:35.989938 eth0  Out IP 192.168.1.52 > 192.168.1.2: ICMP echo reply, id 4140, seq 3, length 64
09:26:37.013684 eth0  P   IP 192.168.1.2 > 192.168.1.52: ICMP echo request, id 4140, seq 4, length 64
09:26:37.013684 br0   In  IP 192.168.1.2 > 192.168.1.52: ICMP echo request, id 4140, seq 4, length 64
09:26:37.013805 br0   Out IP 192.168.1.52 > 192.168.1.2: ICMP echo reply, id 4140, seq 4, length 64
09:26:37.013831 eth0  Out IP 192.168.1.52 > 192.168.1.2: ICMP echo reply, id 4140, seq 4, length 64
09:26:39.061679 eth0  P   IP 192.168.1.2 > 192.168.1.52: ICMP echo request, id 4140, seq 6, length 64
09:26:39.061679 br0   In  IP 192.168.1.2 > 192.168.1.52: ICMP echo request, id 4140, seq 6, length 64
09:26:39.061790 br0   Out IP 192.168.1.52 > 192.168.1.2: ICMP echo reply, id 4140, seq 6, length 64
09:26:39.061811 eth0  Out IP 192.168.1.52 > 192.168.1.2: ICMP echo reply, id 4140, seq 6, length 64
09:26:42.133673 eth0  P   IP 192.168.1.2 > 192.168.1.52: ICMP echo request, id 4140, seq 9, length 64
09:26:42.133673 br0   In  IP 192.168.1.2 > 192.168.1.52: ICMP echo request, id 4140, seq 9, length 64
09:26:42.133841 br0   Out IP 192.168.1.52 > 192.168.1.2: ICMP echo reply, id 4140, seq 9, length 64
09:26:42.133877 eth0  Out IP 192.168.1.52 > 192.168.1.2: ICMP echo reply, id 4140, seq 9, length 64

40 % ping loss. This behaviour also was happening with the Rpi4. Load and CPU usage is normal (below 0,1 of load)

Thanks!

morrownr, 2022-01-17:
Good day @frsarieg 

There were two things that caught my eye when scanning the information you posted:

> Install: bridge-utils

It is not clear to me why you install this package. What setup guide are you using?

> Commandline: apt-get dist-upgrade

If you are starting with a freshly installed sd card, it is not clear why you used this command. If you are doing the upgrade in the terminal, I can see using this command:

$ sudo apt full-upgrade

I have been pondering this issue. The big question is my mind is why am I not seeing more reports that are similar to this? This repo is a very very heavily used repo as the 8812au chipset has been in production for a long time and is popular with Linux users. I have two working theories at this time:

- Your Pi is sitting next to a heavily used microwave oven. Electromagnetic interference could account for this. Maybe.

- You are using a setup guide that is faulty. When I saw that you installed bridge-utils, an alarm went off. I don't use bridge-utils. The guide I use, that I made and continually test, only requires the installation of one package: hostapd. I keep it simple on purpose.

Talk to me about the setup guide you are using.

frsarieg, 2022-01-17:
Hi @morrownr,

This is the content of the bridge-utils package :

```
# apt-file list bridge-utils
bridge-utils: /etc/default/bridge-utils   
bridge-utils: /etc/network/if-down.d/bridge
bridge-utils: /etc/network/if-post-down.d/bridge
bridge-utils: /etc/network/if-pre-up.d/bridge
bridge-utils: /lib/bridge-utils/bridge-utils.sh
bridge-utils: /lib/bridge-utils/ifupdown.sh
bridge-utils: /lib/udev/bridge-network-interface
bridge-utils: /lib/udev/rules.d/60-bridge-network-interface.rules
bridge-utils: /sbin/brctl
bridge-utils: /usr/share/doc/bridge-utils/FAQ
bridge-utils: /usr/share/doc/bridge-utils/FIREWALL
bridge-utils: /usr/share/doc/bridge-utils/HOWTO
bridge-utils: /usr/share/doc/bridge-utils/NEWS.Debian.gz
bridge-utils: /usr/share/doc/bridge-utils/PROJECTS
bridge-utils: /usr/share/doc/bridge-utils/README
bridge-utils: /usr/share/doc/bridge-utils/README.Debian
bridge-utils: /usr/share/doc/bridge-utils/THANKS
bridge-utils: /usr/share/doc/bridge-utils/WISHLIST
bridge-utils: /usr/share/doc/bridge-utils/changelog.Debian.gz
bridge-utils: /usr/share/doc/bridge-utils/changelog.gz
bridge-utils: /usr/share/doc/bridge-utils/copyright
bridge-utils: /usr/share/doc/bridge-utils/examples/bridge.service
bridge-utils: /usr/share/doc/bridge-utils/examples/hibernate
bridge-utils: /usr/share/doc/bridge-utils/examples/interfaces.hotplug.wifi
bridge-utils: /usr/share/doc/bridge-utils/examples/interfaces.networkmanager
bridge-utils: /usr/share/doc/bridge-utils/examples/pm-utils
bridge-utils: /usr/share/man/man5/bridge-utils-interfaces.5.gz
bridge-utils: /usr/share/man/man8/brctl.8.gz
```

I've installed it in order to get the `brctl show` functionality. It's not a part of the "AP setup".
```

# brctl show
bridge name	bridge id		STP enabled	interfaces
br0		8000.00c0caaff497	no		eth0
							wlan0
```

bridge-utils is now purged and the Raspberry Pi rebooted.

I am following the official guide :-) https://github.com/morrownr/8812au-20210629/blob/main/docs/Bridged_Wireless_Access_Point.md

The Raspberry is far from the microwave :D

Thanks!

morrownr, 2022-01-18:
Thought you might like the microwave thing. ;D

frsarieg, 2022-01-19:
> Thought you might like the microwave thing. ;D

Hahahahaha!

I think you've nailed it: not any ping loss nor the "sreset_xmit_status_check tx hang" after the purge of the `bridge-utils` package and the reboot.

This is just for 24 hours... we'll see in the next 48 hours if the problem appears again. If not, I will reinstall the `bridge-utils` again and watch and expect for the issue.

Thanks!

frsarieg, 2022-01-19:
Well... too fast saying...

```
Jan 19 13:06:24 raspberrypi kernel: [142505.925544] RTW: [C2H], C2H_TXBF!!
Jan 19 13:06:24 raspberrypi kernel: [142505.925560] RTW: SOUNDING_FW_VHT_TIMER
Jan 19 13:06:26 raspberrypi kernel: [142507.120124] RTW: rtl8812_sreset_xmit_status_check tx hang
Jan 19 13:06:26 raspberrypi kernel: [142507.367525] RTW: [C2H], C2H_TXBF!!
Jan 19 13:06:26 raspberrypi kernel: [142507.367542] RTW: SOUNDING_FW_VHT_TIMER
Jan 19 13:06:27 raspberrypi kernel: [142508.810396] RTW: [C2H], C2H_TXBF!!
Jan 19 13:06:27 raspberrypi kernel: [142508.810411] RTW: SOUNDING_FW_VHT_TIMER
Jan 19 13:06:28 raspberrypi kernel: [142509.200141] RTW: rtl8812_sreset_xmit_status_check tx hang
Jan 19 13:06:29 raspberrypi kernel: [142510.252492] RTW: [C2H], C2H_TXBF!!
Jan 19 13:06:29 raspberrypi kernel: [142510.252508] RTW: SOUNDING_FW_VHT_TIMER
Jan 19 13:06:30 raspberrypi kernel: [142511.695970] RTW: [C2H], C2H_TXBF!!
Jan 19 13:06:30 raspberrypi kernel: [142511.695985] RTW: SOUNDING_FW_VHT_TIMER
Jan 19 13:06:32 raspberrypi kernel: [142513.176049] RTW: OnDeAuth(wlan0) - Start to Disconnect
Jan 19 13:06:32 raspberrypi kernel: [142513.176094] RTW: OnDeAuth(wlan0) reason=4, ta=ac:d6:18:aa:8c:e3
Jan 19 13:06:32 raspberrypi kernel: [142513.176129] RTW: rtw_cfg80211_indicate_sta_disassoc(wlan0)
Jan 19 13:06:32 raspberrypi kernel: [142513.176205] RTW: rtw_ht_operation_update current operation mode=0x4
Jan 19 13:06:32 raspberrypi kernel: [142513.176228] RTW: rtw_ht_operation_update new operation mode=0x4 changes=0
Jan 19 13:06:32 raspberrypi kernel: [142513.176247] RTW: bss_cap_update_on_sta_leave, updated=0
Jan 19 13:06:32 raspberrypi kernel: [142513.176282] RTW: report_del_sta_event(wlan0) ac:d6:18:aa:8c:e3 mac_id=2, enqueue:1, res:1
Jan 19 13:06:32 raspberrypi kernel: [142513.177240] RTW: cfg80211_rtw_del_key(wlan0) key_index=0, addr=ac:d6:18:aa:8c:e3
Jan 19 13:06:32 raspberrypi kernel: [142513.177669] RTW: cfg80211_rtw_del_key(wlan0) key_index=0, addr=ac:d6:18:aa:8c:e3
Jan 19 13:06:32 raspberrypi kernel: [142513.177853] RTW: cfg80211_rtw_change_station(wlan0) mac:ac:d6:18:aa:8c:e3
Jan 19 13:06:32 raspberrypi kernel: [142513.178012] RTW: +cfg80211_rtw_del_station(wlan0) mac=ac:d6:18:aa:8c:e3
Jan 19 13:06:32 raspberrypi kernel: [142513.178030] RTW: free sta macaddr =ac:d6:18:aa:8c:e3
Jan 19 13:06:32 raspberrypi kernel: [142513.178048] RTW: -cfg80211_rtw_del_station(wlan0)
Jan 19 13:06:32 raspberrypi hostapd: wlan0: STA ac:d6:18:aa:8c:e3 IEEE 802.11: disassociated
Jan 19 13:06:32 raspberrypi kernel: [142513.239938] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
Jan 19 13:06:32 raspberrypi kernel: [142513.349939] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
Jan 19 13:06:32 raspberrypi kernel: [142513.459962] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
Jan 19 13:06:32 raspberrypi kernel: [142513.569953] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
Jan 19 13:06:32 raspberrypi kernel: [142513.680045] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
Jan 19 13:06:32 raspberrypi kernel: [142513.789935] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
Jan 19 13:06:32 raspberrypi kernel: [142513.899939] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
Jan 19 13:06:32 raspberrypi kernel: [142513.902112] RTW: _alloc_mgtxmitframe(wlan0) alloc xmitbuf fail
Jan 19 13:06:32 raspberrypi kernel: [142513.902141] RTW: issue_beacon, alloc mgnt frame fail
Jan 19 13:06:32 raspberrypi kernel: [142513.904007] RTW: _alloc_mgtxmitframe(wlan0) alloc xmitbuf fail
Jan 19 13:06:32 raspberrypi kernel: [142513.904029] RTW: issue_beacon, alloc mgnt frame fail
Jan 19 13:06:32 raspberrypi kernel: [142513.905884] RTW: _alloc_mgtxmitframe(wlan0) alloc xmitbuf fail
Jan 19 13:06:32 raspberrypi kernel: [142513.905906] RTW: issue_beacon, alloc mgnt frame fail
Jan 19 13:06:32 raspberrypi kernel: [142513.907779] RTW: _alloc_mgtxmitframe(wlan0) alloc xmitbuf fail
```

Next step: install from scratch again.

Thanks!

morrownr, 2022-01-19:
Let me suggest this since you are going to do a clean installation again: Set it up for only one adapter. Another adapter can be added later. Use the Alfa ACH. The guide here in the `docs` folder gives examples of dual band and single band setups.

This thread is getting long but I think your other adapter is based on a 8812bu chipset? How about leaving it out for now?

frsarieg, 2022-01-19:
> Let me suggest this since you are going to do a clean installation again: Set it up for only one adapter. Another adapter can be added later. Use the Alfa ACH. The guide here in the `docs` folder [...]

Hi @morrownr ,

I'm using only the Alfa ACH now. Both are based on the 8812au chipset.

The setup uses only 5g.

When I reinstall, I will post about the progress.

Thanks! :-)

morrownr, 2022-01-19:
I'm getting a little slow in my old age so hang in there with me...

Are you saying that you were using two adapters with 8812au chipsets and this driver in the same Pi?

If so, no wonder you are having problems. Realtek drivers don't support 2 adapters with the same chipset in the same system.

frsarieg, 2022-01-19:
> I'm getting a little slow in my old age so hang in there with me...
> 
> Are you saying that you were using two adapters with 8812au chipsets and this driver in the same Pi?
> 
> If so, no wonder you are having problems. Realtek drivers don't support 2 adapters with the same chipset in the same system.

Hi @morrownr.

Only the Alfa is connected to the Pi :-)

Thanks.

frsarieg, 2022-01-26:
Hi @morrownr,

This is the status.

I've tested a few things that didn't help with the issue:

- I've setted a static MAC address on the bridge interface (br0). The MAC is different from the ethernet and WiFi devices.
- I've changed the WiFi device from the Alfa to the Brostrend
- I've tested again the rest of the USB ports
- I've Connected the Pi2 to another switch. Now is connected to the switch in my bedroom sharing with the my desktop PC. I think that this solved the ping loss. Probably a problem with the router, where is connected before. I need to investigate more about this.

This is now the setup and, probably this is the change that matters:

- `rtw_beamform_cap=0` in the options of the driver. I've disabled the previous SU Beamformer setting. Is working now for 72 hours in a row without any problem.

Let's see what happens in the next days. I plan to leave it without any restart until the next week. If the problem doesn't appear, I will reactivate the option.

Thanks!

morrownr, 2022-01-27:
Keep me posted. This is a hair puller.

frsarieg, 2022-02-01:
I must insist: I've never-never-never plugged two adapters at the same time in the same system. Probably it's my fault with my bad english.

I've said this before in:
https://github.com/morrownr/8812au-20210629/issues/22#issuecomment-1016602943

And any of the `lsusb` I've posted show more than one adapter plugged in.

Rpi2 with the `rtw_beamform_cap=0` failed short after:

```
# systemctl status hostapd.service
● hostapd.service - Access point and authentication server for Wi-Fi and Ethernet
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2022-02-01 11:40:16 CET; 6h ago
```

```
Feb  1 15:53:22 raspberrypi kernel: [851632.006884] RTW: rtl8812_sreset_xmit_status_check tx hang
Feb  1 15:53:23 raspberrypi kernel: [851633.035852] RTW: [C2H], C2H_TXBF!!
Feb  1 15:53:23 raspberrypi kernel: [851633.035875] RTW: SOUNDING_FW_VHT_TIMER
Feb  1 15:53:25 raspberrypi kernel: [851634.478276] RTW: [C2H], C2H_TXBF!!
Feb  1 15:53:25 raspberrypi kernel: [851634.478299] RTW: SOUNDING_FW_VHT_TIMER
Feb  1 15:53:26 raspberrypi kernel: [851635.920715] RTW: [C2H], C2H_TXBF!!
Feb  1 15:53:26 raspberrypi kernel: [851635.920736] RTW: SOUNDING_FW_VHT_TIMER
Feb  1 15:53:28 raspberrypi kernel: [851637.363355] RTW: [C2H], C2H_TXBF!!
Feb  1 15:53:28 raspberrypi kernel: [851637.363377] RTW: SOUNDING_FW_VHT_TIMER
Feb  1 15:53:29 raspberrypi kernel: [851638.806004] RTW: [C2H], C2H_TXBF!!
Feb  1 15:53:29 raspberrypi kernel: [851638.806026] RTW: SOUNDING_FW_VHT_TIMER
Feb  1 15:53:30 raspberrypi kernel: [851639.119086] RTW: OnDisassoc(wlan0) - Start to Disconnect
Feb  1 15:53:30 raspberrypi kernel: [851639.119173] RTW: OnDisassoc(wlan0) reason=3, ta=a4:93:3f:8e:4a:58
Feb  1 15:53:30 raspberrypi kernel: [851639.119236] RTW: rtw_cfg80211_indicate_sta_disassoc(wlan0)
Feb  1 15:53:30 raspberrypi kernel: [851639.119323] RTW: rtw_ht_operation_update current operation mode=0x4
Feb  1 15:53:30 raspberrypi kernel: [851639.119352] RTW: rtw_ht_operation_update new operation mode=0x0 changes=1
Feb  1 15:53:30 raspberrypi kernel: [851639.119381] RTW: update_bcn_htcap_ie
Feb  1 15:53:30 raspberrypi kernel: [851639.119414] RTW: update_bcn_htinfo_ie current operation mode=0x0
Feb  1 15:53:30 raspberrypi kernel: [851639.119444] RTW: num_sta_40mhz_intolerant(0), 20mhz_width_req(0), intolerant_ch_rpt(0), olbc(0)
Feb  1 15:53:30 raspberrypi kernel: [851639.119471] RTW: update_bcn_htinfo_ie:switching back to 40Mhz
Feb  1 15:53:30 raspberrypi kernel: [851639.119517] RTW: bss_cap_update_on_sta_leave, updated=1
Feb  1 15:53:30 raspberrypi kernel: [851639.119574] RTW: report_del_sta_event(wlan0) a4:93:3f:8e:4a:58 mac_id=2, enqueue:1, res:1
Feb  1 15:53:30 raspberrypi kernel: [851639.120748] RTW: cfg80211_rtw_del_key(wlan0) key_index=0, addr=a4:93:3f:8e:4a:58
Feb  1 15:53:30 raspberrypi kernel: [851639.121414] RTW: cfg80211_rtw_del_key(wlan0) key_index=0, addr=a4:93:3f:8e:4a:58
Feb  1 15:53:30 raspberrypi kernel: [851639.121697] RTW: cfg80211_rtw_change_station(wlan0) mac:a4:93:3f:8e:4a:58
Feb  1 15:53:30 raspberrypi kernel: [851639.121939] RTW: +cfg80211_rtw_del_station(wlan0) mac=a4:93:3f:8e:4a:58
Feb  1 15:53:30 raspberrypi kernel: [851639.121967] RTW: free sta macaddr =a4:93:3f:8e:4a:58
Feb  1 15:53:30 raspberrypi kernel: [851639.121990] RTW: -cfg80211_rtw_del_station(wlan0)
Feb  1 15:53:30 raspberrypi kernel: [851639.141632] RTW: rtw_stadel_event_callback(mac_id=2)=a4:93:3f:8e:4a:58
Feb  1 15:53:30 raspberrypi kernel: [851639.141665] RTW: rtw_sta_mstatus_disc_rpt(wlan0) - mac_id=2
Feb  1 15:53:30 raspberrypi kernel: [851639.143006] RTW: start_addr=1024, data_H:01040600, data_L:8000302e, macid=2, txrpt_h4b=1040600
Feb  1 15:53:30 raspberrypi kernel: [851639.145027] RTW: wlan0 wakeup macid=2, ori reg_0x4d4=0x00000000
Feb  1 15:53:30 raspberrypi kernel: [851639.145616] RTW: ### Clean STA_(2) info ###
Feb  1 15:53:30 raspberrypi kernel: [851639.145649] RTW: rtw_release_macid(wlan0) if1, mac_addr:a4:93:3f:8e:4a:58 macid:2
Feb  1 15:53:30 raspberrypi kernel: [851639.145682] RTW: rtw_set_rts_bw connect_to_8812=0,enable=1
Feb  1 15:53:30 raspberrypi kernel: [851639.146075] RTW: clear key for addr:a4:93:3f:8e:4a:58, camid:1
Feb  1 15:53:39 raspberrypi kernel: [851648.646893] RTW: rtl8812_sreset_xmit_status_check tx hang
Feb  1 15:53:41 raspberrypi kernel: [851650.726921] RTW: rtl8812_sreset_xmit_status_check tx hang
Feb  1 15:53:43 raspberrypi kernel: [851652.806936] RTW: rtl8812_sreset_xmit_status_check tx hang
Feb  1 15:53:45 raspberrypi kernel: [851654.886966] RTW: rtl8812_sreset_xmit_status_check tx hang
Feb  1 15:53:47 raspberrypi kernel: [851656.967011] RTW: rtl8812_sreset_xmit_status_check tx hang
Feb  1 15:53:49 raspberrypi kernel: [851659.047052] RTW: rtl8812_sreset_xmit_status_check tx hang
Feb  1 15:53:52 raspberrypi kernel: [851661.127037] RTW: rtl8812_sreset_xmit_status_check tx hang
Feb  1 15:53:54 raspberrypi kernel: [851663.207094] RTW: rtl8812_sreset_xmit_status_check tx hang
```

Five hours later the restart, the issue appears.

Now I've changed to `rtw_beamform_cap=3` and waiting...

So **this issue is not solved**: we are now understanding probably where the problem is.

Thanks.

frsarieg, 2022-02-01:
I can confirm the same issue with `rtw_beamform_cap=3`.

Now trying `rtw_beamform_cap=11`: "11= SU Beamformer and SU Beamformee and MU Beamformee"

Thanks!

morrownr, 2022-02-01:
> I must insist: I've never-never-never plugged two adapters at the same time in the same system. Probably it's my fault with my bad english.

My bad. Your English is fine. This thread is so long at this point that I had decided to reread it again to make sure I knew what was going on. I look at a lot of reports. I guess I misunderstood something that was posted.

It seems that you are seeing more stable results. What is your assessment?

frsarieg, 2022-02-01:
Hi @morrownr ,

First of all: thank you so much for your time and effort! :-)

With the `rtw_beamform_cap=0` configured I've never seen a problem in eight days.

I think that the problem is there. `1` and `3` values shows the issue. 11 is now configured and under investigation.

Thanks again :-)

frsarieg, 2022-02-02:
Hi @morrownr,

Same issue after configuring the `rtw_beamform_cap=11`.

Therefore neither of the the SU Beamformer options (`1`, `3` and `11`) are working well.

Now I've reverted to `rtw_beamform_cap=0` (disabled) where I find it 100% stable.

Thanks!

morrownr, 2022-02-02:
Hi @frsarieg 

Would you mind doing an additional test?

`rtw_beamform_cap=10`

That is actually the default that Realtek uses. If you do not have `rtw_beamform_cap=` in 8812au.conf then that `10` is what is used. I'd appreciate you testing to see what the results are.

I have researched beamforming a lot. You almost need to be an EE to fully understand it and the benefits as shown by real world testing are certainly very limited. The original reason I documented `rtw_beamform_cap=` was so that users could turn it off although I don't say that anywhere. One person that is pretty knowledgable dug into the beamforming code in these Linux drivers and was not impressed.

Given what you have seen here and depending on the results of `10`, it may be best if I use a default of `0` as I like the defaults to lead to a driver that is as trouble free as possible.

Thanks.

frsarieg, 2022-02-08:
Hi @morrownr,

Working without any issue for almost 5 days in a row. `rtw_beamform_cap=10`.

I guess that the problem is with the SU Beamformer config.

Did you test it with your devices and setups?

Thanks!

morrownr, 2022-03-14:
Hi @frsarieg 

Been sick lately and am behind. What is the status?

> Assuming that there are a problem with the beamformer capabilities, Is there any way in which we can help? Are the engineers at Realtek receiving bugs or comments?

You would have to ask. There is no known way for end users to report bugs to Realtek. I have gone to great lengths to find a way.

frsarieg, 2022-03-17:
Hi @morrownr ,

All is working fine now with the beamformer/beamforming options disabled.

```
root@raspberrypi:~# systemctl status hostapd.service 
● hostapd.service - Access point and authentication server for Wi-Fi and Ethernet
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2022-02-03 18:42:06 CET; 1 months 11 days ago
```
Working hard without any issue.

Thanks!

morrownr, 2022-03-19:
Hi @frsarieg ,

It is only took 55 messages over 4 months!