morrownr/8812au-20210629 #24: Suggest to add a stable branch

heku, 2021-12-22:
Hi Nick,
I've found that the 'main' branch has some commits day by day, thanks for your effort firstly.
As there are changes day by day, I'm not sure when it's a good chance to update my local driver, so I suggest to add a stable branch, so that if there is a batch of changes you think is completed and stable, then you push to that branch. for end-users like me, then we know it's chance to update our local driver.

morrownr, 2021-12-22:
Hi @heku 

I'm glad you brought this issue up. I have pondered this issue to one degree or another for the last two years. I know Linus releases a new stable kernel every 2.5 months or so which is part of the Linux development model. It works because there was flexibility to change in order to take advantage of the resources that are available. Maybe a stable branch release schedule similar to the one Linus uses would work?

Some things that may not be obvious to you and others that stop by to get a driver is that there is far more going on here than this repo. I realize that most people show up here via a link that takes them directly into a repo so they don't see the other repos. What does this have to do with anything? A lot actually. I have 6 separate repos for modern Realtek drivers. Five of the six are public. What I do with the Realtek drivers is I take fixes and improvements from one driver and apply them to the other drivers if applicable. That is why you see work going on here that does not appear to be related to a bug report. That allows bugs to be fixed and enhancements to be applied before they are even reported in most cases. This particular repo has seen over 3,200 clones in the last 2 weeks. Isn't it amazing how few bug reports there are? My point here is that I do not use Main as a test branch. I keep a separate test branch on my local system where I work and test. I have a little lab and I test with multiple OSs on various hardware. I do not push to Main unless I think I am pushing stable code.

So, could keep additional branches and development a release schedule? I could and I will think about it. However, that would add to complexity and work load. If you take a look around, I think you will see that my work load is already heavy. I also maintain a repo for Mediatek drivers but it is informational since the drivers are in the kernel. I also maintain USB-WiFi, which I consider the Crown Jewel:

https://github.com/morrownr/USB-WiFi

That repo has a lot of information regarding using USB WiFi adapters on Linux and it currently has 76 direct links to adapters that use in-kernel drivers. Using adapters that use in-kernel drivers has big benefits and the main thing that keeps that from happening is the lack of information available to Linux users. Hopefully USB-WiFi is helping in that regard.

If you can come up with a way for me to help with your concern that does not add to the complexity and workload, I am all for it. Let me know.

Nick

heku, 2021-12-22:
Thanks for the detail explanation, yes, lots of things out of my understand before, I feel very grateful for your heavy work. Nick, thank you!

closed 2021-12-22 by heku