morrownr/8812au-20210629 #25: airgeddon does not work properly when concurrent mode is enabled

LuckyFishGeek, 2021-12-28:
iw dev :
[image]

When I select wlan0 as the device, I can scan and attack, but can't establish a phishing hotspot and start gon correctly
[image]

When I select wlan1 as the device, it can only scan but not attack

Can this issue be further addressed?

Very much looking forward to it.

Thank you for your work.

usama7628674, 2021-12-28:
Concurrent mode does not work as you would expect it to work with other adapters from different chipset manufacturer. The reason is Realtek does not support virtual interface because the driver does not get written for `mac80211`. To launch an evil twin attack you need to get a second adapter or an adapter that supports a virtual interface.

usama7628674, 2021-12-28:
More info here #9

closed 2021-12-28 by LuckyFishGeek