morrownr/8812au-20210629 #26: build errors 8812au-20210629 on RH8

imuji2022, 2022-01-03:
Fresh install of Redhat 8 - I am getting following error in os_intfs.c
I tried both  8812AU and 8821AU drivers.

[root@dev1 8812au-20210629]# uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev
Linux dev1.localdomain 4.18.0-348.7.1.el8_5.x86_64 #1 SMP Wed Dec 8 21:51:17 EST 2021 x86_64 x86_64 x86_64 GNU/Linux
EFI variables are not supported on this system
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 003: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 008 Device 002: ID 046d:c315 Logitech, Inc. Classic Keyboard 200
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 002: ID 1912:2824  
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 005: ID 0bda:0811 Realtek Semiconductor Corp. Realtek 8812AU/8821AU 802.11ac WLAN Adapter [USB Wireless Dual-Band Adapter 2.4/5Ghz]
Bus 001 Device 007: ID 04e8:6863 Samsung Electronics Co., Ltd Galaxy series, misc. (tethering mode)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
[root@dev1 8812au-20210629]# 

Errors 1:

/root/wifi_drivers/8812au-20210629/os_dep/linux/os_intfs.c:1865:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, void *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, void *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
  .ndo_select_queue = rtw_select_queue,
                      ^~~~~~~~~~~~~~~~
/root/wifi_drivers/8812au-20210629/os_dep/linux/os_intfs.c:1865:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:316: /root/wifi_drivers/8812au-20210629/os_dep/linux/os_intfs.o] Error 1
make[1]: *** [Makefile:1571: _module_/root/wifi_drivers/8812au-20210629] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-348.7.1.el8_5.x86_64'
make: *** [Makefile:2508: modules] Error 2

After editing Makefile with 
USER_EXTRA_CFLAGS=-Wno-error=incompatible-pointer-types
I get following Error:

Error2:

  CC [M]  /root/wifi_drivers/8812au-20210629/os_dep/linux/ioctl_cfg80211.o
/root/wifi_drivers/8812au-20210629/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
/root/wifi_drivers/8812au-20210629/os_dep/linux/ioctl_cfg80211.c:469:3: error: too few arguments to function ‘cfg80211_ch_switch_started_notify’
   cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /root/wifi_drivers/8812au-20210629/include/osdep_service_linux.h:98,
                 from /root/wifi_drivers/8812au-20210629/include/osdep_service.h:54,
                 from /root/wifi_drivers/8812au-20210629/include/drv_types.h:27,
                 from /root/wifi_drivers/8812au-20210629/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:7673:6: note: declared here
 void cfg80211_ch_switch_started_notify(struct net_device *dev,
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/wifi_drivers/8812au-20210629/os_dep/linux/ioctl_cfg80211.c: At top level:
/root/wifi_drivers/8812au-20210629/os_dep/linux/ioctl_cfg80211.c:10477:3: error: ‘struct cfg80211_ops’ has no member named ‘mgmt_frame_register’
  .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
   ^~~~~~~~~~~~~~~~~~~
/root/wifi_drivers/8812au-20210629/os_dep/linux/ioctl_cfg80211.c:10477:25: warning: initialization of ‘int (*)(struct wiphy *, struct wireless_dev *, u64)’ {aka ‘int (*)(struct wiphy *, struct wireless_dev *, long long unsigned int)’} from incompatible pointer type ‘void (*)(struct wiphy *, struct wireless_dev *, u16,  bool)’ {aka ‘void (*)(struct wiphy *, struct wireless_dev *, short unsigned int,  _Bool)’} [-Wincompatible-pointer-types]
  .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/wifi_drivers/8812au-20210629/os_dep/linux/ioctl_cfg80211.c:10477:25: note: (near initialization for ‘rtw_cfg80211_ops.mgmt_tx_cancel_wait’)
make[2]: *** [scripts/Makefile.build:316: /root/wifi_drivers/8812au-20210629/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:1571: _module_/root/wifi_drivers/8812au-20210629] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-348.7.1.el8_5.x86_64'
make: *** [Makefile:2508: modules] Error 2

morrownr, 2022-01-03:
Hi @imuji2022 

I am aware that there could be a problem with RT8 because a user of the 88x2bu repo brought it to my attention and provided a PR that fixed the problem.

https://github.com/morrownr/88x2bu-20210702

I am willing to port the needed changes to this driver IF you tell me that you will stick around and test the version with the changes includes. It may take me a few hours of a day or two before I can post a patched edition of this driver because I will have to test to make sure the changes do not break things.

Let me know.

Regards

imuji2022, 2022-01-04:
Sure. I can do that.

morrownr, 2022-01-04:
Hi @imuji2022 

The patches have been ported and merged. The easiest way to do a clean update is probably to run `./remove-driver.sh` and then delete the driver folder followed by a fresh installation.

Please test and report back.

Regards,

Nick

imuji2022, 2022-01-04:
Hello Nick,

I ran "make" command after cloning new repo. I tried three times.

1. First as is. I got "Result 1"
2. Second, I modified Makefile, inserted "USER_EXTRA_CFLAGS=-Wno-error=incompatible-pointer-types" to line 1. See Result 2.
3. I fixed, os_intfs.c:1725:70: error: missing '(' in expression by removing extra ")"  from  "defined(RHEL8))". See Result 3

Thank you,
Imuji

Result 1:
-------------
/root/wifi_drivers/8812au-20210629/os_dep/linux/os_intfs.c:1725:70: error: missing '(' in expression
  #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 19, 0) || defined(RHEL8))
                                                                      ^
/root/wifi_drivers/8812au-20210629/os_dep/linux/os_intfs.c:1865:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, void *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, void *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
  .ndo_select_queue = rtw_select_queue,
                      ^~~~~~~~~~~~~~~~
/root/wifi_drivers/8812au-20210629/os_dep/linux/os_intfs.c:1865:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:316: /root/wifi_drivers/8812au-20210629/os_dep/linux/os_intfs.o] Error 1
make[1]: *** [Makefile:1571: _module_/root/wifi_drivers/8812au-20210629] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-348.7.1.el8_5.x86_64'
make: *** [Makefile:2512: modules] Error 2

Result 2
----------------
/root/wifi_drivers/8812au-20210629/os_dep/linux/os_intfs.c:1725:70: error: missing '(' in expression
  #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 19, 0) || defined(RHEL8))
                                                                      ^
/root/wifi_drivers/8812au-20210629/os_dep/linux/os_intfs.c:1865:22: warning: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, void *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, void *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Wincompatible-pointer-types]
  .ndo_select_queue = rtw_select_queue,
                      ^~~~~~~~~~~~~~~~
/root/wifi_drivers/8812au-20210629/os_dep/linux/os_intfs.c:1865:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
make[2]: *** [scripts/Makefile.build:316: /root/wifi_drivers/8812au-20210629/os_dep/linux/os_intfs.o] Error 1
make[1]: *** [Makefile:1571: _module_/root/wifi_drivers/8812au-20210629] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-348.7.1.el8_5.x86_64'
make: *** [Makefile:2513: modules] Error 2

Result 3
-----------------
# make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.18.0-348.7.1.el8_5.x86_64/build M=/root/wifi_drivers/8812au-20210629  modules
make[1]: Entering directory '/usr/src/kernels/4.18.0-348.7.1.el8_5.x86_64'
  CC [M]  /root/wifi_drivers/8812au-20210629/os_dep/linux/os_intfs.o
  CC [M]  /root/wifi_drivers/8812au-20210629/os_dep/linux/usb_intf.o
  CC [M]  /root/wifi_drivers/8812au-20210629/os_dep/linux/usb_ops_linux.o
  CC [M]  /root/wifi_drivers/8812au-20210629/os_dep/linux/ioctl_linux.o
  CC [M]  /root/wifi_drivers/8812au-20210629/os_dep/linux/xmit_linux.o
  CC [M]  /root/wifi_drivers/8812au-20210629/os_dep/linux/mlme_linux.o
  CC [M]  /root/wifi_drivers/8812au-20210629/os_dep/linux/recv_linux.o
  CC [M]  /root/wifi_drivers/8812au-20210629/os_dep/linux/ioctl_cfg80211.o
/root/wifi_drivers/8812au-20210629/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
/root/wifi_drivers/8812au-20210629/os_dep/linux/ioctl_cfg80211.c:469:3: error: too few arguments to function ‘cfg80211_ch_switch_started_notify’
   cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /root/wifi_drivers/8812au-20210629/include/osdep_service_linux.h:98,
                 from /root/wifi_drivers/8812au-20210629/include/osdep_service.h:54,
                 from /root/wifi_drivers/8812au-20210629/include/drv_types.h:27,
                 from /root/wifi_drivers/8812au-20210629/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:7673:6: note: declared here
 void cfg80211_ch_switch_started_notify(struct net_device *dev,
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:316: /root/wifi_drivers/8812au-20210629/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:1571: _module_/root/wifi_drivers/8812au-20210629] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-348.7.1.el8_5.x86_64'
make: *** [Makefile:2513: modules] Error 2

imuji2022, 2022-01-05:
I got it to build. I modified ioctl_cfg80211.c file and added missing argument. I ran ./install-driver.sh, skipped option editing  and rebooted when prompted.

However, after the reboot "wifi adapter not found" error is displayed in the Settings->Wifi

Any suggestions / recommendations or reading that might help.

Thank you,

morrownr, 2022-01-05:
Hi @imuji2022 

Okay. Well, just because the compiler likes it does not mean it will work. This is my bad.

I have now reverted the patches. I should have created a branch to do this work but I didn't and that was a mistake on my part. At least you can see basically what needs to be done. Yes, I made a syntax error or two which contributed to the problems we are seeing. Fundamentally, I should have started by waiting until I have time to install RHEL8. Would you be so kind as to point me to where I can download a copy of RHEL8 that I can install so that I can test?

Regards

imuji2022, 2022-01-05:
Hello Nick,
I signed up for redhat developer program
https://developers.redhat.com/articles/getting-red-hat-developer-subscription-what-rhel-users-need-know

This allowed me to login to portal and download iso.

Here is update on my current progress:

Running install-driver.sh version 20211212
Starting installation...
Copying source files to: /usr/src/rtl8812au-5.13.6
Copying 8812au.conf to: /etc/modprobe.d
Creating symlink /var/lib/dkms/rtl8812au/5.13.6/source -> /usr/src/rtl8812au-5.13.6

Building module:
cleaning build area...
'make' -j8 KVER=4.18.0-348.7.1.el8_5.x86_64 KSRC=/lib/modules/4.18.0-348.7.1.el8_5.x86_64/build................
cleaning build area...

8812au.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.18.0-348.7.1.el8_5.x86_64/extra/
Adding any weak-modules
depmod: ERROR: fstatat(4, 8812au.ko.xz): No such file or directory
depmod....
The driver was installed successfully.
Do you want to edit the driver options file now? [y/N] n
Do you want to reboot now? (recommended) [y/N] n

Thank you,

morrownr, 2022-01-05:
Here is what I suggest for now. Go to:

https://github.com/morrownr/88x2bu-20210702

Go to `Pull Requests"

You will see one closed Pull Request. Open and see if you can start a conversation with the person that submitted the PR. My schedule is such that I may not have any to look at this and since I won't have an iso to install, I need to ponder things.

imuji2022, 2022-01-05:
Hello Nick,

I found your post regarding WiFi Adapters at https://github.com/morrownr/USB-WiFi 
Great info. I wish I had found it earlier. 

For now I will review https://github.com/morrownr/88x2bu-20210702 and keep searching...

Thank you for your time and efforts.

morrownr, 2022-01-06:
Hi @imuji2022 

> I found your post regarding WiFi Adapters at https://github.com/morrownr/USB-WiFi Great info. I wish I had found it earlier.

Well, my advertising budget is low... actually it is zero so people have to find it. It is getting 1,250 hits per week and growing these days so people are finding it.
 
> For now I will review https://github.com/morrownr/88x2bu-20210702 and keep searching...
> 
> Thank you for your time and efforts.

You are welcome. This is something that can be fixed, it may just take a while. I likely got close but there are significant differences between the driver that I was using as an example and this driver. 

Regards

morrownr, 2022-01-06:
If budget is an issue, take a look at this:

https://www.amazon.com/gp/product/B086L3D3NB

I'm not sure where you are located but that specific adapter is actually a really good adapter that is single-state and would work without you having to muss and fuss with Realtek drivers. Not many left in stock from this retailer so I would not contemplate long.

imuji2022, 2022-01-15:
Currently, unavailable. But I will keep an eye out. Thank you.

MidlifeRenaissanceMan, 2022-01-30:
Hey Nick
Just letting you know that your repository is showing up top on Duck Duck Go looking for RTL USB dongles. Shame I forgot exactly what I had typed.

FYI, Rocky Linux is the downstream from RHEL replacing the old CentOS that Big Blue Hat canned on us all. 

Grab yourself a copy of that if you need. 

I managed to get the 8812BU working with a small change to a compiler directive. 

Thanks for your hard work :-)

closed 2023-03-17 by morrownr