morrownr/8812au-20210629 #29: Module compilation on kernel 5.16 en Debian 11 breaks

JHM-O, 2022-01-20:
Hi,

Below my experience with a first compilation run of this driver on Debian 11 and a 5.16 kernel:

DKMS make.log for rtl8812au-5.13.6 for kernel 5.16.0 (x86_64)
Thu 13 Jan 2022 17:47:44 CET
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.16.0/build M=/var/lib/dkms/rtl8812au/5.13.6/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.16.0'
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_vht.o
/var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.c: In function ‘rtw_txpwr_init_regd’:
/var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.c:304:20: error: this statement may fall through [-Werror=implicit-fallthrough=]
  304 |                 if (rfctl->txpwr_lmt_name)
      |                    ^
/var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.c:307:9: note: here
  307 |         default:
      |         ^~~~~~~
/var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.c:1695:38: error: this statement may fall through [-Werror=implicit-fallthrough=]
 1695 |                         ptable->func = &OnAuthClient;
      |                         ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.c:1697:9: note: here
 1697 |         case WIFI_ASSOCREQ:
      |         ^~~~
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/monitor/rtw_radiotap.o
cc1: all warnings being treated as errors
make[2]: *** [scripts/Makefile.build:287: /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1846: /var/lib/dkms/rtl8812au/5.13.6/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.16.0'
make: *** [Makefile:2512: modules] Error 2

And also:

sudo uname -a; mokutil --sb-state; lsusb; /usr/sbin/rfkill list all; /usr/sbin/dkms status; /sbin/iw dev
[sudo] password for jouthuis: 
Linux Groargh 5.16.0 #1 SMP PREEMPT Thu Jan 13 16:49:50 CET 2022 x86_64 GNU/Linux
EFI variables are not supported on this system
Bus 006 Device 004: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 006 Device 003: ID 04a9:190a Canon, Inc. CanoScan LiDE 210
Bus 006 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 006 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 7392:a822 Edimax Technology Co., Ltd 802.11n NIC
Bus 003 Device 002: ID 145f:0203 Trust Trust USB Camera
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 0b33:0402 Contour Design, Inc. RollerMouse Free 2
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
rtl8812au/5.13.6: added
rtl8812au/5.9.3.2, 5.10.0-3-amd64, x86_64: installed
rtl8812au/5.9.3.2, 5.13.0, x86_64: installed
rtl8812au/5.9.3.2, 5.14.0, x86_64: installed

A fix would be welcome. I will give that a test asap.

Thanks

Jan Outhuis

morrownr, 2022-01-20:
Hi Jan

> A fix would be welcome.

Thanks for posting the information. I suspect the fix will have to happen on your end. Let me suggest some things:

> rtl8812au/5.13.6: added
> rtl8812au/5.9.3.2, 5.10.0-3-amd64, x86_64: installed
> rtl8812au/5.9.3.2, 5.13.0, x86_64: installed
> rtl8812au/5.9.3.2, 5.14.0, x86_64: installed

The last 3 lines show that you already have v5.9.3.2 of a driver for this chipset installed. That driver needs to be removed. Please follow the removal instructions for that driver and reboot.

The top line shows that the failed installation did add the driver to dkms. That needs to be removed. Please run the following per the removal instructions in the README:

./remove-driver.sh

Once those are complete, follow the installation instructions very carefully and let's see where we are.

My opinion: This is a very heavily used repo (hundreds of installations per day). I do my best to keep the driver maintained. If there were a lot of people having problems, we would see `Issues` full of reports. However, there are very few reports. I know it is fun to upgrade kernels and various parts of the os, I do it myself but I also know very well that some things may not work if I do so. If you are running Debian 11 with kernel 5.16, that tells me that your os is what I call a Frankenstein OS. It is put together with various parts that were not part of the original distro. The problem with this approach is that the various parts have not been tested together and issues may happen. I keep multiple distros available for testing but they are stock setups. I can best fix things if I can duplicate the problem. With Frankenstein OS's, I have no practical way to duplicate the setup. We can try to fix this but it may not work.

Regards

JHM-O, 2022-01-25:
I did a recompile against kernel 5.14. This went well, no break. Module works correct on install. 

Below the make.log. Same sort of notification as above for kernel 5.16 in the first couple of lines. This time however it is labeled a 'warning', whereas above for 5.16 it is labeled an 'error'

Is there anything I can do about that?

DKMS make.log for rtl8812au-5.13.6 for kernel 5.14.0 (x86_64)
Tue 25 Jan 2022 12:14:49 CET
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.14.0/build M=/var/lib/dkms/rtl8812au/5.13.6/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.14.0'
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_wlan_util.o
/var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.c: In function ‘rtw_txpwr_init_regd’:
/var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.c:304:20**: warning:** this statement may fall through [-Wimplicit-fallthrough=]
  304 |                 if (rfctl->txpwr_lmt_name)
      |                    ^
/var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.c:307:9: note: here
  307 |         default:
      |         ^~~~~~~
/var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.c:1695:38: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1695 |                         ptable->func = &OnAuthClient;
      |                         ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.c:1697:9: note: here
 1697 |         case WIFI_ASSOCREQ:
      |         ^~~~
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/monitor/rtw_radiotap.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/wds/rtw_wds.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ft.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_wnm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mbo.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rm_util.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_roch.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-internal.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-internal-enc.o
/var/lib/dkms/rtl8812au/5.13.6/build/core/efuse/rtw_efuse.c: In function ‘rtw_efuse_analyze’:
/var/lib/dkms/rtl8812au/5.13.6/build/core/efuse/rtw_efuse.c:971:17: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
[... 113 lines cut ...]
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/phydm_pow_train.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/phydm_lna_sat.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/phydm_pmac_tx_setting.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/phydm_mp.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/phydm_cck_rx_pathdiv.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/halrf/halrf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/halrf/halrf_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/halrf/halphyrf_ce.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/halrf/halrf_powertracking_ce.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/halrf/halrf_powertracking.o
[... cut ...]
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/rtl8812a/phydm_rtl8812a.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/txbf/haltxbfjaguar.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/platform/platform_ops.o
  LD [M]  /var/lib/dkms/rtl8812au/5.13.6/build/8812au.o
  MODPOST /var/lib/dkms/rtl8812au/5.13.6/build/Module.symvers
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/8812au.mod.o
  LD [M]  /var/lib/dkms/rtl8812au/5.13.6/build/8812au.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.14.0'

morrownr, 2022-01-25:
Hi @JHM-O 

Fix merged. How about you do a removal of the current version that you have:

sudo ./remove-driver.sh

Then you can either delete the directory and start over or `git pull` should work... then `sudo ./install-driver.sh`

It appears you were able to get rid of the lingering previously installed drivers.

Let me know how this update works.

Regards

JHM-O, 2022-01-27:
Hi Nick,

Your fixes have solved the problem. Clean compile on 5.16 now. Module
works.

Below tthe make.log, which looks more than fine.

Thanks and kind greetings.

DKMS make.log for rtl8812au-5.13.6 for kernel 5.16.0 (x86_64)
Thu 27 Jan 2022 11:25:11 CET
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.16.0/build
M=/var/lib/dkms/rtl8812au/5.13.6/build modules
make[1]: Entering directory '/usr/src/linux-headers-5.16.0'
warning: the compiler differs from the one used to build the kernel
The kernel was built by: gcc (Debian 11.2.0-13) 11.2.0
You are using: gcc (Debian 11.2.0-14) 11.2.0
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_cmd.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_security.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_debug.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_io.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ioctl_query.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ioctl_set.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ieee80211.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mi.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_wlan_util.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_vht.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_pwrctrl.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rf.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_chplan.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/monitor/rtw_radiotap.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_recv.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_sta_mgt.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ap.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/wds/rtw_wds.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/mesh/rtw_mesh.o
CC [M]
/var/lib/dkms/rtl8812au/5.13.6/build/core/mesh/rtw_mesh_pathtbl.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/mesh/rtw_mesh_hwmp.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_xmit.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_p2p.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rson.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_tdls.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_br_ext.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_iol.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_sreset.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_btcoex_wifionly.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_btcoex.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_beamforming.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_odm.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rm.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rm_fsm.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ft.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_wnm.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mbo.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rm_util.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/efuse/rtw_efuse.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_roch.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-internal.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-internal-
enc.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-gcm.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-ccm.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-omac1.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/ccmp.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/gcmp.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-siv.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-ctr.o
[... 136 lines cut ...]
/var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/rtl8812a/halhwimg8812a_m
ac.o
CC [M]
/var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/rtl8812a/halhwimg8812a_b
b.o
CC [M]
/var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/rtl8812a/halhwimg8812a_r
f.o
CC [M]
/var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/halrf/rtl8812a/halrf_881
2a_ce.o
CC [M]
/var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/rtl8812a/phydm_regconfig
8812a.o
CC [M]
/var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/rtl8812a/phydm_rtl8812a.
o
CC [M]
/var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/txbf/haltxbfjaguar.o
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/platform/platform_ops.o
LD [M] /var/lib/dkms/rtl8812au/5.13.6/build/8812au.o
MODPOST /var/lib/dkms/rtl8812au/5.13.6/build/Module.symvers
CC [M] /var/lib/dkms/rtl8812au/5.13.6/build/8812au.mod.o
LD [M] /var/lib/dkms/rtl8812au/5.13.6/build/8812au.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.16.0'

morrownr, 2022-01-27:
Hi @JHM-O 

> Your fixes have solved the problem. Clean compile on 5.16 now. Module works.

That is good to hear. Thanks for the report. 

Nick