morrownr/8812au-20210629 #31: it's work? on arm64 kali ??

lastime1650, 2022-01-28:
I install and use arm64-based Kalilinux on Raspberry Pie. I'm very detained to see if the rtl8812au driver can be installed successfully.

morrownr, 2022-01-28:
I haven't tested it on arm64 Kali yet but I am not aware of issues. If you decide to give it a try, I recommend you read the installations carefully and ask questions here if uncertain about anything before actually doing the installation.

For Step 9, I recommend:

./raspiOS-64.sh

Regards

lastime1650, 2022-01-28:
I tried it several times, but I failed.

lastime1650, 2022-01-28:
Kalilinux detects the wifi adapter through the lsusb command, but it doesn't work.

morrownr, 2022-01-28:
Can you post the output of:

$ lsusb

Can you provide me a link to arm64 Kali and I see what is going on?

lastime1650, 2022-01-28:
현재 상황을 보고드립니다. 드라이버가 변경되었습니다. 윈도우 운영 체제에서 8812au 드라이버로 사용할 수 있었지만, 웬일인지 라즈베리 파이 암64 칼리 리눅스는 rtl88x2.bu으로 인식한다. 그래서 당신의 github에서 rtl88x2.bu 드라이버를 찾았고 wifi 어댑터를 Kali 리눅스에 연결했더니 모든 것이 멈췄고 Kalinux의 데스크톱은 망가졌습니다.

lastime1650, 2022-01-28:
I'm reporting the current situation. The driver has changed. It was available as an 8812au driver on Windows operating systems, but somehow raspberry pi arm64 kali linux recognizes it as rtl88x2.bu. So I found rtl88x2.bu driver in your github. and I connected wifi adapter to Kali linux, everything stopped, and Kalinux's desktop was ruined.(broken)

lastime1650, 2022-01-28:
When I connect the Wi-Fi adapter, everything freezes and I have to turn it off and on forcibly. This is a big problem. Will it be solved only when the driver is reinstalled again?

morrownr, 2022-01-28:
> When I connect the Wi-Fi adapter, everything freezes and I have to turn it off and on forcibly.

Then don't connect the WiFi adapter.

Let me see if I can get your attention for a moment:

There is a saying in my country that goes like this: If you find that you are digging yourself a hole, stop digging.

I need you to stop digging and start listening. Please provide the output of the following commands:

$ dkms status
$ lsmod

Thank you.

lastime1650, 2022-01-28:
rtl88x2bu/5.13.1, 5.4.83-Re4son-V8l+, aarch64: installed

lsusb: ID 0bda:B812 Realtek Semiconductor Corp. RTL88x2bu (AC1200 Techkey)

morrownr, 2022-01-28:
I want the entire output of those commands, not the ones that you cherry pick.

morrownr, 2022-01-31:
Closed due to lack of response.

closed 2022-01-31 by morrownr

LorenzoAncora, 2022-01-31:
> When I connect the Wi-Fi adapter, everything freezes and I have to turn it off and on forcibly.
>> Then don't connect the WiFi adapter.
>> **Let me see if I can get your attention for a moment**: There is a saying in my country that goes like this: **If you find that you are digging yourself a hole, stop digging**.  **I need you to stop digging and start listening.** I want the entire output of those commands, **not the ones that you cherry pick**.
>> **Closed due to lack of response.**

@morrownr I've just browsed by and this discussion looks a bit hustled to me: if you are so eager to mark the Issues as closed, I suggest that a more polite approach could lead to quicker answers.

morrownr, 2022-01-31:
Hi @LorenzoAncora 

> I've just browsed by and this discussion looks a bit hustled to me: if you are so eager to mark the Issues as closed, I suggest that a more polite approach could lead to quicker answers.

I understand your perspective. Now please try to understand my perspective:

The third item listed on the main page (README.md) makes it clear what info should be provided with issues. That this guidance was not followed does not sit well with me. I then took my time, and was polite, over a series of messages where I requested information. Little or no additional information that was useful was provided.

The following link shows the repos that I maintain for the benefit of Linux users:

https://github.com/morrownr?tab=repositories

I am not paid anything. I do this for free. I generally enjoy it but when a user reports an issue but fails to assist in the resolution of the problem, I can only assume their report is for their entertainment and is not serious. At that point, it is time to close the issue.

If the OP wants to start over by making a new issue that includes the needed information. That is fine. 

Regards

LorenzoAncora, 2022-01-31:
Hi @morrownr

> The third item listed on the main page **(README.md) makes it clear what info should be provided with issues**.

Your README.md is 556 lines long and extremely complex for a non-native speaker, featuring complex nesting, blocks of code and many outbound links, all stuff that should be moved to a [GitHub Wiki](https://docs.github.com/en/communities/documenting-your-project-with-wikis/about-wikis), so to ensure proper accessibility. The README.md is definitely the wrong place to write bug reporting instructions, as you should create a standard [Issue Template](https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/about-issue-and-pull-request-templates) for that.
After reading that, please don't forget to make one, it helps a lot. 😉 

> I generally enjoy it but **when a user reports an issue but fails to assist in the resolution of the problem, I can only assume their report is for their entertainment and is not serious**. At that point, it is time to close the issue. [...] That **this guidance was not followed does not sit well with me**. I then took my time, and was polite, over a series of messages where I requested information. **Little or no additional information that was useful was provided**.

Unless they write you otherwise, end users lack development or system administration skills. When they fail to cooperate, first blame your explanations and try to repeat them with simpler wordings, explaining what the commands are supposed to do (because users need to trust you before executing anything, because the system may not be theirs or they may not be acquainted with it).

Really, try not to see the world in grey: unless you've offended someone before, by logic there is never a valid reason for thinking your end users are nagging you for their personal amusing. Believe in their good faith. :-)

> If the OP wants to start over by making a new issue that includes the needed information. That is fine.

The consequence of closing Issues too early is that you'll gradually reduce the quality of your code and documentation, because users are always more likely to participate to an existing open issue rather than opening a new one, especially if another one already exists and was ignored or the answer wasn't friendly enough to encourage further discussion.
Closing the Issues without solving the problem at their root will only make users diffident and you volunteer work harder.

> **I am not paid anything. I do this for free.**

In the same way your end users aren't paid anything to test your code, follow your instructions or give you insights on their system configuration. They employ their time to give you the possibility to fix the code for the good of the whole community, for free.

Be likeable to your end users and they'll to their best to reciprocate. 👋🏻