morrownr/8812au-20210629 #32: Humble request for channels 52-64 and/or 100-140

Bartholomew42, 2022-02-03:
Hello, I've recently got an USB device with a rtl8812au chip, ID 0bda:8812. It's working fine using this driver on a fresh Ubuntu 21.10 and for channels 36, 40, 44 and 48. But as the ISP-preconfigured wireless routers from three of my neighbours span 80MHz-networks over this frequency range, it's a bit crowded. 
Using an Ubuntu 20.04 with kernel 5.4, I was able to use Gordboys driver: https://github.com/gordboy/rtl8812au-5.9.3.2 
With his driver and the option rtw_country_code=DE (compare to https://github.com/gordboy/rtl8812au-5.9.3.2/issues/18 ), I can connect to my DFS-capable access point on the upper channels, tested with channels 52 and 100. So I know my hardware is able to do this. 
As Gordboy is not maintaining his repository anymore, I'd very much appreciate if you could find the time to get the upper channels usable with your driver, too. Maybe one could salvage some code that deals with the correct region setting from Gordboy's driver? 

(As I am not a native english speaker, I hope my text is understandable.)

Thank your for all your efforts so far!

morrownr, 2022-02-03:
Hi @Bartholomew42 

> (As I am not a native english speaker, I hope my text is understandable.)

Your English is very good.

> With his driver and the option rtw_country_code=DE (compare to gordboy/rtl8812au-5.9.3.2#18 ), I can connect to my DFS-capable access point on the upper channels

Try this from the driver directory:

```
sudo ./edit-options.sh
```

... and add `rtw_country_code=DE` to the `options` lines. Then reboot.

Now, with that said, this is a bug but it is not a bug in this driver. It is a bug in Ubuntu.

Go back out to the main README.md and go down to the section called `Check and set regulatory domain`. Follow the instructions. See if that fixes the problem. It may not as I have seen another solution but I don't remember what it was and I forgot to document it. However, if you run over to the Ubuntu `Networking and Wireless` forum and search or ask, I'll bet you get a good answer. Some smart folks over there. Please let me know what works.

Regards

P.S. My 8812au adapter is connected to channel 100 right now.  It will also do channel 100 in AP mode.

Bartholomew42, 2022-02-04:
Hello Nick, 
splendid, the option `rtw_country_code=DE` unlocked the missing channels. I'm now alone on channel 132+136 (40MHz) with -45dB signal strength, hooray. Thanks a lot for your quick and useful response! 

I did change the content of `/etc/default/crda` to `REGDOMAIN=DE` and verified it after a reboot with `$iw reg get` before I opened this issue. The driver does not seem to care about the thus set region of my system. 

```
$ iw reg get
global
country DE: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)
```

I did some testing: My system happens to be equipped with a 2,4GHz Singleband Intel Centrino Wireless-N 105 mini PCIe card (hence the urge to get a second interface with 5GHz capability). 
If I start the machine up with `/etc/default/crda` set to `REGDOMAIN=DE` , `$iw phy0 info` states that channel 1-13 are available with 15.0 dBm for the Centrino device, with the parameter `(no IR)` for channels 12+13. 
If I start with `REGDOMAIN=US`, `$iw phy0 info` states that now channel 1-11 are available with 15.0 dBm and channels 12+13 are disabled (as expected). 

`$iw phy1 info` for the rtl8812au device gives the same output (allowance for channels 1-13) with /etc/default/crda set to `REGDOMAIN=DE`, US, JP or 00 (checked after rebooting):
```
Wiphy phy1
...
	Band 1:
...
		Frequencies:
			* 2412 MHz [1] (17.0 dBm)
			* 2417 MHz [2] (17.0 dBm)
			* 2422 MHz [3] (17.0 dBm)
			* 2427 MHz [4] (17.0 dBm)
			* 2432 MHz [5] (17.0 dBm)
			* 2437 MHz [6] (17.0 dBm)
			* 2442 MHz [7] (17.0 dBm)
			* 2447 MHz [8] (17.0 dBm)
			* 2452 MHz [9] (17.0 dBm)
			* 2457 MHz [10] (17.0 dBm)
			* 2462 MHz [11] (17.0 dBm)
			* 2467 MHz [12] (17.0 dBm)
			* 2472 MHz [13] (17.0 dBm)
			* 2484 MHz [14] (disabled) 
...
```

Do I understand it right that this behaviour is unexpected, and that it is more a problem of Ubuntu not letting the driver know the current region setting, compared to the driver module not fetching the right setting? 
If I can help by trying things out on my system, let me know. 

As for now, maybe the `readme.md` should be amended to mention the `rtw_country_code:charp` option, and maybe also the default `8812au.conf `. I was smart enough to tinker with `rtw_dfs_region_domain:int` before opening this issue because that option was mentioned in your default `8812au.conf `, but not to try `$ modinfo 8812au |grep rtw_country_code ` to check if this driver would already accept the option that did previously help me with Gordboy's driver. 

By the way, if I use the option `rtw_dfs_region_domain=0` , `$iw phy1 info` delivers
```
		Frequencies:
			* 5180 MHz [36] (16.0 dBm)
			* 5200 MHz [40] (16.0 dBm)
			* 5220 MHz [44] (16.0 dBm)
			* 5240 MHz [48] (16.0 dBm)
			* 5260 MHz [52] (16.0 dBm) (radar detection)
			* 5280 MHz [56] (16.0 dBm) (radar detection)
			* 5300 MHz [60] (16.0 dBm) (radar detection)
			* 5320 MHz [64] (16.0 dBm) (radar detection)
			* 5500 MHz [100] (16.0 dBm) (radar detection)
			* 5520 MHz [104] (16.0 dBm) (radar detection)
			* 5540 MHz [108] (16.0 dBm) (radar detection)
			* 5560 MHz [112] (16.0 dBm) (radar detection)
			* 5580 MHz [116] (16.0 dBm) (radar detection)
			* 5600 MHz [120] (16.0 dBm) (radar detection)
			* 5620 MHz [124] (16.0 dBm) (radar detection)
			* 5640 MHz [128] (16.0 dBm) (radar detection)
			* 5660 MHz [132] (16.0 dBm) (radar detection)
			* 5680 MHz [136] (16.0 dBm) (radar detection)
			* 5700 MHz [140] (16.0 dBm) (radar detection)
			* 5720 MHz [144] (disabled)
			* 5745 MHz [149] (13.0 dBm)
			* 5765 MHz [153] (13.0 dBm)
			* 5785 MHz [157] (13.0 dBm)
			* 5805 MHz [161] (13.0 dBm)
			* 5825 MHz [165] (13.0 dBm)
			* 5845 MHz [169] (disabled)
			* 5865 MHz [173] (disabled)
			* 5885 MHz [177] (disabled)
```
			
If i use the option `rtw_dfs_region_domain=3` (for ETSI) , `$iw phy1 info` gives me  
```
		Frequencies:
			* 5180 MHz [36] (16.0 dBm)
			* 5200 MHz [40] (16.0 dBm)
			* 5220 MHz [44] (16.0 dBm)
			* 5240 MHz [48] (16.0 dBm)
			* 5260 MHz [52] (16.0 dBm)
			* 5280 MHz [56] (16.0 dBm)
			* 5300 MHz [60] (16.0 dBm)
			* 5320 MHz [64] (16.0 dBm)
			* 5500 MHz [100] (16.0 dBm)
			* 5520 MHz [104] (16.0 dBm)
			* 5540 MHz [108] (16.0 dBm)
			* 5560 MHz [112] (16.0 dBm)
			* 5580 MHz [116] (16.0 dBm)
			* 5600 MHz [120] (16.0 dBm)
			* 5620 MHz [124] (16.0 dBm)
			* 5640 MHz [128] (16.0 dBm)
			* 5660 MHz [132] (16.0 dBm)
			* 5680 MHz [136] (16.0 dBm)
			* 5700 MHz [140] (16.0 dBm)
			* 5720 MHz [144] (disabled)
			* 5745 MHz [149] (13.0 dBm)
			* 5765 MHz [153] (13.0 dBm)
			* 5785 MHz [157] (13.0 dBm)
			* 5805 MHz [161] (13.0 dBm)
			* 5825 MHz [165] (13.0 dBm)
			* 5845 MHz [169] (disabled)
			* 5865 MHz [173] (disabled)
			* 5885 MHz [177] (disabled)
```
			
Which one should I use? 

Regards!

morrownr, 2022-02-04:
> Which one should I use?

The `rtw_dfs_region_domain=` parameter is for AP mode. If not using AP mode, leave it at 0. If you are using DFS channels in AP mode, use the setting that is appropriate for your country.

Thanks for the ideas about documenting `rtw_country_code=`. I'll put that on my to-do list to evaluate. I had not documented it because using it in most situations, including yours, should not be needed.

Regards

morrownr, 2022-02-16:
Hi @Bartholomew42 

> Can you tell me if the Alfa AWUS036ACM works with DFS channels under Windows 10?

I can test this if you want. I did have a severe shortage of machines that run Windows until I got a new little laptop for XMas but it has Windows 11. Of couse I promply add an sdd and installed Linux to it but the original drive still has Windows. I'm a little sick these days will try to do the test soon.

I know the ALFA ACM does DFS in client mode as I use it like that regularly. I also know the code for AP mode DFS is in the 7612u driver and I am currently engaged with the Mediatek team in an effort to see if it can be turned on.

Regards

morrownr, 2022-02-17:
I just tested the ALFA ACM on my new laptop with Windows 11. The ACM was plug and play. It connected to my router that shows 5 GHz channel 60, which is DFS. I surfed for a while and then shut things down.

crmmc, 2022-03-23:
I got the Same Problem on my RaspberryPi 4B 1G Module, and new driver on X86 Kali Linux.  I'm using TP-Link WDR6200 adapter，And when i plugin it into my Kali Linux and using driver from aircrack-ng/rtl8812au branch v5.6.4.2 , it can work well on monitor mode and it can using all bands including Band 1-14 36-165, but when i using every driver i can found on the net built on my RPI4B ,only band 1-13 149-161 can be used, i tried ( /morrownr/8812au-20210629/ ), ( /aircrack-ng/rtl8812au branch v6.4 / 6.13) and some custom driver, it telled me lost of bands were disabled. i'm very confuse about it
i try to change the source code on os_dep/linux/wifi_regd.c to confirm the country setting, but failed.It seems like Driver doesn't accept my costom country setting, i had tryed to write a file with "rtw_dfs_region_domain=0 rtw_set_country=US" it doesn't work.
[Picture For My Adapter]
[image]
[It can use all channel on driver v6.4 On X86 Kali Linux]
[image]

when i use newest driver on X86 Kali Linux or any driver on Armhf RPI OS on my RPI4B1G it doesn't work on bands 36-145 165
[image]

i hope you can help me, thanks

What bothers me most is that there is no problem with running a specific version of the driver on Kali Linux. You can start AP mode，However, this version（aircrack-ng v5.6 version) will have restrictions on my raspberryPi,  not only it disable lots of bands, but it will cause kernel panic when listening mode is turned on too.