morrownr/8812au-20210629 #34: bug report: failure with Ubuntu 20 on HP Z230

keithbriggs, 2022-02-04:
I've got an HP Z230 desktop which ran an earlier version of this driver under Ubuntu 18 without problems.   I upgraded to Ubuntu 20, and now I cannot get the latest driver version to work.   I even bought a new dongle, the AWUS036ACS was was claimed to be tested and compatible.  The transcript below seems to show that installation worked without problems.   I did "sudo bash ./install_driver.sh" and got a success message.   I include the other requested output below.   The device is showing up with `lsusb`. I would appreciate any suggestions as to what to try next.

```
Running install-driver.sh version 20220108
Ubuntu 20.04.3 LTS
5.4.0-97-generic
x86_64
Starting installation...
Copying source files to: /usr/src/rtl8812au-5.13.6
Copying 8812au.conf to: /etc/modprobe.d

Creating symlink /var/lib/dkms/rtl8812au/5.13.6/source ->
                 /usr/src/rtl8812au-5.13.6

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j8 KVER=5.4.0-97-generic KSRC=/lib/modules/5.4.0-97-generic/build...........
Signing module:
 - /var/lib/dkms/rtl8812au/5.13.6/5.4.0-97-generic/x86_64/module/8812au.ko
Secure Boot not enabled on this system.
cleaning build area...

DKMS: build completed.

8812au.ko:
Running module version sanity check.

Good news! Module version v5.13.6-15-gc40b977e2.20210629 for 8812au.ko
exactly matches what is already found in kernel 5.4.0-97-generic.
DKMS will not replace this module.
You may override by specifying --force.

DKMS: install completed.
The driver was installed successfully.
Do you want to edit the driver options file now? [y/N] y
Do you want to reboot now? (recommended) [y/N] N

Z230:~/Downloads/8812au-20210629-main> sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev
Linux home-z230 5.4.0-97-generic #110-Ubuntu SMP Thu Jan 13 18:22:13 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
SecureBoot disabled
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 0bda:0301 Realtek Semiconductor Corp. multicard reader
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 018: ID 0bda:0811 Realtek Semiconductor Corp. 802.11ac WLAN Adapter 
Bus 003 Device 003: ID 04f3:0210 Elan Microelectronics Corp. Optical Mouse
Bus 003 Device 002: ID 03f0:0024 HP, Inc KU-0316 Keyboard
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
rtl8812au, 5.13.6, 5.4.0-97-generic, x86_64: installed (WARNING! Diff between built and installed module!)

Z230:~/Downloads/8812au-20210629-main> iwconfig 
lo        no wireless extensions.
eno1      no wireless extensions.
```

morrownr, 2022-02-04:
Hi @keithbriggs 

> I even bought a new dongle, the AWUS036ACS was was claimed to be tested and compatible.

The AWUS036ACS uses the 8811au chipset. The proper driver for the 8811au chipset is:

https://github.com/morrownr/8821au-20210708

Regards

keithbriggs, 2022-02-05:
Thanks very much - it's working now!