morrownr/8812au-20210629 #38: how to wpa3 issue #17 not there

ABerTSC, 2022-02-18:
hi i still would like to activate wpa3 on my awus036ach i did the steps in the wpa_supplicant txt file but how do i active wpa3 then? the tutorial is not there anymore :S

morrownr, 2022-02-18:
FYI: I am trying to do some housecleaning. Look in the docs folder for an upgraded wpa_supplicant guide. That probably won't fix what you are running into though. You may be running into problems with Network Manager. On some distros you have to kill or stop NM and use wpa_supplicant to connect with WPA3. 

Before you say anything... I know you are thinking it because I am thinking it also... Yes, this is a mess.

While we are now enjoying a cup of coffee... did you know that I have not found an adapter that uses in-kernel drivers that does work painlessly with WPA3? 

Vent mode off. What distro are you using?

morrownr, 2022-02-19:
hi @ABerTSC 

I use Linux Mint as my main dev distro. I keep other distros on test systems but that is another story.

I have watched as big new changes have happened over the years and it takes time to get the bugs worked out... however, WAP3 is working well on Linux for many people. There are two issues at play:

1. Realtek does not make Linux standards compliant drivers for their USB Adapters. This causes problems in a variety of ways and one of the ways concerns WPA3 support. on the other hand, Mediatek based USB Adapters work well already as they are Linus standards compliant. I do maintain a repo that provides more info:

https://github.com/morrownr/USB-WiFi

2. Linux Mint is a nice distro but it only rebases on the LTS versions of Ubuntu. What that means is that near the end of the 2 year LTS cycle, which is where we are now, Linux Mint may not be supporting recent changes very well and that includes WPA3. I suspect things will be greatly improved with Mint 22.0 and it is only a few months away.

Regards

closed 2023-03-17 by morrownr