morrownr/8812au-20210629 #39: Add possibility of using secure boot

purevw, 2022-02-18:
I found that the kernel will not allow insertion of the drive on my system. openSUSE 15.3 on an x86_64 system.   Per the openSUSE forum, I found that the issue was caused because I used secure boot. Disabling secure boot fixed the issue, but I have Windows 11 on the same machine, which requires secure boot. Is there any chance of building an efi version or addition to the driver?

morrownr, 2022-02-18:
I just grabbed my laptop and my 8812au based adapter. A quick removal of the driver and reboot later, I followed the installation steps from the beginning. Install went smooth and a quick reboot later, I was logging in. Iperf3 is showing about 400 Mbps and that it from one side of the house to the other.

I checked to ensure secure boot was enabled and it is.

I am not that familiar with openSUSE. The only thing I can think of right now would be if a basic manual installation is what is being used. I use dkms with the install script and one of the reasons is because it handles keeping secure boot happy.

morrownr, 2022-02-21:
See #35

purevw, 2022-03-14:
> Status?

I apologize for not replying, I thought I had. The openSUSE builder (Sauerland) has updated drivers that I think are based on Git drivers. He also provides an 8812AU_8821AU-ueficert, which addresses the issue.

closed 2022-03-14 by purevw

morrownr, 2022-03-14:
Hi @purevw 

The name of the driver that you posted, rtl8812AU_8821AU-kmp-default, suggests to me that it is based on very old source code. Support for both the 8812au and 8821au chipsets in the same driver was phased out several years ago. The driver here is only for the 8812au and is recent and has many more features than the older drivers, not to mention that it is rock solid.

I try to support a wide range of distros, but many distros make it difficult. My opinion of what would best serve users is that Sauerland and I cut a deal. He shows me what is required for my repo to work with openSUSE and I push this driver with its newer source to him so that he can have a better rtl8812AU-kmp-default. If he does not like the idea of a driver that only supports the 8812au, tell him I also have a very modern version of the 8821au driver:

https://github.com/morrownr/8821au-20210708

Regards

purevw, 2022-03-14:
I'll send him a message but of course I can't speak for him.