morrownr/8812au-20210629 #40: hostapd not editable

RunningTarg3t, 2022-02-28:
Hello, 

I don´t know if this issue really depends on the driver, but every time I edit my hostapd, save the file and reboot - my entries (channel, ssid...) are gone and replaced with the default ones. 

Is there a solution or what I´m doing wrong?

Thank you.

morrownr, 2022-02-28:
Hi @RunningTarg3t 

I assume you are talking about hostapd.conf. Show me the full command line that you are using to start the editor.

Regards

RunningTarg3t, 2022-03-01:
Hi, thanks for this quick answer. 

I'm using:
"sudo nano /etc/hostapd/hostapd.conf"

RunningTarg3t, 2022-03-01:
Just now, I tried to edit the configuration again and added the line "driver=nl80211". Now it seems to work on another channel with another SSID. I dont know if that was the solution (without this line, the ap-mode worked too), but I'm happy. 

Big THX for the quick answer again. Have a good day!

RunningTarg3t, 2022-03-01:
Forget my last comment - it's the same issue. When I reboot my Raspi, the content of hostapd.conf is reset to default and I need help again o_0

morrownr, 2022-03-01:
> I'm using:  "sudo nano /etc/hostapd/hostapd.conf"

My best guess is that you are not saving the file after making changes. The following nano key combination will save the file:

Ctrl + o

Then `Ctrl + x` will exit.

RunningTarg3t, 2022-03-02:
Hehe, you're right, I'm a real noob 😁, but I saved my changes everytime. Sometimes it worked and after a reboot are my edit lines already there, but sometimes the hostapd is the default one. 

Perhaps it is my distro- named "max2play" based on buster. The web interface shows own AP- options and I think it overwrites my changes on a special point which I didn't found yet.

Bartholomew42, 2022-03-08:
Hi @[RunningTarg3t](https://github.com/RunningTarg3t) ,
> The web interface shows own AP- options and I think it overwrites my changes on a special point which I didn't found yet.

The following command gives you the _change time_ of your hostapd.conf: 
`ls -l --time-style=full-iso /etc/hostapd/hostapd.conf`
You may need to add a sudo as a prefix. 

Then you will at least know when the file was changed the last time, either by you or by your operating system. You can investigate further from there. 

Regards!

morrownr, 2022-03-08:
> Perhaps it is my distro- named "max2play" based on buster. The web interface shows own AP- options and I think it overwrites my changes on a special point which I didn't found yet.

After pondering the issue, I think the above is the key to this issue. I have no knowledge of `max2play` but if it already has AP support built in, your best bet is probably to see if that distro has forums where you can ask questions.

If you were running Raspberry Pi OS, I have a good guide for AP mode Access Point setup in the `docs` folder.

morrownr, 2022-03-19:
Status?

RunningTarg3t, 2022-03-20:
Sorry, for my absence, I have actual no time left to test further, but the informations I got are sufficient for me. Thank you.

closed 2022-03-20 by RunningTarg3t