morrownr/8812au-20210629 #41: Connection activation failed: (53) The Wi-Fi network could not be found

tplusone, 2022-03-10:
After following installation instructions, I successfully installed the driver for an _AWUS036AC_ adapter. However, I cannot connect to a Wi-Fi network no matter what network I choose (all of them exist). I receive an _Wi-Fi network could not be found_ issue. I'm working on HP/Kali VirtualBox installed on Ubuntu.

Output of **_sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev_**:

```
Linux kali 5.16.0-kali3-amd64 #1 SMP PREEMPT Debian 5.16.11-1kali1 (2022-03-03) x86_64 GNU/Linux
EFI variables are not supported on this system  
Bus 001 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub  
Bus 002 Device 002: ID 80ee:0021 VirtualBox USB Tablet  
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub  
0: phy0: Wireless LAN  
        Soft blocked: no  
        Hard blocked: no  
rtl8812au/5.13.6, 5.16.0-kali3-amd64, x86_64: installed  
phy#0  
        Interface wlan0  
                ifindex 3  
                wdev 0x1  
                addr 7e:3d:90:81:af:a6  
                type managed  
                txpower -100.00 dBm  
```

Result of **_iwconfig wlan0_**  is:
```
wlan0     unassociated  Nickname:"WIFI@RTL8812AU"
          Mode:Managed  Frequency=2.462 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

```

morrownr, 2022-03-10:
In the FAQ, there is a link to an article that might help.

morrownr, 2022-03-19:
Status?

tplusone, 2022-03-20:
thanks, it works. Turned out I had USB 2 activated in virtualbox. When I switched to USB 3 it connected successfully.

closed 2022-03-20 by tplusone