morrownr/8812au-20210629 #44: Compiled but does not show up in ip a

techflashYT, 2022-03-19:
It compiled and loaded but does not show up in `ip a`  
The output if that command at the top of the README is:
```
root@Optiplex-380:~# sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev
Linux Optiplex-380 5.10.0-10-amd64 #1 SMP Debian 5.10.84-1 (2021-12-08) x86_64 GNU/Linux
EFI variables are not supported on this system
Bus 005 Device 005: ID 0bda:0811 Realtek Semiconductor Corp. Realtek 8812AU/8821AU 802.11ac WLAN Adapter [USB Wireless Dual-Band Adapter 2.4/5Ghz]
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 005: ID 093a:2532 Pixart Imaging, Inc. Gaming Mouse
Bus 003 Device 004: ID 1a2c:417c China Resource Semico Co., Ltd USB Keyboard
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
rtl8812au/5.13.6, 5.10.0-10-amd64, x86_64: installed
```
And the output of an `ip a` is:
root@Optiplex-380:~# ip a
```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: enp2s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether b8:ac:6f:35:bd:98 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.25/24 brd 192.168.0.255 scope global enp2s0
       valid_lft forever preferred_lft forever
    inet6 fe80::baac:6fff:fe35:bd98/64 scope link
       valid_lft forever preferred_lft forever
```
Not sure what this is about, but this seems like it could be an issue, considering the `UNCLAIMED`
```
root@Optiplex-380:~# lshw
...
*-usb UNCLAIMED
                   description: Generic USB device
                   product: 802.11ac WLAN Adapter
                   vendor: Realtek
                   physical id: 7
                   bus info: usb@5:7
                   version: 2.00
                   serial: 00e04c000001
                   capabilities: usb-2.10
                   configuration: maxpower=500mA speed=480Mbit/s
```
Also a couple of things that I thought might be useful:
```
root@Optiplex-380:~# lspci
00:00.0 Host bridge: Intel Corporation 4 Series Chipset DRAM Controller (rev 03)
00:01.0 PCI bridge: Intel Corporation 4 Series Chipset PCI Express Root Port (rev 03)
00:1b.0 Audio device: Intel Corporation NM10/ICH7 Family High Definition Audio Controller (rev 01)
00:1c.0 PCI bridge: Intel Corporation NM10/ICH7 Family PCI Express Port 1 (rev 01)
00:1d.0 USB controller: Intel Corporation NM10/ICH7 Family USB UHCI Controller #1 (rev 01)
00:1d.1 USB controller: Intel Corporation NM10/ICH7 Family USB UHCI Controller #2 (rev 01)
00:1d.2 USB controller: Intel Corporation NM10/ICH7 Family USB UHCI Controller #3 (rev 01)
00:1d.3 USB controller: Intel Corporation NM10/ICH7 Family USB UHCI Controller #4 (rev 01)
00:1d.7 USB controller: Intel Corporation NM10/ICH7 Family USB2 EHCI Controller (rev 01)
00:1e.0 PCI bridge: Intel Corporation 82801 PCI Bridge (rev e1)
00:1f.0 ISA bridge: Intel Corporation 82801GB/GR (ICH7 Family) LPC Interface Bridge (rev 01)
00:1f.1 IDE interface: Intel Corporation 82801G (ICH7 Family) IDE Controller (rev 01)
00:1f.2 IDE interface: Intel Corporation NM10/ICH7 Family SATA Controller [IDE mode] (rev 01)
00:1f.3 SMBus: Intel Corporation NM10/ICH7 Family SMBus Controller (rev 01)
01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] RV740 PRO [Radeon HD 4770]
01:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] RV710/730 HDMI Audio [Radeon HD 4000 series]
02:00.0 Ethernet controller: Broadcom Inc. and subsidiaries NetLink BCM57780 Gigabit Ethernet PCIe (rev 01)
```
```
root@Optiplex-380:~# apt list --installed | grep "wlan"

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

```

```
root@Optiplex-380:~# apt list --installed | grep "ip"

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

bzip2/testing,unstable,now 1.0.8-5 amd64 [installed]
ghostscript/testing,unstable,now 9.55.0~dfsg-3 amd64 [installed,automatic]
gzip/stable,testing,unstable,now 1.10-4 amd64 [installed]
iproute2/testing,unstable,now 5.16.0-2 amd64 [installed]
iptables/stable,testing,unstable,now 1.8.7-1 amd64 [installed,automatic]
iputils-ping/testing,unstable,now 3:20211215-1 amd64 [installed]
libcairo-script-interpreter2/stable,testing,unstable,now 1.16.0-5 amd64 [installed,automatic]
libip4tc2/stable,testing,unstable,now 1.8.7-1 amd64 [installed]
libip6tc2/stable,testing,unstable,now 1.8.7-1 amd64 [installed,automatic]
libipc-system-simple-perl/stable,testing,unstable,now 1.30-1 all [installed,automatic]
libpipeline1/testing,unstable,now 1.5.5-1 amd64 [installed,automatic]
libpipewire-0.3-0/testing,unstable,now 0.3.48-1 amd64 [installed,automatic]
libpipewire-0.3-common/testing,unstable,now 0.3.48-1 all [installed,automatic]
libpipewire-0.3-modules/testing,unstable,now 0.3.48-1 amd64 [installed,automatic]
libqt5qmlworkerscript5/testing,unstable,now 5.15.2+dfsg-10 amd64 [installed,automatic]
pipewire-bin/testing,unstable,now 0.3.48-1 amd64 [installed,automatic]
pipewire-media-session/testing,unstable,now 0.4.1-2 amd64 [installed,automatic]
pipewire/testing,unstable,now 0.3.48-1 amd64 [installed,automatic]
python3-pyqt5.sip/testing,unstable,now 12.9.1-1 amd64 [installed,automatic]
python3-sip/testing,unstable,now 4.19.25+dfsg-3+b1 amd64 [installed,automatic]
qlipper/stable,testing,unstable,now 1:5.1.2-1+b1 amd64 [installed]
unzip/stable,testing,unstable,now 6.0-26 amd64 [installed]
whiptail/testing,unstable,now 0.52.21-5+b1 amd64 [installed]
zip/stable,testing,unstable,now 3.0-12 amd64 [installed,automatic]
```
Current GUI info (aka: I don't have a network flyout on my taskbar):
```
Display Manager: `sddm/stable,testing,unstable,now 0.19.0-3 amd64 [installed]`
Window Manager: `twm/stable,testing,unstable,now 1:1.0.10-1 amd64 [installed]`
```
PC Specs:
```
Dell Optiplex 380
Boots with LILO and systemd
64-bit
CPU: Intel Pentium Dual-Core E5400 @ 2.70GHz
RAM: 4GB of DDR3 (2 800MHz sticks, unsure if it's running in dual-channel)
GPU: AMD/ATI Radeon HD 4770
```
Please help,  I want to get this PC off of my desk where it is currently hooked up to my router, as that is the only way it can get internet connectivity currently.

morrownr, 2022-03-19:
Hi @techflashYT 

> ID 0bda:0811

The above device ID gives away the problem. If you look in the file `supported-device-IDs` here in this repo you won't find this device ID as a supported device for this driver.

If you go over to:

https://github.com/morrownr/8821au-20210708

...and look in `supported-device-IDs`, you will see your device ID as supported.

So, you have installed the wrong driver.

Recommend you go to the section `Removal of the Driver ( remove-driver.sh )` and follow the instructions to remove this 8812au driver and then proceed to install the 8821au driver.

Good luck and let me know how it goes.

Cheers

techflashYT, 2022-03-19:
Alright, thanks!  This is the repo that my friend sent me, and he's kinda a linux genius, so I didn't expect him to be wrong.

techflashYT, 2022-03-19:
Update:  I just compiled it and it's showing up in `ip a`!
```
3: wlx0013eff20449: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 00:13:ef:f2:04:49 brd ff:ff:ff:ff:ff:ff
```