morrownr/8812au-20210629 #46: Removal script error

freibadschwimmer, 2022-03-23:
Hi, thanks a lot for the driver!

I ran into some minor error when attempting to remove the driver. When running remove-driver.sh, the driver is being removed from dkms, but the files in /etc/modprobe.d/ and /usr/src/ remain on the system and have to be deleted manually.
The script throws the message:
remove-driver.sh: 51: Syntax error: word unexpected (expecting ")")

morrownr, 2022-03-24:
Hi @freibadschwimmer 

I tried unsuccessfully to duplicate the problem this morning. I also looked over the code in the script and don't see a problem.

That leaves us in a position where we need to dig deeper.

Did you edit the script at any point before running it?

What distro and hardware are you using?

Did you use `sudo` with `./remove-driver.sh`?

What directory did you save the driver in when you downloaded it?

I am pondering what could cause the problem but don't have any good ideas to pursue right now so we will just have to explore. I checked and this specific driver has been downloaded 1,260 times in the last 2 weeks so it sees a lot of action but I can't say how many folks are running the removal script.

Regards

morrownr, 2022-04-20:
Hi @freibadschwimmer 

I think I remember working on this issue and resolving it. Is it resolved?

I usually mark resolved issues as solved. Not sure what happened.

Regards

closed 2022-05-03 by morrownr