morrownr/8812au-20210629 #47: Installed but Not working on ubuntu 20.04 5.13.0-37

VM-137, 2022-03-24:
Hello, I have installed the driver with a clean and new installation of ubuntu, but it does not work, any suggestion?
thanks

Linux peter-B85M-D3H 5.13.0-37-generic #42~20.04.1-Ubuntu SMP Tue Mar 15 15:44:28 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
SecureBoot disabled
Platform is in Setup Mode
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 006: ID 18f8:0f97 [Maxxter] USB OPTICAL MOUSE 
Bus 003 Device 008: ID 148f:3070 Ralink Technology, Corp. RT2870/RT3070 Wireless Adapter
Bus 003 Device 005: ID 048d:1366 Integrated Technology Express, Inc. MultiCard Device
Bus 003 Device 004: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 003 Device 003: ID 2357:011e TP-Link 802.11ac WLAN Adapter 
Bus 003 Device 002: ID 0c45:7603 Microdia USB Keyboard
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: hci0: Bluetooth
	Soft blocked: yes
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
rtl8812au, 5.13.6, 5.13.0-37-generic, x86_64: installed
rtl88x2bu, 5.8.7.1, 5.13.0-37-generic, x86_64: installed
phy#0
	Interface wlx000db002de98
		ifindex 3
		wdev 0x1
		addr 00:0d:b0:02:de:98
		ssid NETGEAR37
		type managed
		channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
		txpower 20.00 dBm

morrownr, 2022-03-25:
Hi @VM-137 

> I have installed the driver with a clean and new installation of ubuntu, but it does not work, any suggestion?

Ummmm... you installed a driver for a chipset that you do not have. This driver is for usb wifi adapters based on the 8812au chipset. I see that you have two wifi adapters. One is based on the rt3070 chipset and is supported with in-kernel drivers. The other is based the rtl8811au chipset.

See issue #45 

Regards

VM-137, 2022-03-25:
Hello, I was searching and it uses or at least in 99% of the forums people says it works with rtl8812au drivers (Device 'TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano]), I have checked the  supported devices IDs before posting, so maybe it is wrong?
I have try https://github.com/aircrack-ng/rtl8812au and it work, can you give me an explanation of why it works if it is not the chipset rtl8812au? if you can sure.... just for learn. Thanks!

morrownr, 2022-03-25:
Hi @VM-137 

Glad to answer your questions.

> I have try https://github.com/aircrack-ng/rtl8812au and it work,

That driver is based on source code that is a few years old. At one time, the source code for the 8812au did also support the 8811au/8821au. That driver is maintained for a very specific purpose- supporting monitor mode work so moving to a modern source may be more work than the maintainers have time for.

> I was searching and it uses or at least in 99% of the forums people says it works with rtl8812au drivers...

One of the problems with the internet is that old, outdated information hangs around. There is also the problem of authors continuing to publish old information because they are too lazy to do any research.

One of the objectives I have for this site and all of the repos here is to provide the most current drivers and information. It takes a lot of work. What you have here in this repo is the most current 8812au driver that is available. It is a MUCH better driver than versions that are years old. This is a very stable driver with numerous capabilities that either did not work or did not exist in older versions.

Now, does that mean I recommend Linux users looking to buy new adapters run out any buy adapters based on Realtek chipsets? No. These Realtek drivers are based on the wrong technology and will be problematic and there is nothing I can do about that. I'm just doing the best I can to help users with Realtek based adapters while pointing out to them that there are alternatives:

https://github.com/morrownr/USB-WiFi

We are watching the driver for the new mt7921u chipset go into the kernel right now. Indications are that we will see usb wifi adapters based on the mt7921u become available in the second half of this year. That chipset supports WiFi 6. In the mean time, there are adapters available that use good in-kernel drivers that use the mt7612u and mt7610u chipsets and the above repo shows links to many adapters that use in-kernal drivers.

Hopefully this clears up a few things.

Regards