morrownr/8812au-20210629 #49: small error while compilig against 5.17.0

JHM-O, 2022-04-20:
Hello,

Sorry to bother again. Have waited some months before reporting. But the build still breaks with kernel 5.17. Seems to be small flaw, though.
So, below the log, hoping it will be a small effort this time to fix the error.

sudo uname -a; mokutil --sb-state; lsusb; /usr/sbin/rfkill list all; /usr/sbin/dkms status; /sbin/iw dev
Linux Groargh 5.17.0 #1 SMP PREEMPT Tue Mar 22 14:39:44 CET 2022 x86_64 GNU/Linux
EFI variables are not supported on this system
Bus 003 Device 004: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 003 Device 003: ID 04a9:190a Canon, Inc. CanoScan LiDE 210
Bus 003 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 003: ID 7392:a822 Edimax Technology Co., Ltd 802.11n NIC
Bus 005 Device 002: ID 145f:0203 Trust Trust USB Camera
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 0b33:0402 Contour Design, Inc. RollerMouse Free 2
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
rtl8812au/5.13.6: added

DKMS make.log for rtl8812au-5.13.6 for kernel 5.17.0 (x86_64)
Wed 20 Apr 2022 15:33:12 CEST
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.17.0/build M=/var/lib/dkms/rtl8812au/5.13.6/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.17.0'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Debian 11.2.0-18) 11.2.0
  You are using:           gcc (Debian 11.2.0-20) 11.2.0
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/monitor/rtw_radiotap.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/wds/rtw_wds.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ft.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_wnm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mbo.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rm_util.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_roch.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-internal.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-internal-enc.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-gcm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-ccm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-omac1.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/ccmp.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/gcmp.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-siv.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-ctr.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/sha256-internal.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/sha256.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/sha256-prf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/rtw_crypto_wrap.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_swcrypto.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/rtw_android.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/rtw_proc.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/nlrtw.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/rtw_rhashtable.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_intf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_com.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_com_phycfg.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_phy.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_dm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_dm_acs.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_btcoex.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_mp.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_mcc.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_hci/hal_usb.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/led/hal_led.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/led/hal_usb_led.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/HalPwrSeqCmd.o
/var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_hci/hal_usb.c: In function ‘usb_init_recv_priv’:
/var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_hci/hal_usb.c:29:22: error: cast between incompatible function types from ‘void (*)(void *)’ to ‘void (*)(long unsigned int)’ [-Werror=cast-function-type]
   29 |                      (void(*)(unsigned long))usb_recv_tasklet,
      |                      ^
cc1: all warnings being treated as errors
make[2]: *** [scripts/Makefile.build:288: /var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_hci/hal_usb.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1831: /var/lib/dkms/rtl8812au/5.13.6/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.17.0'
make: *** [Makefile:2514: modules] Error 2

morrownr, 2022-04-20:
Hi @JHM-O 

You are probably correct. This may be a minor error but I need to finish some unrelated work on this repo and then I will try to discover what the problem is. I see you are using a Debian based distro but it is not clear to me what distro it is. Can you tell me?

Regards

JHM-O, 2022-04-20:
It's basic Debian  Bookworm.Greetings,JhmVerzonden vanaf mijn Galaxy

morrownr, 2022-04-26:
Hi @JHM-O 

Message to let you know I have not forgot about this. I am looking at which test system to work this on.

Regards

morrownr, 2022-05-19:
Hi @JHM-O 

I merged a PR this morning that likely fixed this issue. Can I get you to test?

Regards,

Nick

JHM-O, 2022-05-19:
Hi Nick,That's nice. I'll look into this tomorrow.GreetsJhmVerzonden vanaf mijn Galaxy

JHM-O, 2022-05-20:
Hi Nick,

Ran a build against the fixed code. Gives a clean make-log and a
working module.

So, we're out of trouble again.

Great job. Thanks!

Greets
JHM

Finoderi, 2022-05-20:
I guess I had similar problem on Arch. The module refused to compile with 5.17 kernel. Although it did just that after the new PR as well. DKMS was confused. I had to launch remove-driver.sh script and then update the kernel.

Thank you for the fix.