morrownr/8812au-20210629 #51: (feedback) Some feedback on running this driver on Ubuntu 18.04, 20.04, 22.04

Bartholomew42, 2022-05-01:
TLDR: With a caveat, the driver can be considered tested and well-working with the current Ubuntu 22.04. 

The long version:

Hi Nick, 

I recently updated my desktop to Ubuntu 22.04 (Kernel 5.15). In this version, the crda package is no longer installed by default and got uninstalled with the update. Compare https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1958918 . It seems Gentoo did the same: https://bugs.gentoo.org/show_bug.cgi?id=462032 
So, `/etc/default/crda` is no longer read and the default region no longer set in this way. 

If I am running your driver with the `rtw_country_code=DE` option, it continues to work great, same as with 18.04 and 20.04 on my laptop. 

I observed the following:
I have a Zyxel router rented from my internet service provider that broadcasts an 802.11g-network on channel 13 and an 802.11n-network on channel 132+136. Also, I have temporarily set up a Netgear router (running openWRT) that broadcasts an 802.11n-network on channel 1 that I set up to use the region ruleset for the US. 

My rudimentary understanding of the set wireless region I can read out via `iw reg get `on my ubuntu clients is this:
When I am not connected to any wireless network at all, the region is 00 (or whatever I set before using `sudo iw reg set`). When I connect to my network on channel 13, it stays there. If I connect to the network on channel 1, it switches to US. If I disconnect the connection, it goes back to 00. If I connect it to the network on channel 132, it changes to DE. This works both with the 8812au USB adapter and with the build-in Centrino wireless-N 105. 
So, the basic idea in Ubuntu seems to be that an Ubuntu wireless client gets its region setting from the AP, if the latter is configured correctly (I can't get the network on channel 13 to broadcast the DE information). If the network connection is severed, the set region of the Ubuntu client goes back to the state it was in before. So, either 00 or anything that was set via `sudo iw reg set `(let it be CA, JP, ...). The output of `iwlist chan` and `iw list` changes accordingly with the output from `iw reg get`.

Now, the interesting bit (replicated on both laptop and desktop):
If I connect to the network on channel 1 (the one that gets the client to the US setting) using my trusty Alfa AWUS036ACM USB adapter (or the build-in interface, either Centrino wireless-n 105 or Realtek BCM4312 running with the b43 driver on ubuntu 18.04 kernel 5.4), I can no longer connect to the network on channel 13 with the other wifi adapter. This seems ok to me as the region on the client got set to US, thus inhibiting the wireless interfaces from sending radio signals on channel 12 and 13. If I connect to the network on channel 13 first, I can make a second connection with the other wifi adapter to the network on Channel 1. 

My USB adapter with the 8812 chipset will connect to the network on channel 13 anytime (if the driver was loaded with the `rtw_country_code=DE` option), regardless of the output of `iw reg get`. This is sensible. If I reload it without that option using `sudo modprobe -r 8812au & sudo modprobe 8812au` , the output from `iwlist chan` is

```
wlxdc4ef405e467  17 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
```

and it is indifferent to the output of `iw reg get`. With the three other wifi adapters, the output of `iwlist chan` and `iw list` changes with the output of `iw reg get.` 

My conclusion:
Your driver on my Ubuntu systems is indifferent to the wifi region set via `iw`. It will albeit and reliably use the region set that it receives via the option `rtw_country_code=__` when the driver gets loaded by the OS. 
With this caveat, the driver can be considered tested and well-working with the current Ubuntu 22.04. 

Let me know if I can test anything for you. 
Regards!

morrownr, 2022-05-02:
Hi @Bartholomew42 

Thank you for the detailed report. 

I have uploaded changes to address this issue today. If I could get you to test and provide feedback, I would appreciate it.

I do not currently have time to do a full investigation so what I did is a few modifications of the installation script and the .conf file so as to bring it to the attention of users that they need to set the country code. If you see anything else I can do that would make things better please let me know. It is important to get this right as I have 4 additional Realtek driver likely also need this change. 

Thank you

Bartholomew42, 2022-05-03:
Hi Nick, 
I just reinstalled your amended driver on Ubuntu 22.04. The automated invocation of nano /etc/modprobe.d/8812au.conf worked fine. 

About documentation: 
After the sencence` Important: Set rtw_country_code to the proper code for your country. See documentation section below.`, you might add something like `When you travel to another country, you might want to reset the country code with sudo ./edit_options.sh to conform to local regulations. `  as a friendly reminder. 

Also, the current README.md has a subsection called "Check and set regulatory domain". You might want to alter this section, too. 
In subsection "Driver Options ( edit-options.sh )" , you might change 
`Note: Documentation for Driver Options is included in the file 88x2bu.conf` to 
`Note: Documentation for Driver Options is included in the file 8812au.conf` .

Greetings!

morrownr, 2022-05-03:
Hi @Bartholomew42 

> I just reinstalled your amended driver on Ubuntu 22.04. The automated invocation of nano /etc/modprobe.d/8812au.conf worked fine.

Excellent. This is not the way I would have preferred to fix this issue but I do not have time to research what is behind it for now. If you run onto information, Please let me know.
 
> About documentation: After the sencence` Important: Set rtw_country_code to the proper code for your country. See documentation section below.`, you might add something like `When you travel to [...]

Done.
 
> Also, the current README.md has a subsection called "Check and set regulatory domain". You might want to alter this section, too.

For now I deleted section. When I have time I will figure out what is going on.

> In subsection "Driver Options ( edit-options.sh )" , you might change `Note: Documentation for Driver Options is included in the file 88x2bu.conf` to `Note: Documentation for Driver Options is included in the file 8812au.conf` .

Done.

Thanks.

Bartholomew42, 2022-05-04:
Hi Nick, 
I'm glad to be of help :)

In the changed 8812au.conf, you made a little mistake: 
```
To edit this file
# in the future:
#
# $ sudo nano ./edit_options.sh
```
The proposed command needs to be changed to `$ sudo ./edit_options.sh`. 

Greetings!

Bartholomew42, 2022-05-04:
Supplement: 
in README.md, you changed` 88x2bu.conf` to `88x2au.conf`, not to `8812au.conf`.

morrownr, 2022-05-04:
Your help is greatly appreciated.

> The proposed command needs to be changed to $ sudo ./edit_options.sh

Good catch. I messed this up. Actually it should be:

$ sudo ./edit-options.sh

I have edited this and even added another way to do it.

> 88x2au.conf

I have now fixed this as well. 

If stopping by to help is of interest to you, maybe once ever week or two, I could use some extra eyes. This repo and the following repo could certainly use some extra eyes:

https://github.com/morrownr/USB-WiFi

Thanks again and if you run across what is actually going on with this issue, please let me know because what we did here is really temp patch, not really a good permanent fix.

Regards

Nick

Bartholomew42, 2022-05-05:
Hi Nick, 
I did a proofread of the first section, see [https://github.com/morrownr/USB-WiFi/issues/74](https://github.com/morrownr/USB-WiFi/issues/74).

I do fully understand that giving the region as a static option to the driver is only a quick fix. I tried out the older aircrack-ng driver and I could not get it to work properly without loading the country code as an option, too. I hope that some information about what went wrong will pop up around the Ubuntu 22.04.1 release. 

Greetings!

morrownr, 2022-05-14:
Hi @Bartholomew42 

I've been sick so as behind.

Thanks for this information. It should really help.

I have been turning the README.md at USB-WiFi into a main menu. If you have time, take a look and offer any good ideas you have. It is a work in progress. The main thing that needs checking right now is if all of the links in each document still work.

Regards

Bartholomew42, 2022-05-23:
Hi Nick, 
I did a quick check of the new README.md at USB-WiFi, all links seem to be working as expected.

Concerning the README.md of this repository: In section "Recommended WiFi Router/ Access Point Settings", you recommend using Channels 1, 6 and 11. This is sound advice for most of North, Central and South America. For most of the rest of the world, channels 12 and 13 are legally usable and the use of channels 1, 5, 9 and 13 provides four overlap free 20 MHz channels instead of three. This in theory helps to decongest the 2.4Ghz band. 
Why in theory? Well, in the early years of 802.11, 
a) 802.11b with its 22MHz wide channels was still in use and 
b) it was advised not to use channel 12+13 in Europe to avoid unhappy clients that could not connect to these because their owner was unable to set the proper WiFi region in their printer or phone or whatever. 
So, nowadays we _could_ have four instead of three overlap free channels in Europe, and all that is stopping us now is the grown status quo with many APs still set to channels 1, 6 and 11. My personal advise to anybody who wants to set up a new 2.4Ghz WiFi network is to use channel 13. This sets an incentive for all the channel 11 users to move to channel 9, and those from channel 6 to move to channel 5. Well, in theory, as most users never touch their AP again as soon as it is set up for the first time. 
At least, for most of the bold channel 13 users, the upper half of the channel's bandwidth will be without background noise. 

My rant is over. Cheers!

Bartholomew42, 2022-05-23:
Supplement: 
In section https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapter_Information_for_Linux.md , I found a typing error in a date:
"WPA-3 SAE support: 2121-09-15"