morrownr/8812au-20210629 #52: 5GHz WIFI repeatedly asks for password never connects. Kernel v5.16|v15.5  - Ubuntu|PopOS

pygaurav, 2022-05-03:
I am using PopOS currently, I am unable to connect to 5GHz wifi, it repeatedly asks for password & never connects.

Bartholomew42, 2022-05-03:
Hi @pygaurav , 

what channel is that wifi network on? 
What is the output of `iwlist chan` ? 
What is the output of `iw list | grep MHz`? 
What is the output of `grep [[:alnum:]] /sys/module/8812au/parameters/* | grep country` ?

(Because I myself had trouble with the country code setting, my mind is primed with the idea that you might have a similar problem. I might be completely wrong and your problem might originate from something completely different.)

Greetings!

pygaurav, 2022-05-03:
Hi @Bartholomew42 ,

Thank you for your comment. Please find the below output for your reference.

Channel is 44
Output of `iwlist chan`:

    lo        no frequency information.
    
    enp3s0    no frequency information.
    
    wlp2s0    32 channels in total; available frequencies :
              Channel 01 : 2.412 GHz
              Channel 02 : 2.417 GHz
              Channel 03 : 2.422 GHz
              Channel 04 : 2.427 GHz
              Channel 05 : 2.432 GHz
              Channel 06 : 2.437 GHz
              Channel 07 : 2.442 GHz
              Channel 08 : 2.447 GHz
              Channel 09 : 2.452 GHz
              Channel 10 : 2.457 GHz
              Channel 11 : 2.462 GHz
              Channel 12 : 2.467 GHz
              Channel 13 : 2.472 GHz
              Channel 34 : 5.17 GHz
              Channel 36 : 5.18 GHz
              Channel 38 : 5.19 GHz
              Channel 40 : 5.2 GHz
              Channel 42 : 5.21 GHz
              Channel 44 : 5.22 GHz
              Channel 46 : 5.23 GHz
              Channel 48 : 5.24 GHz
              Channel 52 : 5.26 GHz
              Channel 56 : 5.28 GHz
              Channel 60 : 5.3 GHz
              Channel 64 : 5.32 GHz
              Channel 100 : 5.5 GHz
              Channel 104 : 5.52 GHz
              Channel 108 : 5.54 GHz
              Channel 112 : 5.56 GHz
              Channel 116 : 5.58 GHz
              Channel 120 : 5.6 GHz
              Channel 124 : 5.62 GHz
              Current Frequency:2.462 GHz (Channel 11)
    
    wlxd03745aeb6c3  32 channels in total; available frequencies :
              Channel 01 : 2.412 GHz
              Channel 02 : 2.417 GHz
              Channel 03 : 2.422 GHz
              Channel 04 : 2.427 GHz
              Channel 05 : 2.432 GHz
              Channel 06 : 2.437 GHz
              Channel 07 : 2.442 GHz
              Channel 08 : 2.447 GHz
              Channel 09 : 2.452 GHz
              Channel 10 : 2.457 GHz
              Channel 11 : 2.462 GHz
              Channel 36 : 5.18 GHz
              Channel 40 : 5.2 GHz
              Channel 44 : 5.22 GHz
              Channel 48 : 5.24 GHz
              Channel 52 : 5.26 GHz
              Channel 56 : 5.28 GHz
              Channel 60 : 5.3 GHz
              Channel 64 : 5.32 GHz
              Channel 100 : 5.5 GHz
              Channel 104 : 5.52 GHz
              Channel 108 : 5.54 GHz
              Channel 112 : 5.56 GHz
[... 106 lines cut ...]
			* 5660 MHz [132] (24.0 dBm)
			* 5680 MHz [136] (24.0 dBm)
			* 5700 MHz [140] (24.0 dBm)
			* 5745 MHz [149] (30.0 dBm)
			* 5765 MHz [153] (30.0 dBm)
			* 5785 MHz [157] (30.0 dBm)
			* 5805 MHz [161] (30.0 dBm)
			* 5825 MHz [165] (30.0 dBm)
			* 5920 MHz [184] (disabled)
			* 5940 MHz [188] (disabled)
			* 5960 MHz [2] (disabled)
			* 5980 MHz [6] (disabled)
			* 6000 MHz [10] (disabled)
			* 6020 MHz [14] (disabled)
			* 6040 MHz [18] (disabled)
			* 6060 MHz [22] (disabled)
			* 6080 MHz [26] (disabled)

Output of ``grep [[:alnum:]] /sys/module/8812au/parameters/* | grep rtw_c``

    /sys/module/8812au/parameters/rtw_channel:1
    /sys/module/8812au/parameters/rtw_channel_plan:255
    /sys/module/8812au/parameters/rtw_check_hw_status:0
    /sys/module/8812au/parameters/rtw_chip_version:0
    /sys/module/8812au/parameters/rtw_ch_switch_offload:0

Bartholomew42, 2022-05-03:
Hi @pygaurav , 
it seems to me that my initial idea was probably wrong. 
I think this could be an authentication error that has little to do with the driver. 

Such errors can occur when 
a) the access point and the client are far apart and can't receive each other very well (low signal strength on any side). Moving the client nearer to the access point will help in this situation. 
b) the access point won't allow the client to log in for any reason (the access point does not like the clients MAC address, some wifi security setting is wrong, a typing error in the password). 

Can you connect to the desired network using your other wifi network adapter (wlp2s0)? 
Do you have access to the log files of the access point?
Can you test if you can connect to any other 5GHz network with your 8812au adapter?

Greetings!

pygaurav, 2022-05-03:
Hi @Bartholomew42 ,

I was using Ubuntu 20.04 in the morning & it was working fine (as it was using the older kernel version). But when I upgraded to Ubuntu 22.04 problem start appearing (I am sure the error is because of the kernel version i.e. v5.16).

I am able to connect with my inbuilt adapter but the error is coming with the USB one.

Could you please suggest how can I change my `rtw_country_code`?

pygaurav, 2022-05-03:
I have changed value of `rtw_country_code` it is working fine now.

closed 2022-05-03 by pygaurav

Bartholomew42, 2022-05-03:
To change it temporarily until the next loading of the driver (in most times, the next booting of Ubuntu), use `sudo modprobe -r 8812au` (this unloads the driver), the reload it using `sudo modprobe 8812au rtw_country_code=XY` whereas XY has to be the identifier for your country (BR for Brazil, CN for China, CH for Switzerland and so on: [https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2#Officially_assigned_code_elements](url) ). 
You can check if the driver was loaded with the desired option with `grep [[:alnum:]] /sys/module/8812au/parameters/*|grep rtw_co`. In my case, the output is `/sys/module/8812au/parameters/rtw_country_code:DE` .
If you want to set the option permanently, use `sudo nano /etc/modprobe.d/8812au.conf` and add `rtw_country_code=XY` to the line with all the other options at the top of the config file. You can apply the change with a reboot or with `sudo modprobe -r 8812au & modprobe 8812au`.