morrownr/8812au-20210629 #53: TP-Link - Archer AC600 T2U Nano

zeph, 2022-05-10:
`{USB_DEVICE(0x2357, 0x011F), .driver_info = RTL8821}, /* TP-Link - Archer AC600 T2U Nano */`
this from the https://github.com/aircrack-ng/rtl8812au but neither of the 2 codebases work properly... if I hack your code with this 
```diff
diff --git a/os_dep/linux/usb_intf.c b/os_dep/linux/usb_intf.c
index 242ea99..b3c4d5b 100644
--- a/os_dep/linux/usb_intf.c
+++ b/os_dep/linux/usb_intf.c
@@ -184,6 +184,7 @@ static struct usb_device_id rtw_usb_id_tbl[] = {
        {USB_DEVICE(0x2357, 0x010D), .driver_info = RTL8812}, /* TP-Link - Archer T4U AC1300 */
        {USB_DEVICE(0x2357, 0x010E), .driver_info = RTL8812}, /* TP-Link - Archer T4UH AC1300 */
        {USB_DEVICE(0x2357, 0x010F), .driver_info = RTL8812}, /* TP-Link - T4UHP */
+       {USB_DEVICE(0x2357, 0x011F), .driver_info = RTL8812}, /* TP-Link - Archer AC600 T2U Nano */
        {USB_DEVICE(0x20F4, 0x805B), .driver_info = RTL8812}, /* TRENDnet - TEW-805UB */
        {USB_DEVICE(0x0411, 0x025D), .driver_info = RTL8812}, /* Buffalo - WI-U3-866D */
        {USB_DEVICE(0x050D, 0x1109), .driver_info = RTL8812}, /* Belkin F9L1109 - SerComm */
```

it matches, loads and gives me details... but an `iw scan` returns no networks

I'm testing with your code cuz the other repo/driver behaviour is pretty glitchy (lots of packet loss)

p.s. https://linux-hardware.org/?id=usb:2357-011f

zeph, 2022-05-10:
ok, thanks for the feedback @morrownr

closed 2022-05-10 by zeph

morrownr, 2022-05-16:
Hi @zeph 

I get the feeling that you did not read issue 45. In issue 45 I point users to the driver they should be using:

https://github.com/morrownr/8821au-20210708

That driver is for adapters that use 8821au and 8811au chipsets. It is a very good driver (as far as Realtek drivers are concerned.) Please give it a try and let me know what the results are.