morrownr/8812au-20210629 #55: Speed is not seen

castillofrancodamian, 2022-05-23:
[image]

morrownr, 2022-05-23:
Hola @castillofrancodamian 

There are various ways to see what your link speed is if that is what you want.

```
$ nmcli dev wifi
IN-USE  BSSID              SSID  MODE   CHAN  RATE        SIGNAL  BARS  SECURITY 
*       8C:59:73:FE:8B:F5  Bass  Infra  100   540 Mbit/s  82      ▂▄▆█  WPA2     
```

$ sudo apt install wavemon
$ wavemon

If you prefer a gui, your distro probably provides as much. If you tell me what distro are running, I might be able to help.

There is also iperf3:

$ sudo apt install iperf3

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.241 port 58658 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  46.1 MBytes   386 Mbits/sec    2    515 KBytes       
[  5]   1.00-2.00   sec  45.0 MBytes   377 Mbits/sec    2    407 KBytes       
[  5]   2.00-3.00   sec  43.8 MBytes   367 Mbits/sec    0    482 KBytes       
[  5]   3.00-4.00   sec  45.0 MBytes   377 Mbits/sec    0    544 KBytes       
[  5]   4.00-5.00   sec  43.8 MBytes   367 Mbits/sec    2    445 KBytes       
[  5]   5.00-6.00   sec  43.8 MBytes   367 Mbits/sec    0    516 KBytes       
[  5]   6.00-7.00   sec  45.0 MBytes   378 Mbits/sec    0    567 KBytes       
[  5]   7.00-8.00   sec  45.0 MBytes   377 Mbits/sec    0    600 KBytes       
[  5]   8.00-9.00   sec  42.5 MBytes   357 Mbits/sec    6    264 KBytes       
[  5]   9.00-10.00  sec  45.0 MBytes   377 Mbits/sec    2    266 KBytes       
```

To use it on a client system, you need iperf3 running as a server on another system on your network. It provides measured throughput.

Saludos

castillofrancodamian, 2022-05-23:
I have Arch Linux and KDE plasma.

castillofrancodamian, 2022-05-24:
```
[franco@pc openwrt]$ nmcli dev wifi
IN-USE  BSSID              SSID                     MODE   CHAN  RATE        SIGNAL  BARS  SECURITY  
        36:5D:9E:B4:E9:56  TeleCentro Wifi          Infra  1     195 Mbit/s  89      ▂▄▆█  --
```

morrownr, 2022-05-24:
Hmmm...

Given the signal reading of 89, I would expect a higher rate. 

What is the result of:

$ lsusb -t

castillofrancodamian, 2022-05-24:
```
[franco@pc openwrt]$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
    |__ Port 5: Dev 2, If 0, Class=Mass Storage, Driver=uas, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 3: Dev 2, If 0, Class=Vendor Specific Class, Driver=rtl8812au, 480M
    |__ Port 8: Dev 3, If 1, Class=Wireless, Driver=btusb, 12M
    |__ Port 8: Dev 3, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 12: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 12: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 12M

```

morrownr, 2022-05-25:
> Port 3: Dev 2, If 0, Class=Vendor Specific Class, Driver=rtl8812au, 480M

You are running is USB2 mode. Is that what you want?

castillofrancodamian, 2022-05-25:
> > Port 3: Dev 2, If 0, Class=Vendor Specific Class, Driver=rtl8812au, 480M
> 
> You are running is USB2 mode. Is that what you want?

What does it have to do? Besides, it is in a USB 3 (blue). That the speed does not appear does not have to have something to do with being "connected" to USB 2.0? Here I have a speed test connected to 5Ghz. In KDE neon I see the speed in the same panel.
```
[franco@pc ~]$ iperf3 -c enrutador.lan
Connecting to host enrutador.lan, port 5201
[  5] local 192.168.1.114 port 48414 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  16.3 MBytes   137 Mbits/sec    0    217 KBytes       
[  5]   1.00-2.00   sec  19.1 MBytes   160 Mbits/sec    0    217 KBytes       
[  5]   2.00-3.00   sec  19.7 MBytes   165 Mbits/sec    0    217 KBytes       
[  5]   3.00-4.00   sec  22.4 MBytes   188 Mbits/sec    0    217 KBytes       
[  5]   4.00-5.00   sec  22.7 MBytes   190 Mbits/sec    0    217 KBytes       
[  5]   5.00-6.00   sec  21.7 MBytes   182 Mbits/sec    0    217 KBytes       
[  5]   6.00-7.00   sec  21.9 MBytes   184 Mbits/sec    0    217 KBytes       
[  5]   7.00-8.00   sec  22.5 MBytes   188 Mbits/sec    0    217 KBytes       
[  5]   8.00-9.00   sec  21.8 MBytes   183 Mbits/sec    0    217 KBytes       
[  5]   9.00-10.00  sec  22.4 MBytes   188 Mbits/sec    0    217 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   210 MBytes   177 Mbits/sec    0             sender
[  5]   0.00-15.07  sec   210 MBytes   117 Mbits/sec                  receiver

iperf Done.
```

morrownr, 2022-05-25:
> What does it have to do?

Run the following from your driver directory:

$ sudo ./edit-options.sh

Copy and post the results of the Options line.

While you are in the 8812au.conf file, read the docs.

Regards

castillofrancodamian, 2022-05-25:
What do I have to look for specifically?

morrownr, 2022-05-25:
If you want the adapter to run in USB3 mode:

rtw_switch_usb_mode=1

morrownr, 2022-05-25:
> rtw_vht_enable=2

Are you running AP Mode, 5 GHz? If not, you need this setting to be:

rtw_vht_enable=1

A bad setting there and Client mode performance immediately goes to terrible.

castillofrancodamian, 2022-05-25:
> > rtw_vht_enable=2
> 
> Are you running AP Mode, 5 GHz? If not, you need this setting to be:
> 
> rtw_vht_enable=1
> 
> A bad setting there and Client mode performance immediately goes to terrible.

I didn't realize I had to uncomment the line. Now that I've uncommented it I can see `Port 3: Dev 2, If 0, Class=Vendor Specific Class, Driver=rtl8812au, 5000M`. Now, if I comment out the line again, it doesn't go back to USB 2. It stayed on USB 3. I've also noticed some instability in Ping and general usage whether I modify the file or not:

```
[franco@pc 8812au-20210629]$ ping 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=57 time=10.6 ms
64 bytes from 1.1.1.1: icmp_seq=2 ttl=57 time=11.0 ms
64 bytes from 1.1.1.1: icmp_seq=3 ttl=57 time=10.2 ms
64 bytes from 1.1.1.1: icmp_seq=4 ttl=57 time=12.3 ms
64 bytes from 1.1.1.1: icmp_seq=5 ttl=57 time=10.9 ms
64 bytes from 1.1.1.1: icmp_seq=6 ttl=57 time=11.3 ms
64 bytes from 1.1.1.1: icmp_seq=7 ttl=57 time=11.9 ms
64 bytes from 1.1.1.1: icmp_seq=8 ttl=57 time=10.7 ms
64 bytes from 1.1.1.1: icmp_seq=9 ttl=57 time=10.9 ms
64 bytes from 1.1.1.1: icmp_seq=10 ttl=57 time=11.8 ms
64 bytes from 1.1.1.1: icmp_seq=11 ttl=57 time=12.0 ms
64 bytes from 1.1.1.1: icmp_seq=12 ttl=57 time=10.7 ms
64 bytes from 1.1.1.1: icmp_seq=13 ttl=57 time=11.5 ms
64 bytes from 1.1.1.1: icmp_seq=14 ttl=57 time=11.0 ms
64 bytes from 1.1.1.1: icmp_seq=15 ttl=57 time=10.5 ms
64 bytes from 1.1.1.1: icmp_seq=16 ttl=57 time=10.9 ms
64 bytes from 1.1.1.1: icmp_seq=18 ttl=57 time=2424 ms
64 bytes from 1.1.1.1: icmp_seq=17 ttl=57 time=3447 ms
64 bytes from 1.1.1.1: icmp_seq=19 ttl=57 time=1410 ms
64 bytes from 1.1.1.1: icmp_seq=20 ttl=57 time=397 ms
64 bytes from 1.1.1.1: icmp_seq=21 ttl=57 time=10.0 ms
64 bytes from 1.1.1.1: icmp_seq=22 ttl=57 time=11.0 ms
64 bytes from 1.1.1.1: icmp_seq=23 ttl=57 time=11.8 ms
64 bytes from 1.1.1.1: icmp_seq=24 ttl=57 time=11.3 ms
64 bytes from 1.1.1.1: icmp_seq=25 ttl=57 time=13.8 ms
From 192.168.1.114 icmp_seq=27 Destination Host Unreachable
From 192.168.1.114 icmp_seq=28 Destination Host Unreachable
From 192.168.1.114 icmp_seq=29 Destination Host Unreachable
From 192.168.1.114 icmp_seq=30 Destination Host Unreachable
From 192.168.1.114 icmp_seq=31 Destination Host Unreachable
From 192.168.1.114 icmp_seq=32 Destination Host Unreachable
From 192.168.1.114 icmp_seq=33 Destination Host Unreachable
From 192.168.1.114 icmp_seq=34 Destination Host Unreachable
From 192.168.1.114 icmp_seq=35 Destination Host Unreachable
From 192.168.1.114 icmp_seq=36 Destination Host Unreachable
From 192.168.1.114 icmp_seq=37 Destination Host Unreachable
From 192.168.1.114 icmp_seq=38 Destination Host Unreachable
^C
--- 1.1.1.1 ping statistics ---
39 packets transmitted, 25 received, +12 errors, 35.8974% packet loss, time 38231ms
rtt min/avg/max/mdev = 10.031/316.572/3447.267/834.304 ms, pipe 4
[franco@pc 8812au-20210629]$ ping 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
^C
--- 1.1.1.1 ping statistics ---
4 packets transmitted, 0 received, 100% packet loss, time 3048ms
```

In iperf3 I have also noticed a lot of instability:

```
[franco@pc 8812au-20210629]$ iperf3 -c enrutador.lan
Connecting to host enrutador.lan, port 5201
[  5] local 192.168.1.114 port 45874 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  80.3 KBytes   658 Kbits/sec   20   1.20 KBytes
[  5]   1.00-2.00   sec  1.65 MBytes  13.8 Mbits/sec   13   3.60 KBytes
[... 17 lines cut ...]
[  5] local 192.168.1.114 port 49366 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  10.0 MBytes  84.3 Mbits/sec   16   50.4 KBytes
[  5]   1.00-2.00   sec  16.8 MBytes   141 Mbits/sec   16   74.4 KBytes
[  5]   2.00-3.00   sec  16.2 MBytes   136 Mbits/sec   14   73.2 KBytes
[  5]   3.00-4.00   sec  16.3 MBytes   137 Mbits/sec    5   70.8 KBytes
[  5]   4.00-5.00   sec  12.7 MBytes   106 Mbits/sec   25   56.4 KBytes
[  5]   5.00-6.00   sec  15.8 MBytes   133 Mbits/sec   24   45.6 KBytes
[  5]   6.00-7.00   sec  15.0 MBytes   126 Mbits/sec    2   95.9 KBytes
[  5]   7.00-8.00   sec  20.1 MBytes   169 Mbits/sec    9   87.5 KBytes
[  5]   8.00-9.00   sec  15.9 MBytes   133 Mbits/sec   30    101 KBytes
[  5]   9.00-10.00  sec  18.4 MBytes   154 Mbits/sec   13    112 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   157 MBytes   132 Mbits/sec  154             sender
[  5]   0.00-10.00  sec   157 MBytes   131 Mbits/sec                  receiver

iperf Done.

```
I think it is due to using the latest kernel because previously with KDE neon with kernel 5.11 (if I remember correctly) this did not happen.
By the way, the device is a TP-Link Archer T4UHP.
To all this, I still see the problem of the main image. And when doing tests I see that there are several problems.

Then testing with the line uncommented and having changed `rtw_vht_enable=2` to `rtw_vht_enable=1`, resulting in the line at `options 8812au rtw_drv_log_level=0 rtw_led_ctrl=0 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_beamform_cap=0 rtw_dfs_region_switchus_domain=1 rtw_domain=1 1`, I still have all the problems presented.

morrownr, 2022-05-25:
There is definitely a problem.

> I think it is due to using the latest kernel because previously with KDE neon with kernel 5.11 (if I remember correctly) this did not happen.

This could be the case because I, and others that help, have to regularly update this driver to keep up with changes in the Linux kernel. What kernel version are you using now?

morrownr, 2022-05-25:
I'd recommend that you remove the driver you correctly have but do not reboot after running the script:

$ sudo ./remove-driver.sh

When it asks if you want to reboot, say no. That should leave the driver active so you can download the driver again.

Now, delete the driver directory. Once that is complete, start at the beginning of the installation instructions and when ask if you want to reboot by the installation script, say yes.

Side note: While I do not change the version of the driver as time goes along, unless Realtek does release a new version, there is work done on a regular basis to keep it working with new kernels. Rule of thumb: If you upgrade kernel such as 5.8 to 5.17, you really need to remove this driver before the upgrade and then install a freshly download version once the upgrade is complete. I realize this is pain but it is what Realtek has provided... which is why I recommend Mediatek based adapters to those who are looking for an adapter. I keep more info at:

https://github.com/morrownr/USB-WiFi

castillofrancodamian, 2022-05-25:
> I'd recommend that you remove the driver you correctly have but do not reboot after running the script:
>
> $ sudo ./remove-driver.sh
>
> When it asks if you want to reboot, say no. That should leave the [...]

So far I have had the controller directory updated with git pull. I still have a cable connection to be able to download.

castillofrancodamian, 2022-05-26:
I have installed KDE neon on another partition and I get the following:

```
franco@pc:~$ uname -a
Linux pc 5.13.0-44-generic #49~20.04.1-Ubuntu SMP Wed May 18 18:44:28 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

franco@pc:~$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
    |__ Port 3: Dev 4, If 0, Class=Vendor Specific Class, Driver=rtl8812au, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 8: Dev 2, If 1, Class=Wireless, Driver=btusb, 12M
    |__ Port 8: Dev 2, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 12: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 12: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
```

```
franco@pc:~$ nmcli dev wifi
IN-USE  BSSID              SSID                     MODE   CHAN  RATE        SIGNAL  BARS  SECURITY
*       50:C7:BF:75:81:AF  LAG                      Infra  36    270 Mbit/s  100     ▂▄▆█  WPA2
        52:C7:BF:75:81:AF  Invitados                Infra  36    270 Mbit/s  100     ▂▄▆█  WPA2
        B0:FC:36:87:D5:A0  Fibertel WiFi889 2.4GHz  Infra  6     195 Mbit/s  97      ▂▄▆█  WPA1 WPA2
        52:C7:BF:75:81:B0  Invitados                Infra  1     195 Mbit/s  95      ▂▄▆█  WPA2
        50:C7:BF:75:81:B0  LAG                      Infra  1     195 Mbit/s  92      ▂▄▆█  WPA2
        58:2F:F7:8D:03:8C  Fibertel WiFi950 2.4GHz  Infra  11    260 Mbit/s  92      ▂▄▆█  WPA1 WPA2
        46:AD:B1:2A:06:11  TeleCentro Wifi          Infra  2     195 Mbit/s  80      ▂▄▆_  --
        40:3F:8C:DD:CC:24  TP-Link_CC24             Infra  9     270 Mbit/s  80      ▂▄▆_  WPA2
        F8:08:4F:7E:3E:EA  TeleCentro-3ee5          Infra  11    260 Mbit/s  80      ▂▄▆_  WPA1 WPA2
        FA:08:4F:7E:3F:EC  TeleCentro Wifi          Infra  11    260 Mbit/s  80      ▂▄▆_  --
        36:5D:9E:B4:E9:56  TeleCentro Wifi          Infra  6     195 Mbit/s  77      ▂▄▆_  --
        34:5D:9E:B4:E7:55  Telecentro-e750          Infra  6     195 Mbit/s  75      ▂▄▆_  WPA2
        44:AD:B1:2A:04:10  Telecentro-040b          Infra  2     195 Mbit/s  74      ▂▄▆_  WPA2
        B8:D9:4D:68:75:14  TeleCentro-750f          Infra  4     260 Mbit/s  74      ▂▄▆_  WPA2
        BA:D9:4D:68:76:16  TeleCentro Wifi          Infra  4     260 Mbit/s  72      ▂▄▆_  --
        76:4F:57:AD:56:7E  Invitados                Infra  1     130 Mbit/s  64      ▂▄▆_  WPA2
        72:4F:57:AD:56:7E  LAG                      Infra  1     130 Mbit/s  60      ▂▄▆_  WPA2
        B0:FC:36:87:D5:A8  Fibertel WiFi889 5.8GHz  Infra  56    405 Mbit/s  49      ▂▄__  WPA1 WPA2
        58:2F:F7:8D:03:8D  Fibertel WiFi950 5.8GHz  Infra  64    540 Mbit/s  45      ▂▄__  WPA1 WPA2
        34:5D:9E:B4:E7:56  Telecentro-e750          Infra  149   540 Mbit/s  30      ▂___  WPA2
        36:5D:9E:B4:E8:57  --                       Infra  149   540 Mbit/s  30      ▂___  WPA2
        36:5D:9E:B4:E8:50  TeleCentro Wifi          Infra  149   540 Mbit/s  30      ▂___  --
        F8:08:4F:7E:3E:EB  TeleCentro-3ee5-5G       Infra  149   540 Mbit/s  20      ▂___  WPA1 WPA2
        FA:08:4F:7E:40:ED  TeleCentro Wifi          Infra  149   540 Mbit/s  20      ▂___  --
        A0:39:EE:B0:C8:27  Fibertel WiFi690 5.8GHz  Infra  64    405 Mbit/s  19      ▂___  WPA2
        A0:39:EE:B0:CD:C7  Laura Wifi 5.8           Infra  149   195 Mbit/s  19      ▂___  WPA2
```

```
franco@pc:~$ ping 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=57 time=9.96 ms
64 bytes from 1.1.1.1: icmp_seq=2 ttl=57 time=11.3 ms
64 bytes from 1.1.1.1: icmp_seq=3 ttl=57 time=11.9 ms
64 bytes from 1.1.1.1: icmp_seq=4 ttl=57 time=11.9 ms
64 bytes from 1.1.1.1: icmp_seq=5 ttl=57 time=11.0 ms
64 bytes from 1.1.1.1: icmp_seq=6 ttl=57 time=11.9 ms
64 bytes from 1.1.1.1: icmp_seq=7 ttl=57 time=11.0 ms
64 bytes from 1.1.1.1: icmp_seq=8 ttl=57 time=11.4 ms
64 bytes from 1.1.1.1: icmp_seq=9 ttl=57 time=9.97 ms
64 bytes from 1.1.1.1: icmp_seq=10 ttl=57 time=13.2 ms
64 bytes from 1.1.1.1: icmp_seq=11 ttl=57 time=11.9 ms
64 bytes from 1.1.1.1: icmp_seq=12 ttl=57 time=21.5 ms
64 bytes from 1.1.1.1: icmp_seq=13 ttl=57 time=11.3 ms
64 bytes from 1.1.1.1: icmp_seq=14 ttl=57 time=27.4 ms
64 bytes from 1.1.1.1: icmp_seq=15 ttl=57 time=12.6 ms
64 bytes from 1.1.1.1: icmp_seq=16 ttl=57 time=11.7 ms
64 bytes from 1.1.1.1: icmp_seq=17 ttl=57 time=11.8 ms
64 bytes from 1.1.1.1: icmp_seq=18 ttl=57 time=25.6 ms
64 bytes from 1.1.1.1: icmp_seq=19 ttl=57 time=11.1 ms
64 bytes from 1.1.1.1: icmp_seq=20 ttl=57 time=32.3 ms
[... 71 lines cut ...]

```
franco@pc:~$ iperf3 -c enrutador.lan
Connecting to host enrutador.lan, port 5201
[  5] local 192.168.1.235 port 56362 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  24.5 MBytes   205 Mbits/sec    0    215 KBytes
[  5]   1.00-2.00   sec  22.6 MBytes   190 Mbits/sec    0    215 KBytes
[  5]   2.00-3.00   sec  24.6 MBytes   206 Mbits/sec    0    215 KBytes
[  5]   3.00-4.00   sec  24.0 MBytes   201 Mbits/sec    0    215 KBytes
[  5]   4.00-5.00   sec  25.6 MBytes   215 Mbits/sec    0    215 KBytes
[  5]   5.00-6.00   sec  23.3 MBytes   195 Mbits/sec    0    215 KBytes
[  5]   6.00-7.00   sec  26.3 MBytes   221 Mbits/sec    0    215 KBytes
[  5]   7.00-8.00   sec  23.9 MBytes   200 Mbits/sec   30    167 KBytes
[  5]   8.00-9.00   sec  23.5 MBytes   197 Mbits/sec    0    211 KBytes
[... cut ...]
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   241 MBytes   202 Mbits/sec   30             sender
[  5]   0.00-10.01  sec   241 MBytes   202 Mbits/sec                  receiver

iperf Done.
```
I have uncommented the options line to `options 8812au rtw_drv_log_level=0 rtw_led_ctrl=0 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_beamform_cap=0 rtw_dfs_region_domain=1 rtw_switch_usb_mode=1`.
I have still noticed some instability but not as much as with Arch Linux.

morrownr, 2022-05-26:
This new data shows a more solid connection as there are few retries. The iperf3 value of 202 Mb/s is solid but doesn't tell us much as this is a sharing technology so other systems on the same channel could be eating into the available bandwidth.

>  I have uncommented the options line

I'n curious about this commend as the `options` line is not commented out in a default installation.

castillofrancodamian, 2022-05-26:
I am the only one using channel 36, so there is no interference.

> I'n curious about this commend as the options line is not commented out in a default installation.

How not? I see [here](# options 8812au rtw_drv_log_level=0 rtw_led_ctrl=0 rtw_vht_enable=2 rtw_power_mgnt=1 rtw_beamform_cap=0 rtw_dfs_region_domain=1 rtw_switch_usb_mode=1) that it is commented.

castillofrancodamian, 2022-06-17:
Why would it have to go to 400 Mbps? I have approx. 250 Mbps. Perhaps due to having a not so powerful router (Archer C60).

```
[franco@pc ~]$ iperf3 -c enrutador.lan
Connecting to host enrutador.lan, port 5201
[  5] local 192.168.1.235 port 34868 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  30.3 MBytes   254 Mbits/sec    0    218 KBytes       
[  5]   1.00-2.00   sec  29.8 MBytes   250 Mbits/sec    0    218 KBytes       
[  5]   2.00-3.00   sec  29.8 MBytes   250 Mbits/sec    0    218 KBytes       
[  5]   3.00-4.00   sec  30.2 MBytes   253 Mbits/sec    0    218 KBytes       
[  5]   4.00-5.00   sec  29.8 MBytes   250 Mbits/sec    0    218 KBytes       
[  5]   5.00-6.00   sec  1.74 MBytes  14.6 Mbits/sec    1   1.41 KBytes       
[  5]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  5]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  5]   8.00-9.00   sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  5]   9.00-10.00  sec  27.0 MBytes   227 Mbits/sec    1    219 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   179 MBytes   150 Mbits/sec    2             sender
[  5]   0.00-10.01  sec   178 MBytes   149 Mbits/sec                  receiver
```

I have also modified the correct line but the problems/errors are still there.

morrownr, 2022-06-18:
> Why would it have to go to 400 Mbps?

You don't have to have 400 Mb/s. To get 400 everything has to be capable of pushing it and that may not be the case. I should have said since you have no competing traffic on your wifi channel you could see up to 400 if everything is up to the task.

I see that the bitrate dropped off to zero. That is not good. Sometimes it can be a good idea to run the following, then delete the folder with the driver and start over:

$ ,/remove-driver.sh

castillofrancodamian, 2022-12-05:
To this day I still have the same symptoms. High latency every few seconds as can be seen from Ping and connection speed at 0 Bits/s.
I have to clarify that I have the same symptoms if I use the [aircrack-ng](https://github.com/aircrack-ng/rtl8812au/issues/1019) driver.

```
[franco@pc ~]$ ping 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=57 time=1339 ms
64 bytes from 1.1.1.1: icmp_seq=2 ttl=57 time=313 ms
64 bytes from 1.1.1.1: icmp_seq=3 ttl=57 time=10.2 ms
64 bytes from 1.1.1.1: icmp_seq=4 ttl=57 time=10.6 ms
64 bytes from 1.1.1.1: icmp_seq=5 ttl=57 time=10.8 ms
64 bytes from 1.1.1.1: icmp_seq=6 ttl=57 time=11.2 ms
64 bytes from 1.1.1.1: icmp_seq=7 ttl=57 time=11.0 ms
64 bytes from 1.1.1.1: icmp_seq=8 ttl=57 time=11.8 ms
64 bytes from 1.1.1.1: icmp_seq=9 ttl=57 time=11.8 ms
64 bytes from 1.1.1.1: icmp_seq=10 ttl=57 time=11.2 ms
64 bytes from 1.1.1.1: icmp_seq=11 ttl=57 time=12.0 ms
64 bytes from 1.1.1.1: icmp_seq=12 ttl=57 time=10.8 ms
64 bytes from 1.1.1.1: icmp_seq=14 ttl=57 time=2422 ms
64 bytes from 1.1.1.1: icmp_seq=13 ttl=57 time=3430 ms
64 bytes from 1.1.1.1: icmp_seq=16 ttl=57 time=396 ms
64 bytes from 1.1.1.1: icmp_seq=15 ttl=57 time=1409 ms
64 bytes from 1.1.1.1: icmp_seq=17 ttl=57 time=11.3 ms
64 bytes from 1.1.1.1: icmp_seq=18 ttl=57 time=12.2 ms
64 bytes from 1.1.1.1: icmp_seq=19 ttl=57 time=10.4 ms
64 bytes from 1.1.1.1: icmp_seq=20 ttl=57 time=10.3 ms
64 bytes from 1.1.1.1: icmp_seq=21 ttl=57 time=12.7 ms
64 bytes from 1.1.1.1: icmp_seq=22 ttl=57 time=18.1 ms
64 bytes from 1.1.1.1: icmp_seq=23 ttl=57 time=11.2 ms
64 bytes from 1.1.1.1: icmp_seq=24 ttl=57 time=11.6 ms
64 bytes from 1.1.1.1: icmp_seq=25 ttl=57 time=11.7 ms
64 bytes from 1.1.1.1: icmp_seq=26 ttl=57 time=11.5 ms
64 bytes from 1.1.1.1: icmp_seq=27 ttl=57 time=3406 ms
64 bytes from 1.1.1.1: icmp_seq=28 ttl=57 time=2397 ms
64 bytes from 1.1.1.1: icmp_seq=29 ttl=57 time=1389 ms
64 bytes from 1.1.1.1: icmp_seq=30 ttl=57 time=376 ms
64 bytes from 1.1.1.1: icmp_seq=31 ttl=57 time=11.7 ms
64 bytes from 1.1.1.1: icmp_seq=32 ttl=57 time=11.6 ms
64 bytes from 1.1.1.1: icmp_seq=33 ttl=57 time=11.4 ms
64 bytes from 1.1.1.1: icmp_seq=34 ttl=57 time=13.5 ms
64 bytes from 1.1.1.1: icmp_seq=35 ttl=57 time=9.60 ms
64 bytes from 1.1.1.1: icmp_seq=36 ttl=57 time=10.9 ms
64 bytes from 1.1.1.1: icmp_seq=37 ttl=57 time=10.8 ms
64 bytes from 1.1.1.1: icmp_seq=38 ttl=57 time=11.4 ms
64 bytes from 1.1.1.1: icmp_seq=39 ttl=57 time=9.75 ms
64 bytes from 1.1.1.1: icmp_seq=40 ttl=57 time=13.5 ms
64 bytes from 1.1.1.1: icmp_seq=41 ttl=57 time=3589 ms
64 bytes from 1.1.1.1: icmp_seq=42 ttl=57 time=2580 ms
64 bytes from 1.1.1.1: icmp_seq=44 ttl=57 time=557 ms
64 bytes from 1.1.1.1: icmp_seq=43 ttl=57 time=1571 ms
64 bytes from 1.1.1.1: icmp_seq=45 ttl=57 time=11.3 ms
64 bytes from 1.1.1.1: icmp_seq=46 ttl=57 time=9.87 ms
^C
--- 1.1.1.1 ping statistics ---
46 packets transmitted, 46 received, 0% packet loss, time 45186ms
rtt min/avg/max/mdev = 9.598/555.245/3588.521/1034.367 ms, pipe 4
```

[image]

```
[franco@pc ~]$ uname -a
Linux pc 6.0.11-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 02 Dec 2022 17:25:31 +0000 x86_64 GNU/Linux
```

```
[franco@pc ~]$ nmcli dev wifi
IN-USE  BSSID              SSID                      MODE   CHAN  RATE        SIGNAL  BARS  SECURITY  
        A2:39:EE:A2:FB:12  Personal Wifi Zone        Infra  1     195 Mbit/s  100     ▂▄▆█  --        
        A0:39:EE:A2:FA:1E  LAG                       Infra  1     195 Mbit/s  100     ▂▄▆█  WPA2      
[... 195 lines cut ...]
# hw ht capab: 0x862
# ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
#
# hw vht capab: 0x03c001a2
# vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
#
# -----

```

```
[franco@pc ~]$ iwconfig
lo        no wireless extensions.

enp1s0    no wireless extensions.

wlan0     IEEE 802.11AC  ESSID:"LAG"  Nickname:"WIFI@RTL8812AU"
          Mode:Managed  Frequency:5.56 GHz  Access Point: A0:39:EE:A2:FA:1F   
          Bit Rate:867 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=64/100  Signal level=88/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

castillofrancodamian, 2023-06-15:
@morrownr Hello, I still have the problems. And this is the worst. It really is very annoying.
```
[franco@pc ~]$ ping archer_c60
PING archer_c60.lan (10.1.0.1) 56(84) bytes of data.
64 bytes from _gateway (10.1.0.1): icmp_seq=1 ttl=64 time=1.71 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=2 ttl=64 time=17.6 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=3 ttl=64 time=2.21 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=4 ttl=64 time=3.15 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=5 ttl=64 time=3.76 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=6 ttl=64 time=2.24 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=7 ttl=64 time=3.41 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=8 ttl=64 time=2.60 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=9 ttl=64 time=2.03 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=10 ttl=64 time=2.65 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=11 ttl=64 time=3228 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=12 ttl=64 time=2182 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=13 ttl=64 time=1161 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=14 ttl=64 time=137 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=15 ttl=64 time=1.95 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=16 ttl=64 time=2.09 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=17 ttl=64 time=1.72 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=18 ttl=64 time=11.5 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=19 ttl=64 time=2.32 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=20 ttl=64 time=2.01 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=21 ttl=64 time=2.19 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=22 ttl=64 time=2.31 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=23 ttl=64 time=2.67 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=24 ttl=64 time=1.93 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=25 ttl=64 time=3308 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=26 ttl=64 time=2288 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=27 ttl=64 time=1264 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=28 ttl=64 time=241 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=29 ttl=64 time=4.40 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=30 ttl=64 time=3.63 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=31 ttl=64 time=1.72 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=32 ttl=64 time=2.33 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=33 ttl=64 time=2.19 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=34 ttl=64 time=4.04 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=35 ttl=64 time=2.54 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=36 ttl=64 time=3.95 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=37 ttl=64 time=2.77 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=38 ttl=64 time=2.20 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=39 ttl=64 time=3440 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=40 ttl=64 time=2421 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=41 ttl=64 time=1399 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=42 ttl=64 time=373 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=43 ttl=64 time=2.60 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=44 ttl=64 time=2.75 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=45 ttl=64 time=2.65 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=46 ttl=64 time=3.70 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=47 ttl=64 time=2.13 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=48 ttl=64 time=2.84 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=49 ttl=64 time=2.14 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=50 ttl=64 time=1.80 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=51 ttl=64 time=3.03 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=52 ttl=64 time=3.61 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=53 ttl=64 time=3706 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=54 ttl=64 time=2690 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=55 ttl=64 time=1666 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=56 ttl=64 time=642 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=57 ttl=64 time=2.05 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=58 ttl=64 time=2.59 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=59 ttl=64 time=2.32 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=60 ttl=64 time=2.72 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=61 ttl=64 time=1.89 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=62 ttl=64 time=2.27 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=63 ttl=64 time=2.30 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=64 ttl=64 time=2.48 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=65 ttl=64 time=2.67 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=66 ttl=64 time=2.87 ms
[... 6 lines cut ...]
64 bytes from _gateway (10.1.0.1): icmp_seq=73 ttl=64 time=2.12 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=74 ttl=64 time=2.52 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=75 ttl=64 time=2.81 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=76 ttl=64 time=2.68 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=77 ttl=64 time=2.29 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=78 ttl=64 time=2.31 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=79 ttl=64 time=2.11 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=80 ttl=64 time=2.58 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=81 ttl=64 time=3995 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=82 ttl=64 time=2974 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=83 ttl=64 time=1950 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=84 ttl=64 time=926 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=85 ttl=64 time=2.15 ms
[... cut ...]
64 bytes from _gateway (10.1.0.1): icmp_seq=88 ttl=64 time=2.23 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=89 ttl=64 time=2.36 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=90 ttl=64 time=2.10 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=91 ttl=64 time=1.78 ms
64 bytes from _gateway (10.1.0.1): icmp_seq=92 ttl=64 time=1.77 ms
^C
--- archer_c60.lan ping statistics ---
92 packets transmitted, 92 received, 0% packet loss, time 91518ms
rtt min/avg/max/mdev = 1.709/537.135/3994.994/1082.410 ms, pipe 4
```

morrownr, 2023-06-15:
I may have found and fixed the issue. Please test.

$ git pull
$ sudo sh install-driver.sh

castillofrancodamian, 2023-06-15:
@morrownr keeps malfunctioning 😢
```
[franco@pc ~]$ git clone https://github.com/morrownr/8812au-20210629.git
Clonando en '8812au-20210629'...
remote: Enumerating objects: 1394, done.
remote: Counting objects: 100% (284/284), done.
remote: Compressing objects: 100% (130/130), done.
remote: Total 1394 (delta 164), reused 252 (delta 147), pack-reused 1110
Recibiendo objetos: 100% (1394/1394), 3.79 MiB | 8.69 MiB/s, listo.
Resolviendo deltas: 100% (703/703), listo.
[franco@pc ~]$ cd 8812au-20210629/
[franco@pc 8812au-20210629]$ sudo sh install-driver.sh
[sudo] contraseña para franco:
: ---------------------------
: install-driver.sh v20230227
: x86_64 (architecture)
: 4/4 (in-use/total processing units)
: 7826492 (total system memory)
: 6.3.8-zen1-1-zen (kernel version)
: gcc (GCC) 13.1.1 20230429
: dkms-3.0.11
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 8812au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8812au-5.13.6
Creating symlink /var/lib/dkms/rtl8812au/5.13.6/source -> /usr/src/rtl8812au-5.13.6
The driver was added to dkms successfully.
: ---------------------------
Sign command: /usr/lib/modules/6.3.8-zen1-1-zen/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
./dkms-make.sh...........................
Signing module /var/lib/dkms/rtl8812au/5.13.6/build/8812au.ko
Cleaning build area...
Compile time: 200.27 seconds
The driver was built by dkms successfully.
: ---------------------------

8812au.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/6.3.8-zen1-1-zen/updates/dkms/
depmod....
The driver was installed by dkms successfully.
: ---------------------------

Info: Upgrade this driver with the following commands as needed:
$ git pull
$ sudo sh install-driver.sh
Note: Upgrades to this driver should be performed before distro upgrades.
Note: Upgrades can be performed as often as you like.
Note: Work on this driver is continuous.
: ---------------------------

Do you want to edit the driver options file now? (recommended) [Y/n]

options 8812au rtw_drv_log_level=1 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_switch_usb_mode=1 rtw_country_code=AR

Do you want to apply the new options by rebooting now? (recommended) [Y/n] n
```
[... 63 lines cut ...]
64 bytes from 1.1.1.1: icmp_seq=16 ttl=57 time=20.6 ms
64 bytes from 1.1.1.1: icmp_seq=17 ttl=57 time=26.4 ms
64 bytes from 1.1.1.1: icmp_seq=18 ttl=57 time=18.6 ms
64 bytes from 1.1.1.1: icmp_seq=19 ttl=57 time=20.0 ms
64 bytes from 1.1.1.1: icmp_seq=20 ttl=57 time=20.7 ms
64 bytes from 1.1.1.1: icmp_seq=21 ttl=57 time=16.4 ms
64 bytes from 1.1.1.1: icmp_seq=22 ttl=57 time=21.2 ms
64 bytes from 1.1.1.1: icmp_seq=23 ttl=57 time=20.3 ms
64 bytes from 1.1.1.1: icmp_seq=25 ttl=57 time=2632 ms
64 bytes from 1.1.1.1: icmp_seq=26 ttl=57 time=1608 ms
64 bytes from 1.1.1.1: icmp_seq=24 ttl=57 time=3672 ms
64 bytes from 1.1.1.1: icmp_seq=27 ttl=57 time=596 ms
64 bytes from 1.1.1.1: icmp_seq=28 ttl=57 time=19.0 ms
64 bytes from 1.1.1.1: icmp_seq=29 ttl=57 time=24.1 ms
64 bytes from 1.1.1.1: icmp_seq=30 ttl=57 time=14.6 ms
64 bytes from 1.1.1.1: icmp_seq=31 ttl=57 time=22.3 ms
64 bytes from 1.1.1.1: icmp_seq=32 ttl=57 time=21.5 ms
64 bytes from 1.1.1.1: icmp_seq=33 ttl=57 time=23.3 ms
64 bytes from 1.1.1.1: icmp_seq=34 ttl=57 time=18.5 ms
64 bytes from 1.1.1.1: icmp_seq=35 ttl=57 time=21.5 ms
^C
--- 1.1.1.1 ping statistics ---
35 packets transmitted, 35 received, 0% packet loss, time 34233ms
rtt min/avg/max/mdev = 14.627/515.873/3810.154/1064.091 ms, pipe 4
```

morrownr, 2023-06-16:
I reverted my attempts.

I will continue looking,

morrownr, 2023-06-16:
I have been unable to duplicate the problem again. I add this to my regular checks that I perform on all of the drivers.

If this is making you loco then you could take a look at the Main Menu. Menu item 2 has information on many adapters that use in-kernel drivers:

https://github.com/morrownr/USB-WiFi

FYI: I used to live in AR.

morrownr, 2023-07-28:
Status?

castillofrancodamian, 2023-07-28:
@morrownr It has never been corrected. I have tried changing USB ports, Linux distribution, the network I connect to to test, Kernel version (stable or ZEN) and everything remains the same. I have also tried the [Aircrack version](https://github.com/aircrack-ng/rtl8812au) and I have the same "symptoms".

morrownr, 2023-07-28:
I have been keeping an eye open to see if I could see other situations like this and I have not. I just reread the thread and I have nothing to add for now. This repo averages about 150 clones per day so there are a lot of people using this driver. Given the amount of users, there are not many bug reports and I can't see any with this problem.

I'll continue to keep an eye open.

castillofrancodamian, 2023-07-28:
@morrownr In Windows it works perfect. What I do notice is that it doesn't always happen. I now have unlimited Ping to 1.1.1.1 and see ~20ms stable responses.

dubhater, 2023-11-21:
The connection speed in the screenshot in the first post is supposed to be the current transmit rate used by your Wifi adapter. For some reason the Realtek out-of-kernel drivers don't report it. This is nothing to worry about.

(As far as I can tell, the „RATE” reported by ``nmcli dev wifi`` is each AP's current transmit rate.)

The ping issue could be due to power saving. You can disable it using a module parameter: ``rtw_lps_level=0``

closed 2023-12-21 by morrownr

michelwi, 2024-01-08:
> This repo averages about 150 clones per day so there are a lot of people using this driver. Given the amount of users, there are not many bug reports and I can't see any with this problem.

Hello, I would like to report that I, unfortunately, seem to have the same problem with these latency spikes that @castillofrancodamian reported (while pinging my router about 4s of latency every 10s, very consistently).

Let me start off with the info about my system:
Debian 11 ([openmediavault](https://www.openmediavault.org/) version 6)
```
# uname -a
Linux omv 6.1.0-0.deb11.7-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.20-2~bpo11+1 (2023-04-23) x86_64 GNU/Linux
# lsusb
Bus 002 Device 002: ID 1058:0740 Western Digital Technologies, Inc. My Passport Essential (WDBACY)
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
Bus 001 Device 003: ID 046a:0023 Cherry GmbH Keyboard
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 5000M
    |__ Port 2: Dev 2, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 9: Dev 4, If 0, Class=Vendor Specific Class, Driver=rtl8812au, 480M
    |__ Port 14: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 14: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
# rfkill list all
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
# dkms status
rtl8812au, 5.13.6, 6.1.0-0.deb11.7-amd64, x86_64: installed
zfs, 2.1.11, 6.1.0-0.deb11.13-amd64, x86_64: installed
zfs, 2.1.11, 6.1.0-0.deb11.7-amd64, x86_64: installed
# iw dev
phy#0
        Interface wlx048d392a5ed7
                ifindex 2
                wdev 0x1
                addr [REDACTED]
                ssid [REDACTED]
                type managed
                channel 52 (5260 MHz), width: 80 MHz, center1: 5290 MHz
                txpower 16.00 dBm
# iw reg get
global
country 00: DFS-UNSET
        (755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)
```

A ping to my wifi router shows consistent 4s latency spikes every 10s:
```
$ ping 192.168.0.1
PING 192.168.0.1 (192.168.0.1) 56(84) bytes of data.
64 bytes from 192.168.0.1: icmp_seq=1 ttl=64 time=5.09 ms
64 bytes from 192.168.0.1: icmp_seq=2 ttl=64 time=3911 ms
64 bytes from 192.168.0.1: icmp_seq=3 ttl=64 time=2902 ms
64 bytes from 192.168.0.1: icmp_seq=4 ttl=64 time=1878 ms
64 bytes from 192.168.0.1: icmp_seq=5 ttl=64 time=855 ms
64 bytes from 192.168.0.1: icmp_seq=6 ttl=64 time=5.94 ms
64 bytes from 192.168.0.1: icmp_seq=7 ttl=64 time=6.25 ms
64 bytes from 192.168.0.1: icmp_seq=8 ttl=64 time=6.45 ms
64 bytes from 192.168.0.1: icmp_seq=9 ttl=64 time=4.88 ms
64 bytes from 192.168.0.1: icmp_seq=10 ttl=64 time=4.67 ms
64 bytes from 192.168.0.1: icmp_seq=11 ttl=64 time=3925 ms
64 bytes from 192.168.0.1: icmp_seq=12 ttl=64 time=2893 ms
64 bytes from 192.168.0.1: icmp_seq=13 ttl=64 time=1869 ms
64 bytes from 192.168.0.1: icmp_seq=14 ttl=64 time=845 ms
64 bytes from 192.168.0.1: icmp_seq=15 ttl=64 time=4.56 ms
64 bytes from 192.168.0.1: icmp_seq=16 ttl=64 time=4.33 ms
64 bytes from 192.168.0.1: icmp_seq=17 ttl=64 time=6.42 ms
64 bytes from 192.168.0.1: icmp_seq=18 ttl=64 time=6.44 ms
64 bytes from 192.168.0.1: icmp_seq=19 ttl=64 time=4.12 ms
64 bytes from 192.168.0.1: icmp_seq=20 ttl=64 time=3900 ms
64 bytes from 192.168.0.1: icmp_seq=21 ttl=64 time=2868 ms
64 bytes from 192.168.0.1: icmp_seq=22 ttl=64 time=1846 ms
64 bytes from 192.168.0.1: icmp_seq=23 ttl=64 time=822 ms
64 bytes from 192.168.0.1: icmp_seq=24 ttl=64 time=5.94 ms
64 bytes from 192.168.0.1: icmp_seq=25 ttl=64 time=4.12 ms
```

As @morrownr suggested in this thread I have uninstalled the driver, removed and cloned again the folder. The installation proceeds without error but the problem persists.

If there are any more steps I can take to help debugging please let me know.

Thanks!

michelwi, 2024-01-08:
Well shiver me timbers.. the things you notice after spending 1h painstakingly preparing a bug report in a Github issue.. 

> > Port 3: Dev 2, If 0, Class=Vendor Specific Class, Driver=rtl8812au, 480M
>
> You are running is USB2 mode. Is that what you want?

Mh.. I also seem to be running in USB2 mode, but I distinctly remember setting `rtw_switch_usb_mode=1` and I distinctly remember selecting a USB3 extension cord and plugging it into an USB3 Port.

After a round of switching extension cables and USB ports on my motherboard I found that I have 2 USB ports which are blue and according to the documentation of my motherboard _should_ be USB3, but apparently are not.

If I plug into the designated USB2 ports, I get USB2 speed and no latency.
If I plug into a different set of USB3 ports, I get USB3 speed and no latency.
If I plug into the easier to reach "USB3" ports, I get sometimes USB3, sometimes USB2, but always the latency spikes.
And for completeness, even the front USB2 does work.

So what do we learn? RTFM
> Moving your USB WiFi adapter to a different USB port has been known to fix a variety of problems.

but also do not trust what you read in your manual. The motherboard is a MSI Z170-A PRO and the upper USB3 ports were the culprit.

dubhater, 2024-01-08:
Did you try the ``rtw_lps_level=0`` module parameter?

morrownr, 2024-01-10:
> Well shiver me timbers.. the things you notice after spending 1h painstakingly preparing a bug report in a Github issue.

Frequently used words.

> So what do we learn? RTFM

>> Moving your USB WiFi adapter to a different USB port has been known to fix a variety of problems.

I added that statement for many good reasons. I've seen many problems with USB3 capable USB WiFi adapters over the years. Why?

- USB3 is not mankind's greatest invention. USB2 was pretty good. 
- Many computers have one or more USB3 ports that just don't seem to work right.
- The Linux USB3 drivers are less than perfect.
- Manufacturing issues with the usb ports in the computer and the ports on the adapter contribute as well.
- Wear and tear on the ports from a lot of use.

You can check which mode the adapter is in with:

$ lsusb -t

 > Did you try the rtw_lps_level=0 module parameter?

I've never seen a problem fixed by modifying power saving on this driver.

I am working on a replacement for this driver in a private repo. Let me know if you are interested in helping test.

@morrownr

castillofrancodamian, 2024-01-10:
> Frequently used words.
>
>
>
> I added that statement for many good reasons. I've seen many problems with USB3 capable USB WiFi adapters over the years. Why?
>
> * USB3 is not mankind's greatest invention. [...]

I'm sorry to say that I tried that "new" driver and the problems continue.

michelwi, 2024-01-10:
> @michelwi I really took that question wrong, because it gives you an answer without really knowing if it is the cause of the problem.

I would say in my case the question lead to investigating in a different angle which eventually uncovered faulty (or incompatible) ports. The exact cause and chain of coincidences remains unknown but having a working solution shall suffice for me.

> I am working on a replacement for this driver in a private repo. Let me know if you are interested in helping test.

I might not be able to guarantee the fastest turnaround times, but I am happy to donate some time to improve this driver. Let me post in the other issue then...

michelwi, 2024-01-15:
Mh.. It seems I have to back-paddle on my back-paddeling: After rebooting my problem reappears independent of the port used. Additional curiosity:
* after a reboot, I get ping latency spikes and this output:
```
user@host:~$ ip a
1: lo: [...]
2: enp2s0: [...]
[...]
3: wlx048d392a5ed7: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether [MY MAC ADDRESS] brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.10/24 brd 192.168.0.255 scope global dynamic noprefixroute wlx048d392a5ed7
       valid_lft 604087sec preferred_lft 604087sec
[...]
4: virbr0:  [...]
5: lxcbr0: [...]
```
* when I then unplug and replug the USB dongle (same or different port), the spikes are gone and
```
user@host:~$ ip a
1: lo: [...]
2: enp2s0: [...]
4: virbr0: [...]
5: lxcbr0: [...]
6: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether [MY MAC ADDRESS] brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.10/24 brd 192.168.0.255 scope global noprefixroute wlan0
       valid_lft forever preferred_lft forever
```
I think there is something very amiss with my configuration.

dubhater, 2024-01-15:
Two conflicting daemons active at the same time?

castillofrancodamian, 2024-01-22:
@dubhater Who was the question for?

michelwi, 2024-01-22:
I think this was a reply to my observations in the comment immediately prior (i.e. question for me).

Unfortunately I did not have much time to debug this further. I was afraid my install of openmediavault (which comes with a lot of additional packages and network management stuff) might have messed up, but I also observe this renaming on a fresh install of debian 12.

dubhater, 2024-01-22:
@castillofrancodamian It was a guess about @michelwi 's guess:

> I think there is something very amiss with my configuration.

alvergallano, 2024-02-03:
hi i have TP-Link Archer T4U v2 Realtek RTL8812AU
have same problem with this 
i try multiple distro and [Aircrack version](https://github.com/aircrack-ng/rtl8812au) still  same problem with these latency spikes that @castillofrancodamian,.
but when i try this [OpenHD version](https://github.com/OpenHD/rtl8812au) the latency spikes are gone

dubhater, 2024-04-24:
This makes the driver report the current TX rate: [tx-rate.txt](https://github.com/morrownr/8812au-20210629/files/15101160/tx-rate.txt)

It's ugly and it's not correct for AP mode. Also I forgot about the channel width. Anyway, that's approximately what the driver needs to do. Realtek simply did not implement it.

castillofrancodamian, 2024-05-08:
> This makes the driver report the current TX rate: [tx-rate.txt](https://github.com/morrownr/8812au-20210629/files/15101160/tx-rate.txt)
>
> It's ugly and it's not correct for AP mode. Also I forgot [...]

And how do I prove that?

dubhater, 2024-05-08:
You apply it with ``patch -p1 -i tx-rate.txt``. Then you run make again.

morrownr, 2024-05-10:
@castillofrancodamian 

Would you prefer that I apply the patch to the repo?  and then all you have to do is;

$ git pull
$ sudo sh install-driver.sh