morrownr/8812au-20210629 #56: Driver appears to install correctly but is not functional in network manager, and I think doest show up with lsmod.

LargePrime, 2022-06-06:
I thank you for your patience. Artic LXqt openRC Linux 5.18.0-artix1-1

sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev

Linux 5.18.0-artix1-1 #1 SMP PREEMPT_DYNAMIC Thu, 26 May 2022 13:26:58 +0000 x86_64 GNU/Linux
bash: mokutil: command not found
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 009 Device 004: ID 13fe:5500 Kingston Technology Company Inc. Flash drive
Bus 009 Device 003: ID 0781:5583 SanDisk Corp. Ultra Fit
Bus 009 Device 002: ID 2109:0817 VIA Labs, Inc. USB3.0 Hub             
Bus 009 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

Bus 008 Device 005: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
^, right?

0: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
2: phy1: Wireless LAN
        Soft blocked: no
        Hard blocked: no

rtl8812au/5.13.6, 5.18.0-artix1-1, x86_64: installed

phy#1
        Interface wlan1
                ifindex 7
                wdev 0x100000002
                addr 9a:20:5c:d8:f1:1f
                type managed
                txpower -100.00 dBm
        Interface wlan0
                ifindex 6
                wdev 0x100000001
                addr 52:1c:c7:53:0a:35
                type managed
                txpower -100.00 dBm

has wan interface, right?

lsmod

Module                  Size  Used by
tun                    61440  2
rfcomm                 94208  4
snd_seq_dummy          16384  0
snd_hrtimer            16384  1
snd_seq                94208  7 snd_seq_dummy
snd_seq_device         16384  1 snd_seq
fuse                  176128  6
qrtr                   57344  4
cmac                   16384  2
algif_hash             16384  1
algif_skcipher         16384  1
af_alg                 36864  6 algif_hash,algif_skcipher
bnep                   32768  2
edac_mce_amd           57344  0
kvm_amd               167936  0
ccp                   131072  1 kvm_amd
rng_core               20480  1 ccp
kvm                  1118208  1 kvm_amd
irqbypass              16384  1 kvm
vfat                   24576  3
fat                    90112  1 vfat
amdgpu               8941568  30
crct10dif_pclmul       16384  1
snd_hda_codec_realtek   167936  1
crc32_pclmul           16384  0
snd_hda_codec_generic    98304  1 snd_hda_codec_realtek
btusb                  65536  0
snd_hda_codec_hdmi     86016  1
ledtrig_audio          16384  1 snd_hda_codec_generic
ghash_clmulni_intel    16384  0
btrtl                  28672  1 btusb
btbcm                  20480  1 btusb
8812au               3039232  0
mxm_wmi                16384  0
snd_hda_intel          61440  4
btintel                49152  1 btusb                                                                                                                                                                              
snd_intel_dspcfg       36864  1 snd_hda_intel                                                                                                                                                                      
aesni_intel           380928  3                                                                                                                                                                                    
btmtk                  16384  1 btusb                                                                                                                                                                              
snd_intel_sdw_acpi     20480  1 snd_intel_dspcfg                                                                                                                                                                   
crypto_simd            16384  1 aesni_intel
snd_hda_codec         184320  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
cryptd                 28672  3 crypto_simd,ghash_clmulni_intel
joydev                 28672  0
r8169                 106496  0
bluetooth             872448  36 btrtl,btmtk,btintel,btbcm,bnep,btusb,rfcomm
k10temp                16384  0
snd_hda_core          118784  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
mousedev               24576  0
fam15h_power           16384  0
pcspkr                 16384  0
realtek                36864  1
mdio_devres            16384  1 r8169
cfg80211             1085440  1 8812au
snd_hwdep              16384  1 snd_hda_codec
libphy                172032  3 r8169,mdio_devres,realtek
sp5100_tco             20480  0
snd_pcm               172032  5 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_core
ecdh_generic           16384  1 bluetooth
i2c_piix4              36864  0
rfkill                 36864  5 bluetooth,cfg80211
snd_timer              49152  3 snd_seq,snd_hrtimer,snd_pcm
gpu_sched              53248  1 amdgpu
drm_ttm_helper         16384  1 amdgpu
snd                   126976  19 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_pcm
ttm                    86016  2 amdgpu,drm_ttm_helper
soundcore              16384  1 snd
drm_dp_helper         167936  1 amdgpu
mac_hid                16384  0
wmi                    45056  1 mxm_wmi
acpi_cpufreq           32768  0
crypto_user            24576  0
ext4                 1011712  3
crc32c_generic         16384  0
crc16                  16384  2 bluetooth,ext4
mbcache                16384  1 ext4
jbd2                  192512  1 ext4
uas                    32768  0
usb_storage            81920  5 uas
hid_elecom             16384  0
crc32c_intel           24576  6
xhci_pci               20480  0
usbhid                 73728  0
xhci_pci_renesas       24576  1 xhci_pci

I thank you for your time.

LargePrime, 2022-06-10:
update

[2022-06-08T11:16:31-0400] [ALPM-SCRIPTLET] ==> dkms install --no-depmod rtl8812au/5.13.6.r119.gedb17ca -k 5.18.2-artix1-1
[2022-06-08T11:16:32-0400] [ALPM-SCRIPTLET] Error! Bad return status for module build on kernel: 5.18.2-artix1-1 (x86_64)
[2022-06-08T11:16:32-0400] [ALPM-SCRIPTLET] Consult /var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/make.log for more information.
[2022-06-08T11:16:32-0400] [ALPM-SCRIPTLET] ==> WARNING: `dkms install --no-depmod rtl8812au/5.13.6.r119.gedb17ca -k 5.18.2-artix1-1' exited 10
[2022-06-08T11:16:32-0400] [ALPM-SCRIPTLET] ==> dkms install --no-depmod rtl8812au/5.13.6 -k 5.18.2-artix1-1
[2022-06-08T11:16:33-0400] [ALPM-SCRIPTLET] Error! Bad return status for module build on kernel: 5.18.2-artix1-1 (x86_64)
[2022-06-08T11:16:33-0400] [ALPM-SCRIPTLET] Consult /var/lib/dkms/rtl8812au/5.13.6/build/make.log for more information.
[2022-06-08T11:16:33-0400] [ALPM-SCRIPTLET] ==> WARNING: `dkms install --no-depmod rtl8812au/5.13.6 -k 5.18.2-artix1-1' exited 10

 /var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/make.log

DKMS make.log for rtl8812au-5.13.6.r119.gedb17ca for kernel 5.18.2-artix1-1 (x86_64)
Wed Jun  8 11:16:32 EDT 2022
make: uname: No such file or directory
make: pwd: No such file or directory
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.18.2-artix1-1/build M=  modules
make: pwd: No such file or directory
make: make: No such file or directory
make: *** [Makefile:2514: modules] Error 127

I am not sure, but is this a driver issue?

Full ./remove and ./install got it working it seems.

LargePrime, 2022-06-10:
>You are right on the bleeding edge with kernel 5.18. It should be working but only minimal testing has been accomplished so far.

It works.

>I am not familiar with your Linux distro. What distro is it based on? Debian, Arch, Slackware, Red Hat, ?

Arch, no systemd see https://artixlinux.org/

>Information above indicates your system is based on a X86_64 processor. Can you confirm that?
Can Confirm

>What results do you get if you go into the driver directory and run the following?

Please be precise as to what directory you would like the makes ran in?

LargePrime, 2022-06-10:
/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build$ make clean
#make -C /lib/modules/5.18.2-artix1-1/build M=/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions

/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build$ make

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.18.2-artix1-1/build M=/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build  modules
make[1]: Entering directory '/usr/lib/modules/5.18.2-artix1-1/build'
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13300’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13302’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13304’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13306’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13308’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13310’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13312’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13314’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13316’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13318’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13320’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13322’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13324’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13326’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13328’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13330’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13332’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13334’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13336’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13338’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13340’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13342’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13344’: Permission denied
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13346’: Permission denied                                                                                                  
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13348’: Permission denied                                                                                                  
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13351’: Permission denied                                                                                                  
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13353’: Permission denied                                                                                                  
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13355’: Permission denied                                                                                                  
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13357’: Permission denied                                                                                                  
mkdir: cannot create directory ‘/var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/.tmp_13359’: Permission denied                                                                                                  
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/core/rtw_cmd.o
In file included from ./arch/x86/include/asm/linkage.h:6,
                 from ./include/linux/linkage.h:8,
                 from ./arch/x86/include/asm/cache.h:5,
                 from ./include/linux/cache.h:6,
[... cut ...]
  AR [M]  /var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/8812au.o
  LD [M]  /var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/8812au.prelink.o
  MODPOST /var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/Module.symvers
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/8812au.mod.o
  LD [M]  /var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/8812au.ko
  BTF [M] /var/lib/dkms/rtl8812au/5.13.6.r119.gedb17ca/build/8812au.ko
make[1]: Leaving directory '/usr/lib/modules/5.18.2-artix1-1/build'

LargePrime, 2022-06-11:
@morrownr

morrownr, 2022-06-13:
Hi @LargePrime 

I'm here but busy. 

When I asked you to run `$ make clean` and `$ make`, I had assumed you would run it is the default directory as is used in the installation instructions: `~/src/8812au-20210629`

You probably should run the following to clean things up and then start over:

$ sudo ./remove-driver.sh
$ sudo ./remove-driver-no-dkms.sh

Regards

LargePrime, 2022-06-14:
Perhaps you can add tested on Artix Linux.

Or would you rather further testing?

LargePrime, 2022-06-15:
I believe there was an update to networkmanager and the associated applet, that fixed any issues I had that I guessed were a driver issue.

Anyway, those updated, removed and reinstalled the driver, and away it went.

When I get a chance to do a clean install to test I will post a new issue.

morrownr, 2022-06-15:
Copy all. I look forward to hearing from you.

Regards