morrownr/8812au-20210629 #57: ibss support

smydoudou, 2022-06-15:
Hi,
I'm using 2 nvidia tx2 boards for wireless module. After installing rtl8812au driver, I checked normal wifi connection is no problem. But in ad-hoc mode the boards can't communicate each other even though they are in same essid, cell, frequency... 
Is this a known issue? 
Thanks a lot.

morrownr, 2022-06-15:
Hi,

> Is this a known issue?

No. However, that does not mean that it is working. This driver is very heavily used but I haven't had any reports of ad-hoc use and it is not a mode that I test. I can try to help as I have time but right now I have several things I am working on. Let me know if you make any progress.

Regards

smydoudou, 2022-06-18:
> 

Thank you for your reply. I will continue test or find other solution in my project.

closed 2022-12-31 by morrownr