morrownr/8812au-20210629 #61: Does not work on Raspberry Pi

lgbelowbridge, 2022-06-26:
>  uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev
Linux presenter1 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux
bash: mokutil: command not found
Bus 001 Device 004: ID 0b05:17d2 ASUSTek Computer, Inc. USB-AC56 802.11a/b/g/n/ac Wireless Adapter [Realtek RTL8812AU]
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
rtl8812au, 5.13.6, 5.10.103-v7+, armv7l: installed (original_module exists)
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 3c:7c:3f:a2:4f:0e
                type managed
                txpower 42949572.96 dBm

Excerpt from /var/log/messages
Jun 26 10:48:22 presenter1 kernel: [83815.799453] ------------[ cut here ]------------
Jun 26 10:48:22 presenter1 kernel: [83815.800139] WARNING: CPU: 3 PID: 434 at /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ieee80211.c:2017 rtw_bss_get_chbw+0xa8/0xcc [8812au]
Jun 26 10:48:22 presenter1 kernel: [83815.800155] Modules linked in: dm_mod fuse binfmt_misc joydev 8812au(O) sha256_generic cfg80211 rfkill raspberrypi_ts raspberrypi_hwmon bcm2835_isp(C) bcm2835_v4l2(C) bcm2835_codec(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev vc_sm_cma(C) mc rpi_backlight backlight uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6
Jun 26 10:48:22 presenter1 kernel: [83815.800531] CPU: 3 PID: 434 Comm: RTW_CMD_THREAD Tainted: G        WC O      5.10.103-v7+ #1529
Jun 26 10:48:22 presenter1 kernel: [83815.800539] Hardware name: BCM2835
Jun 26 10:48:22 presenter1 kernel: [83815.800547] Backtrace:
Jun 26 10:48:22 presenter1 kernel: [83815.800579] [<809f41d4>] (dump_backtrace) from [<809f4564>] (show_stack+0x20/0x24)
Jun 26 10:48:22 presenter1 kernel: [83815.800596]  r7:ffffffff r6:00000000 r5:60000013 r4:80fe5ef0
Jun 26 10:48:22 presenter1 kernel: [83815.800615] [<809f4544>] (show_stack) from [<809f8770>] (dump_stack+0xcc/0xf8)
Jun 26 10:48:22 presenter1 kernel: [83815.800636] [<809f86a4>] (dump_stack) from [<8011eddc>] (__warn+0xfc/0x114)
Jun 26 10:48:22 presenter1 kernel: [83815.800653]  r10:7f4973ac r9:00000009 r8:7f344aa4 r7:000007e1 r6:00000009 r5:7f344aa4
Jun 26 10:48:22 presenter1 kernel: [83815.800663]  r4:7f468310 r3:80f05050
Jun 26 10:48:22 presenter1 kernel: [83815.800680] [<8011ece0>] (__warn) from [<809f4bb0>] (warn_slowpath_fmt+0x70/0xd8)
Jun 26 10:48:22 presenter1 kernel: [83815.800694]  r7:000007e1 r6:7f468310 r5:80f05008 r4:00000000
Jun 26 10:48:22 presenter1 kernel: [83815.801335] [<809f4b44>] (warn_slowpath_fmt) from [<7f344aa4>] (rtw_bss_get_chbw+0xa8/0xcc [8812au])
Jun 26 10:48:22 presenter1 kernel: [83815.801354]  r9:00000000 r8:bb8c6b6e r7:bb8c6c71 r6:bb8c66fa r5:bb8c66f8 r4:bb8c6b6e
Jun 26 10:48:22 presenter1 kernel: [83815.802613] [<7f3449fc>] (rtw_bss_get_chbw [8812au]) from [<7f365a50>] (join_cmd_hdl+0x230/0x484 [8812au])
Jun 26 10:48:22 presenter1 kernel: [83815.802630]  r5:bb8c66f9 r4:bb8c6000
Jun 26 10:48:22 presenter1 kernel: [83815.803854] [<7f365820>] (join_cmd_hdl [8812au]) from [<7f3280b0>] (rtw_cmd_thread+0x498/0x68c [8812au])
Jun 26 10:48:22 presenter1 kernel: [83815.803878]  r10:7f4d1058 r9:7f497000 r8:bb8c6f88 r7:7f461508 r6:bb8c6fa8 r5:a2dff3c0
Jun 26 10:48:22 presenter1 kernel: [83815.803888]  r4:bb8c6000
Jun 26 10:48:22 presenter1 kernel: [83815.804506] [<7f327c18>] (rtw_cmd_thread [8812au]) from [<801439c4>] (kthread+0x170/0x174)
Jun 26 10:48:22 presenter1 kernel: [83815.804528]  r10:84c79c3c r9:bb8c6000 r8:7f327c18 r7:86468000 r6:00000000 r5:858e1880
Jun 26 10:48:22 presenter1 kernel: [83815.804537]  r4:859678c0
Jun 26 10:48:22 presenter1 kernel: [83815.804556] [<80143854>] (kthread) from [<801000ec>] (ret_from_fork+0x14/0x28)
Jun 26 10:48:22 presenter1 kernel: [83815.804566] Exception stack(0x86469fb0 to 0x86469ff8)
Jun 26 10:48:22 presenter1 kernel: [83815.804580] 9fa0:                                     00000000 00000000 00000000 00000000
Jun 26 10:48:22 presenter1 kernel: [83815.804595] 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Jun 26 10:48:22 presenter1 kernel: [83815.804608] 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Jun 26 10:48:22 presenter1 kernel: [83815.804624]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:80143854
Jun 26 10:48:22 presenter1 kernel: [83815.804633]  r4:858e1880
Jun 26 10:48:22 presenter1 kernel: [83815.804645] ---[ end trace b4beed1d438b18d9 ]---

I can build it. It is installed using dkms. But do not work.

galaris, 2022-07-04:
I think I have the same issue https://github.com/morrownr/8812au-20210629/issues/63

lgbelowbridge, 2022-07-04:
I am really sorry. But I looked deeper into this and found the culprit. An old driver which had not been removed by DKMS, when removing old drivers.

used the command: find /lib/modules/5.10.103-v7+/ -name "*88*" 
And found an old 8812 module.

closed 2022-07-04 by lgbelowbridge