morrownr/8812au-20210629 #62: Compile error with Raspberry Pi 5.10

CharlesButcher, 2022-07-02:
This is on a Pi Zero W with Buster (5.10). Am I doing something stupid?

```
DKMS make.log for rtl8812au-5.13.6 for kernel 5.10.103+ (armv6l)
Sat  2 Jul 20:19:32 BST 2022
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.10.103+/build M=/var/lib/dkms/rtl8812au/5.13.6/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.103+'
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_cmd.o
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/Makefile.build:280: /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1825: /var/lib/dkms/rtl8812au/5.13.6/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.103+'
make: *** [Makefile:2514: modules] Error 2
```

morrownr, 2022-07-02:
Hi @CharlesButcher 

> This is on a Pi Zero W with Buster (5.10). Am I doing something stupid?

It can be easy to confuse 32 bit vs 64 bit and I think that is what is going on here, I think you are running the 32 bit version of RasPiOS but you ran the 64 bit setup script during installation. If that is the case, run the following from the driver directory:

$ ./ARM_RPI.sh

Then run the installation script again:

$ sudo ./install-driver.sh

Let me know if that works. It should take some time for this driver to compile on that hardware so don't get in a hurry.

FYI: The Main Menu here at this site is:

https://github.com/morrownr/USB-WiFi

I post this as a lot of users find that using adapters that use in-kernel drivers work out much better in use cases similar to yours. You can find information along those lines in Main Menu item 2.

Regards

CharlesButcher, 2022-07-02:
@morrownr Thank you for such a prompt and tactful reply. If you listen hard you will hear me slapping my head. My only excuse is that I've just set up a Pi 4 with the 64-bit Raspberry OS.

I'm staggered by the quality and scope of this project – not just the code but your documentation too. I'm sure you're right that buying the right adaptor in the first place saves a lot of hassle, and now that I've seen a proper and up-to-date list I'll do that.

But for the moment I will persevere with the Realtek. I already had it, I just need a bit of extra range, and it fits into the sandwich box that houses the hedgehogcam. I'm now 15 minutes into the build and everything looks fine :-)

closed 2022-07-02 by CharlesButcher

morrownr, 2022-07-03:
@CharlesButcher 

Gla everything is working out. Come back anytime. If you ever find anything here that you would like to work on, let me know. I have several folks that kinda adopt parts of some repos and it really takes the load off of me.

The exciting thing that is coming to the world of usb wifi soon is the new WiFi 6 adapters based on the mt7921u chipset. And for the first time ever for a usb wifi adapter in Linux, the driver is already in the kernel waiting on the adapters to be available.

Regards