morrownr/8812au-20210629 #63: Raspberry PI 4 with Kismet is getting errors

galaris, 2022-07-04:
I noticed some messages in dmesg:

```
[   15.808667] Bluetooth: RFCOMM ver 1.11
[   31.709798] cam-dummy-reg: disabling
[   59.467257] ------------[ cut here ]------------
[   59.467281] WARNING: CPU: 2 PID: 663 at /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme.c:3475 rtw_sta_mstatus_report+0x80/0xd8 [8812au]
[   59.467608] Modules linked in: rfcomm cmac algif_hash aes_arm64 algif_skcipher af_alg bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc snd_soc_hdmi_codec 8812au(O) brcmfmac brcmutil v3d gpu_sched cfg80211 vc4 rfkill bcm2835_codec(C) cec bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) drm_kms_helper videobuf2_vmalloc v4l2_mem2mem raspberrypi_hwmon videobuf2_dma_contig videobuf2_memops snd_soc_core videobuf2_v4l2 vc_sm_cma(C) snd_compress snd_bcm2835(C) videobuf2_common snd_pcm_dmaengine i2c_brcmstb videodev mc snd_pcm rpivid_mem snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops uio_pdrv_genirq uio nvmem_rmem i2c_dev drm drm_panel_orientation_quirks backlight fuse ip_tables x_tables ipv6
[   59.467928] CPU: 2 PID: 663 Comm: RTW_CMD_THREAD Tainted: G        WC O      5.15.32-v8+ #1538
[   59.467943] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[   59.467952] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   59.467965] pc : rtw_sta_mstatus_report+0x80/0xd8 [8812au]
[   59.468228] lr : rtw_sta_mstatus_report+0x5c/0xd8 [8812au]
[   59.468484] sp : ffffffc00a773d70
[   59.468491] x29: ffffffc00a773d90 x28: ffffff8042386a00 x27: ffffffc009da21bc
[   59.468515] x26: ffffffc0012fd0f0 x25: 0000000000001180 x24: ffffffc009da2140
[   59.468537] x23: 00000000ffff151b x22: ffffffc001438980 x21: ffffff8048c1e000
[   59.468558] x20: ffffff8042386a00 x19: ffffffc009da1000 x18: 0000000000000000
[   59.468578] x17: 0000000000000000 x16: 0000000000000000 x15: 0000005577696610
[   59.468599] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[   59.468620] x11: 0000000000000000 x10: 00000000000019d0 x9 : ffffffc008087d48
[   59.468641] x8 : 00000000000001f4 x7 : ffffffffffd5c295 x6 : ffffffc00a773c70
[   59.468661] x5 : 0000000000000001 x4 : 0000000000000000 x3 : ffffffc009319000
[   59.468681] x2 : ffffffc009319238 x1 : 0000000100000000 x0 : 0000000000000000
[   59.468703] Call trace:
[   59.468710]  rtw_sta_mstatus_report+0x80/0xd8 [8812au]
[   59.468968]  disconnect_hdl+0x54/0xc0 [8812au]
[   59.469224]  rtw_cmd_thread+0x474/0x648 [8812au]
[   59.469480]  kthread+0x140/0x158
[   59.469495]  ret_from_fork+0x10/0x20
[   59.469510] ---[ end trace f9bd759fa51fe430 ]---
[   83.652664] ------------[ cut here ]------------
[   83.652728] WARNING: CPU: 1 PID: 663 at /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme.c:3475 rtw_sta_mstatus_report+0x80/0xd8 [8812au]
[   83.652902] Modules linked in: rfcomm cmac algif_hash aes_arm64 algif_skcipher af_alg bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc snd_soc_hdmi_codec 8812au(O) brcmfmac brcmutil v3d gpu_sched cfg80211 vc4 rfkill bcm2835_codec(C) cec bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) drm_kms_helper videobuf2_vmalloc v4l2_mem2mem raspberrypi_hwmon videobuf2_dma_contig videobuf2_memops snd_soc_core videobuf2_v4l2 vc_sm_cma(C) snd_compress snd_bcm2835(C) videobuf2_common snd_pcm_dmaengine i2c_brcmstb videodev mc snd_pcm rpivid_mem snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops uio_pdrv_genirq uio nvmem_rmem i2c_dev drm drm_panel_orientation_quirks backlight fuse ip_tables x_tables ipv6
[   83.653021] CPU: 1 PID: 663 Comm: RTW_CMD_THREAD Tainted: G        WC O      5.15.32-v8+ #1538
[   83.653027] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[   83.653031] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   83.653036] pc : rtw_sta_mstatus_report+0x80/0xd8 [8812au]
[   83.653130] lr : rtw_sta_mstatus_report+0x5c/0xd8 [8812au]
[   83.653216] sp : ffffffc00a773d70
[   83.653218] x29: ffffffc00a773d90 x28: ffffff8042386a00 x27: ffffffc009da21bc
[   83.653226] x26: ffffffc0012fd0f0 x25: 0000000000001180 x24: ffffffc009da2140
[   83.653234] x23: 00000000ffff2cb9 x22: ffffffc001438980 x21: ffffff8048c1e000
[   83.653241] x20: ffffff8042386a00 x19: ffffffc009da1000 x18: 0000000000000000
[   83.653248] x17: 0000000000000000 x16: 0000000000000000 x15: 0000007ff8c35300
[   83.653255] x14: ffffffffffffffff x13: 0000000000000028 x12: 0101010101010101
[   83.653262] x11: ffffffc0093a2138 x10: 00000000000019d0 x9 : ffffffc008087d48
[   83.653269] x8 : 00000000000001f4 x7 : ffffff80fb7a5bb0 x6 : ffffffc00a773c70
[   83.653276] x5 : 0000000000000001 x4 : 0000000000000000 x3 : ffffffc009319000
[   83.653283] x2 : ffffffc009319238 x1 : 0000000100000000 x0 : 0000000000000000
[   83.653290] Call trace:
[   83.653293]  rtw_sta_mstatus_report+0x80/0xd8 [8812au]
[   83.653387]  disconnect_hdl+0x54/0xc0 [8812au]
[   83.653473]  rtw_cmd_thread+0x474/0x648 [8812au]
[   83.653559]  kthread+0x140/0x158
[   83.653566]  ret_from_fork+0x10/0x20
[   83.653573] ---[ end trace f9bd759fa51fe431 ]---
[  107.820917] ------------[ cut here ]------------
[  107.820939] WARNING: CPU: 0 PID: 663 at /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme.c:3475 rtw_sta_mstatus_report+0x80/0xd8 [8812au]
[  107.821276] Modules linked in: rfcomm cmac algif_hash aes_arm64 algif_skcipher af_alg bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc snd_soc_hdmi_codec 8812au(O) brcmfmac brcmutil v3d gpu_sched cfg80211 vc4 rfkill bcm2835_codec(C) cec bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) drm_kms_helper videobuf2_vmalloc v4l2_mem2mem raspberrypi_hwmon videobuf2_dma_contig videobuf2_memops snd_soc_core videobuf2_v4l2 vc_sm_cma(C) snd_compress snd_bcm2835(C) videobuf2_common snd_pcm_dmaengine i2c_brcmstb videodev mc snd_pcm rpivid_mem snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops uio_pdrv_genirq uio nvmem_rmem i2c_dev drm drm_panel_orientation_quirks backlight fuse ip_tables x_tables ipv6
[  107.821552] CPU: 0 PID: 663 Comm: RTW_CMD_THREAD Tainted: G        WC O      5.15.32-v8+ #1538
[  107.821566] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[  107.821573] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  107.821585] pc : rtw_sta_mstatus_report+0x80/0xd8 [8812au]
[  107.821810] lr : rtw_sta_mstatus_report+0x5c/0xd8 [8812au]
[  107.822030] sp : ffffffc00a773d70
[  107.822035] x29: ffffffc00a773d90 x28: ffffff8042386a00 x27: ffffffc009da21bc
[  107.822056] x26: ffffffc0012fd0f0 x25: 0000000000001180 x24: ffffffc009da2140
[  107.822075] x23: 00000000ffff4453 x22: ffffffc001438980 x21: ffffff8048c1e000
[  107.822093] x20: ffffff8042386a00 x19: ffffffc009da1000 x18: 0000000000000000
[... 36 lines cut ...]
        Hard blocked: no
rtl8812au, 5.13.6, 5.15.32-v8+, aarch64: installed
phy#1
        Interface wlan1
                ifindex 4
                wdev 0x100000001
                addr 00:c0:cb:aa:ed:1d
                type managed
                channel 2 (2417 MHz), width: 20 MHz, center1: 2417 MHz
                txpower 14.00 dBm
phy#0
        Unnamed/non-netdev interface
                wdev 0x2
                addr e6:5f:01:5b:f5:3d
                type P2P-device
                txpower 31.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr e4:5f:01:8b:f5:5e
                ssid ******
                type managed
                channel 60 (5300 MHz), width: 80 MHz, center1: 5290 MHz
                txpower 31.00 dBm
```

morrownr, 2022-07-04:
Hi @galaris 

I'm seeing some things I am not familiar with:

Can you give me a short rundown of what Kismet is and what it does?

I don't recognize the distro. Can you provide some details?

What hardware are you using? Is it a RasPi4B?

Bartholomew42, 2022-07-10:
Hi there, 
Kismet is a packet sniffer that is included in Kali. See https://en.wikipedia.org/wiki/Kismet_(software) or https://www.kali.org/tools/kismet/ .
Galaris' dmesg contained the line `[   59.467943] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)`. 
Greetings!

morrownr, 2022-07-11:
@Bartholomew42 

Thanks for the help. So RasPi4B with Kali using Kismet.

Hmmm... working this is going to take more time than I have right now so maybe I can take a look at this in a couple of months if @galaris is still interested.

galaris, 2022-07-14:
Hey, sorry for the late response. First of all I appreciate all your work for the wifi interfaces you do :) So yeah, Kismet is a software package for WiFi scanning. Here I did not use kali, but this Ubuntu: ubuntu-21.10-preinstalled-server-arm64+raspi.img.xz
Kali was manually installed on Ubuntu.  I won't be able to detail how Kismet does what it does, but If I had to guess, it's the same way that Airmon and other tools do it, putting the interface in monitor mode and sniff traffic. If I can provide further infor please let me know what you need of me.

morrownr, 2022-12-31:
Finally getting a little time to look at Kismet. Anyone want to tell me what to do to duplicate the issues?

closed 2023-07-28 by morrownr