morrownr/8812au-20210629 #65: Linux verion 5.19.0+ hang and BUG: scheduling while atomic: swapper/1/0/0x00000504

ikegami-t, 2022-08-11:
At first kenel hang is cansed on Linux version 5.19.0+ after 8812au.ko loaded. (Note: I have fixed the build error on Linux version 5.19.0+ locally to build 8812au.kio.)
Then I changed the log level to 1: always to try to debug but the issue was not caused. (Note: Before the log leve change the issue was caused repeated every time.)
```c
CONFIG_RTW_LOG_LEVEL = 1
```
To make sure I check the dmesg log but the folloing error message was recoreded.
```
...
[ 2192.904736] RTW: rtw_ndev_init(wlan0) if1 mac_addr=60:84:bd:f0:58:b8
[ 2192.905124] usbcore: registered new interface driver rtl8812au
[ 2192.905127] RTW: module init ret=0
[ 2192.909604] rtl8812au 1-3:1.0 wlx6084bdf058b8: renamed from wlan0
[ 2194.493842] IPv6: ADDRCONF(NETDEV_CHANGE): wlx6084bdf058b8: link becomes ready
[ 2198.798658] RTW: rtw_set_802_11_connect(wlx6084bdf058b8)  fw_state=0x00000008
[ 2199.116467] RTW: start auth
[ 2199.118372] RTW: auth success, start assoc
[ 2199.126486] BUG: scheduling while atomic: swapper/1/0/0x00000504
[ 2199.126495] Modules linked in: 8812au(OE) uhid rfcomm nvme_fabrics cmac algif_hash algif_skcipher af_alg xt_CHECKSUM xt_MASQUERADE xt_conntrack ipt_REJECT nf_reject_ipv4 xt_tcpudp nft_compat nouveau nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables libcrc32c nfnetlink bridge stp llc snd_hda_codec_realtek snd_hda_codec_generic snd_hda_codec_hdmi ledtrig_audio iwlmvm snd_hda_intel bnep snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec intel_rapl_msr intel_rapl_common snd_usb_audio snd_hda_core video snd_usbmidi_lib snd_hwdep edac_mce_amd drm_ttm_helper mac80211 ttm snd_pcm drm_display_helper uvcvideo snd_seq_midi snd_seq_midi_event kvm_amd snd_rawmidi libarc4 videobuf2_vmalloc videobuf2_memops btusb kvm videobuf2_v4l2 snd_seq iwlwifi videobuf2_common cec btrtl btbcm crct10dif_pclmul ghash_clmulni_intel btintel videodev rc_core aesni_intel btmtk snd_seq_device drm_kms_helper snd_timer crypto_simd bluetooth nls_iso8859_1 cryptd cfg80211 snd mc fb_sys_fops joydev ccp
[ 2199.126915]  syscopyarea sysfillrect input_leds rapl sysimgblt ecdh_generic serio_raw wmi_bmof ecc k10temp mxm_wmi soundcore mac_hid sch_fq_codel ipmi_devintf ipmi_msghandler drm msr parport_pc ppdev lp parport ip_tables x_tables autofs4 hid_logitech_hidpp hid_logitech_dj hid_generic usbhid hid nvme igb nvme_core crc32_pclmul i2c_algo_bit ahci xhci_pci i2c_piix4 libahci dca xhci_pci_renesas wmi
[ 2199.127082] Preemption disabled at:
[ 2199.127085] [<ffffffffb58bed2b>] copy_process+0x12b/0x1c10
[ 2199.127101] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G           OE      5.19.0+ #4
[ 2199.127107] Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./X570 Taichi, BIOS P4.60 08/03/2021
[ 2199.127109] Call Trace:
[ 2199.127112]  <IRQ>
[ 2199.127116]  dump_stack_lvl+0x49/0x63
[ 2199.127122]  ? copy_process+0x12b/0x1c10
[ 2199.127126]  dump_stack+0x10/0x16
[ 2199.127129]  __schedule_bug.cold+0x81/0x92
[ 2199.127135]  __schedule+0xe6a/0x1470
[ 2199.127141]  ? __mod_timer+0x21d/0x470
[ 2199.127148]  schedule+0x6b/0x110
[ 2199.127152]  schedule_timeout+0x87/0x160
[ 2199.127156]  ? timer_migration_handler+0xa0/0xa0
[ 2199.127161]  wait_for_completion_timeout+0x7d/0x140
[ 2199.127166]  usb_start_wait_urb+0x8f/0x180
[ 2199.127172]  usb_control_msg+0xe3/0x150
[ 2199.127179]  usbctrl_vendorreq+0x355/0x42b [8812au]
[ 2199.127289]  usb_read8+0x3e/0x5c [8812au]
[ 2199.127389]  _rtw_read8+0x1f/0x25 [8812au]
[ 2199.127463]  SwLedOn_8812AU+0xf2/0x17e [8812au]
[ 2199.127556]  SwLedControlMode14+0xeb/0x1cb [8812au]
[ 2199.127646]  LedControlUSB+0x1f0f/0x1f14 [8812au]
[ 2199.127729]  rtw_led_control+0x79/0xc4 [8812au]
[ 2199.127812]  rtw_indicate_connect+0x39/0x6b [8812au]
[ 2199.127889]  rtw_joinbss_event_prehandle+0x2cf/0x795 [8812au]
[ 2199.127964]  report_join_res+0xcc/0x11d [8812au]
[ 2199.128025]  OnAssocRsp+0x1d8/0x317 [8812au]
[ 2199.128083]  _mgt_dispatcher+0x73/0xaf [8812au]
...
```
I am not sure if the error log is related the hang issue or not. (Note: The error log is only confirmed at the fist time after the log level change but not tested to repeat yet.)
If any advice or comment about the issue and error please let me know.
Note: Before I used rtl8812au-5.6.4.2 below also and some time found similar the hang issue but it was not repeated after reboot.
https://github.com/ikegami-t/rtl8812au-5.6.4.2
FYI: The local changes are below. (Note: I am thinking to creat a pull request but the changes will be updated.)
```diff
tokunori@tokunori-desktop:~/8812au-20210629$ git diff
diff --git a/Makefile b/Makefile
index c80b777..053f40b 100644
--- a/Makefile
+++ b/Makefile
@@ -111,7 +111,7 @@ endif
 CONFIG_RTW_DEBUG = y
 # default log level is _DRV_INFO_ = 4,
 # please refer to "How_to_set_driver_debug_log_level.doc" to set the available level.
-CONFIG_RTW_LOG_LEVEL = 0
+CONFIG_RTW_LOG_LEVEL = 1

 # enable /proc/net/rtlxxxx/ debug interfaces
 CONFIG_PROC_DEBUG = y
diff --git a/os_dep/linux/ioctl_cfg80211.c b/os_dep/linux/ioctl_cfg80211.c
index 791cd4c..03b9532 100644
--- a/os_dep/linux/ioctl_cfg80211.c
+++ b/os_dep/linux/ioctl_cfg80211.c
@@ -475,7 +475,7 @@ u8 rtw_cfg80211_ch_switch_notify(_adapter *adapter, u8 ch, u8 bw, u8 offset,
        if (!rtw_cfg80211_allow_ch_switch_notify(adapter))
                goto exit;

-       cfg80211_ch_switch_notify(adapter->pnetdev, &chdef);
+       cfg80211_ch_switch_notify(adapter->pnetdev, &chdef, 0);

 #else
        int freq = rtw_ch2freq(ch);
@@ -490,7 +490,7 @@ u8 rtw_cfg80211_ch_switch_notify(_adapter *adapter, u8 ch, u8 bw, u8 offset,
        }

        ctype = rtw_chbw_to_nl80211_channel_type(ch, bw, offset, ht);
-       cfg80211_ch_switch_notify(adapter->pnetdev, freq, ctype);
+       cfg80211_ch_switch_notify(adapter->pnetdev, freq, ctype, 0);
 #endif

 exit:
@@ -1139,7 +1139,7 @@ check_bss:
                #endif

                #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 12, 0)
-               roam_info.bssid = cur_network->network.MacAddress;
+               roam_info.links[0].bssid = cur_network->network.MacAddress;
                roam_info.req_ie = pmlmepriv->assoc_req + sizeof(struct rtw_ieee80211_hdr_3addr) + 2;
                roam_info.req_ie_len = pmlmepriv->assoc_req_len - sizeof(struct rtw_ieee80211_hdr_3addr) - 2;
                roam_info.resp_ie = pmlmepriv->assoc_rsp + sizeof(struct rtw_ieee80211_hdr_3addr) + 6;
@@ -5377,7 +5377,8 @@ static int cfg80211_rtw_change_beacon(struct wiphy *wiphy, struct net_device *nd
        return ret;
 }

-static int cfg80211_rtw_stop_ap(struct wiphy *wiphy, struct net_device *ndev)
+static int cfg80211_rtw_stop_ap(struct wiphy *wiphy, struct net_device *ndev,
+               unsigned int link_id)
 {
        _adapter *adapter = (_adapter *)rtw_netdev_priv(ndev);

@@ -6753,7 +6754,7 @@ exit:

 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 8, 0))
 static int cfg80211_rtw_get_channel(struct wiphy *wiphy,
-       struct wireless_dev *wdev,
+       struct wireless_dev *wdev, unsigned int link_id,
        struct cfg80211_chan_def *chandef)
 {
        _adapter *padapter = wiphy_to_adapter(wiphy);
@@ -10692,7 +10693,7 @@ void rtw_wdev_unregister(struct wireless_dev *wdev)
        rtw_cfg80211_indicate_scan_done(adapter, _TRUE);

        #if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 11, 0)) || defined(COMPAT_KERNEL_RELEASE)
-       if (wdev->current_bss) {
+       if (wdev->links[0].client.current_bss) {
                RTW_INFO(FUNC_ADPT_FMT" clear current_bss by cfg80211_disconnected\n", FUNC_ADPT_ARG(adapter));
                rtw_cfg80211_indicate_disconnect(adapter, 0, 1);
        }
```

morrownr, 2022-08-12:
Hi @ikegami-t 

Give me a chance to investigate this. You are welcome to make a PR but don't include the following:

+CONFIG_RTW_LOG_LEVEL = 1

I'd prefer to continue working that via module parameter.

Can you provide info about the distro, version and compiler version?

Nick

ikegami-t, 2022-08-12:
> Hi @ikegami-t 
> 
> 
> 
> Give me a chance to investigate this. You are welcome to make a PR but don't include the following:
> 
> 
> 
> +CONFIG_RTW_LOG_LEVEL = 1
Yes sure.
> 
> 
> 
> I'd prefer to continue working that via module parameter.
Thank you. I see.
> 
> 
> 
> Can you provide info about the distro, version and compiler version?
Using Ubuntu and both its and compiler versions are latest I think. For detail let me check and update later.

Regards,
Ikegami
> 
> 
> 
> Nick

morrownr, 2022-08-12:
Okay, so Ubuntu is the distro. Can you confirm if it is 22.04?

In the mean time, I will go ahead and run my Ubuntu based distro up to 5.19.1 and see what happens.

ikegami-t, 2022-08-13:
> Okay, so Ubuntu is the distro. Can you confirm if it is 22.04?

Yes I think so.

> 
> 
> 
> In the mean time, I will go ahead and run my Ubuntu based distro up to 5.19.1 and see what happens.

morrownr, 2022-08-13:
>> Okay, so Ubuntu is the distro. Can you confirm if it is 22.04?

> Yes I think so.

Can I get you confirm what version of Ubuntu you are using? It is important.

I just did a clean installation of the driver on my main dev box after running the kernel up to 5.19.1. The distro is Mint 21 which is based on Ubuntu 22.04. The compilation and operation of the driver is clean

iperf3 shows:

```
[  5]   8.00-9.00   sec  57.5 MBytes   482 Mbits/sec    0   1.84 MBytes       
[  5]   9.00-10.00  sec  57.5 MBytes   482 Mbits/sec    0   1.84 MBytes       
[  5]  10.00-11.00  sec  60.0 MBytes   503 Mbits/sec    0   1.84 MBytes       
[  5]  11.00-12.00  sec  58.8 MBytes   493 Mbits/sec    0   1.84 MBytes       
[  5]  12.00-13.00  sec  57.5 MBytes   482 Mbits/sec    0   1.84 MBytes       
```

Not seeing a problem here.

If your Ubuntu is 20.04 and you have run it up to kernel 5.19, that is highly unlikely work well.

ikegami-t, 2022-08-13:
> >> Okay, so Ubuntu is the distro. Can you confirm if it is 22.04?
> 
> 
> 
> > Yes I think so.
> 
> 
> 
> Can I get you confirm what version of Ubuntu you are using? It is important.

Some update or upgrade were done on Ubuntu 22.04 so I will check the detail minor version etc. later. Sorry I am traveling now so will do tomorrow.
> I just did a clean installation of the driver on my main dev box after running the kernel up to 5.19.1. The distro is Mint 21 which is based on Ubuntu 22.04. The compilation and operation of the [...]

I see so it may be depended on the CPU and motherboard.
> 
> 
> 
> If your Ubuntu is 20.04 and you have run it up to kernel 5.19, that is highly unlikely work well.

No actually I use Ubuntu 22.04 Jammy Jellyfish.

Regards,
Ikegami

morrownr, 2022-08-13:
> I see so it may be depended on the CPU and motherboard.

This driver defaults to compiling on Intel/AMD CPU's but can compile on ARM, ARM64 and others. Step 9 of the installation instructions has scripts for compiling for the RasPiOS but you should not run them if your CPU is not an ARM CPU.

ikegami-t, 2022-08-14:
Just confirmed the Ubuntu version as below.
<pre>
tokunori@tokunori-desktop:~$ cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04 LTS"
</pre>

Also to make sure attach the dmesg and cpuinfo log files.
[dmesg.txt](https://github.com/morrownr/8812au-20210629/files/9332816/dmesg.txt)
[cpuinfo.txt](https://github.com/morrownr/8812au-20210629/files/9332818/cpuinfo.txt)

The motherboard infomation is below as output by the dmidecode command.
<pre>
tokunori@tokunori-desktop:~$ sudo dmidecode
...
Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
        Manufacturer: ASRock
        Product Name: X570 Taichi
        Version:
        Serial Number:
        Asset Tag:
        Features:
                Board is a hosting board
                Board is replaceable
        Location In Chassis:
        Chassis Handle: 0x0003
        Type: Motherboard
        Contained Object Handles: 0
</pre>

morrownr, 2022-08-14:
Okay. So a 64 bit 16 core AMD processor. Ubuntu 22.04. Kernel 5.19.

My main dev box has a 64 bit 4 core Intel processor (about 10 years old). Mint 21. Kernel 5.19.1.

From this driver's perspective, these are basically the same systems but something that matters is different. If it were me, I would start by totally wiping any previous attempts to install this driver and simply start over. If you used the `install-driver.sh` script then use the `remove-driver.sh` script to remove the driver and then delete the driver directory. If you used other methods to install the driver then please use appropriate methods to get rid of any traces of the driver.

Once that is complete and you have rebooted, I think going through the installation very carefully is the best thing to do. I can provide exactly what commands you should run based on your setup. Let me know when you are ready.

Nick

P.S. If you have some extra time, a trip to the Main Menu here might be informative:

https://github.com/morrownr/USB-WiFi

ikegami-t, 2022-08-14:
Hi,

Thanks for the comments. I have confirmed as the hang issue is resolved after the remove-driver.sh execution so it seems that my procedure was not correct as the previous driver was not removed correctly. So let me close the issue.
Note: But the BUG error is still caused after the driver installed so it seems that the error is not related to the hang issue.
By the way I have just created the pull request #66 to fix the build error on 5.19.0+ so please review the changes.

closed 2022-08-14 by ikegami-t