morrownr/8812au-20210629 #67: kernel-6.0-rc1

petehg, 2022-08-17:
I cant compile this for Kernel-6.0 and once again I can't work OUT a patch. I keeep creating more errors.
With your original driver files this is the make.log:
 
DKMS make.log for rtl8812au-5.13.6 for kernel 6.0.0-rc1-3.g7bd57d5-default (x86_64)
Wed Aug 17 18:49:04 BST 2022
/bin/sh: line 1: bc: command not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.0.0-rc1-3.g7bd57d5-default/build M=/var/lib/dkms/rtl8812au/5.13.6/build  modules
make[1]: Entering directory '/usr/src/linux-6.0.0-rc1-3.g7bd57d5-obj/x86_64/default'
/bin/sh: line 1: bc: command not found
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/monitor/rtw_radiotap.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/wds/rtw_wds.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rson.o
/var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_sta_mgt.c: In function ‘rtw_mfree_stainfo’:
/var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_sta_mgt.c:391:25: warning: the comparison will always evaluate as ‘true’ for the address of ‘lock’ will never be NULL [-Waddress]
  391 |         if (&psta->lock != NULL)
      |                         ^~
In file included from /var/lib/dkms/rtl8812au/5.13.6/build/include/drv_types.h:112,
                 from /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_sta_mgt.c:17:
/var/lib/dkms/rtl8812au/5.13.6/build/include/sta_info.h:267:17: note: ‘lock’ declared here
  267 |         _lock   lock;
      |                 ^~~~
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ft.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_wnm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mbo.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rm_util.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_roch.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-internal.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-internal-enc.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-gcm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-ccm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-omac1.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/ccmp.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/gcmp.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-siv.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/aes-ctr.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/crypto/sha256-internal.o
[... 35 lines cut ...]
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c: At top level:
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c:10415:20: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, unsigned int)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *)’ [-Werror=incompatible-pointer-types]
10415 |         .stop_ap = cfg80211_rtw_stop_ap,
      |                    ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c:10415:20: note: (near initialization for ‘rtw_cfg80211_ops.stop_ap’)
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c:10459:24: error: initialization of ‘int (*)(struct wiphy *, struct wireless_dev *, unsigned int,  struct cfg80211_chan_def *)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct wireless_dev *, struct cfg80211_chan_def *)’ [-Werror=incompatible-pointer-types]
10459 |         .get_channel = cfg80211_rtw_get_channel,
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c:10459:24: note: (near initialization for ‘rtw_cfg80211_ops.get_channel’)
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_wdev_unregister’:
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c:10695:17: error: ‘struct wireless_dev’ has no member named ‘current_bss’
10695 |         if (wdev->current_bss) {
      |                 ^~
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_com.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_com_phycfg.o
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-6.0.0-rc1-3.g7bd57d5/scripts/Makefile.build:250: /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [../../../linux-6.0.0-rc1-3.g7bd57d5/Makefile:1869: /var/lib/dkms/rtl8812au/5.13.6/build] Error 2
make[1]: Leaving directory '/usr/src/linux-6.0.0-rc1-3.g7bd57d5-obj/x86_64/default'
make: *** [Makefile:2510: modules] Error 2

If you can find the time perhaps you can work this out. I'll be happy to test it again for you.
Thanks.
Pete

morrownr, 2022-08-18:
Hey Pete,

Actually I think @ikegami-t has already worked it out. See PR #66 .

Give me a chance to work this and test... this actually happened beginning with 5.19.2. I had been testing 5.19.1. Not sure what was so important to mess things up with a point release.

Arg, this is going to mess up the other 4 rtw drivers I have up here. I need a raise.

Nick

petehg, 2022-08-18:
Thanks for the response. I've created a patch from @ikegami-t and applied it successfully. So everything's working again.
Thanks for a great job.

morrownr, 2022-08-18:
Pete,

I just merged the fix. Is there any way to talk you into testing?

And thanks for the note about 88x2bu as that is where I am headed next.

Nick

petehg, 2022-08-18:
Nick
I've just downloaded now so I'll let you know how it goes.

Pete

morrownr, 2022-08-18:
Thanks. I am only testing with 5.19.2 right now and am in the process of working the patches into 4 additional drivers. The 8814au driver is an old POS and it is giving me a headache right now. Appreciate you testing.

petehg, 2022-08-18:
Nick
I've remove everything 8812au from my system with k6.0-rc1 (opensuse tumbleweed). I've newly installed your updated version with dkms and I'm pleased to report it works.
Pete

morrownr, 2022-08-18:
Pete,

You are the man as the saying goes. I'll call 8812au complete. I have now posted the patch to the 88x2bu driver. It was a relatively easy, clean job. Anyway to talk you into testing it?

I'm having to take 8814au driver out to the woodshed! That old driver causes more problems that the other 4 combined.

petehg, 2022-08-18:
Yes, Nick. 88x2bu works with k6.0-rc1. A total new install with dkms.

morrownr, 2022-08-18:
Got it. Thanks.

vscsilva, 2022-09-11:
This should be closed, right?

morrownr, 2022-09-11:
It could be closed but I think we should leave it open for a while. What happens is folks upgrade to a new kernel without first upgrading this driver and they come here looking for answers. This issue should help keep the posting of new issues down.

Appreciate your help.