morrownr/8812au-20210629 #71: Cannot compile module against kernel 6.0.0

JHM-O, 2022-10-04:
Hi Nick,

Something strange happens over here. After studying issues 66 and 67 I presumed my troubles with building against kernel 5.19.0 could be over. So I downloaded the latest zip file and gave it a new try with kernel 6.0.0. (Also gave it another try with 5.19.0)

Sadly, both still wouldn't work. So, what am I to think of that?
Below, the error message, and below that my system info:

DKMS make.log for rtl8812au-5.13.6 for kernel 6.0.0 (x86_64)
Tue 04 Oct 2022 14:01:22 CEST
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.0.0/build M=/var/lib/dkms/rtl8812au/5.13.6/build  modules
make[1]: Entering directory '/usr/src/linux-headers-6.0.0'
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/monitor/rtw_radiotap.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/wds/rtw_wds.o
/var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_sta_mgt.c: In function ‘rtw_mfree_stainfo’:
/var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_sta_mgt.c:391:25: error: the comparison will always evaluate as ‘true’ for the address of ‘lock’ will never be NULL [-Werror=address]
  391 |         if (&psta->lock != NULL)
      |                         ^~
In file included from /var/lib/dkms/rtl8812au/5.13.6/build/include/drv_types.h:112,
                 from /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_sta_mgt.c:17:
/var/lib/dkms/rtl8812au/5.13.6/build/include/sta_info.h:267:17: note: ‘lock’ declared here
  267 |         _lock   lock;
      |                 ^~~~
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/mesh/rtw_mesh_pathtbl.o
cc1: all warnings being treated as errors
make[2]: *** [scripts/Makefile.build:249: /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_sta_mgt.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1852: /var/lib/dkms/rtl8812au/5.13.6/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.0.0'
make: *** [Makefile:2510: modules] Error 2

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

**sudo uname -mr; mokutil --sb-state; lsusb; /usr/sbin/rfkill list all; /usr/sbin/dkms status; /usr/sbin/iw dev; /usr/sbin/iw reg get
6.0.0 x86_64**
EFI variables are not supported on this system
Bus 005 Device 004: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 005 Device 003: ID 04a9:190a Canon, Inc. CanoScan LiDE 210
Bus 005 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 003: ID 7392:a822 Edimax Technology Co., Ltd 802.11n NIC
Bus 004 Device 002: ID 145f:0203 Trust Trust USB Camera
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 0b33:0402 Contour Design, Inc. RollerMouse Free 2
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
rtl8812au/5.13.6: added
global
country 00: DFS-UNSET
	(2402 - 2472 @ 40), (6, 20), (N/A)
	(2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

Hopefully, you can give some clarification?

Thank you very much!

Jan Outhuis

morrownr, 2022-10-04:
Good day Jan,

> Hopefully, you can give some clarification?

Hmmm... I wish I could. This is a new one. I'll try to get to it this afternoon. I'll have to reconfigure one of my systems to try to duplicate the issue. What distro and version are you using?

Nick

morrownr, 2022-10-04:
I've run into another problem. Not sure how long it will take.

morrownr, 2022-10-05:
You may want to back down a kernel version or two to get connected. This may take a while. If you have not heard from me after a week or two, give me a shout.

Nick

JHM-O, 2022-10-05:
Thanks for the note, Nick. I'll use 5.18 when needed. No need to rush
on my behalf.

Jan

JHM-O, 2022-11-07:
Hello Nick,

What do you think, do you have some spare time by now to have a look at
this lingering issue? Would be great.

Greetings,
Jan

morrownr, 2022-11-07:
Hello @JHM-O 

Thanks for checking in and reminding me. This bug is on my list and I think I will elevate it to priority one for today.

I may have inadvertently fixed this recently so, if you would try the following in the driver directory:

$ sudo ./remove-driver.sh
$ git pull (or if you are downloading zips, you may need to delete the directory and start over with a fresh download)
$ sudo ./install-driver.sh

Let's try that first and if it does not work, I have a test system ready to go.

Nick

JHM-O, 2022-11-07:
Well Nick, appears that you can allow yourself a day off: module builds
cleanly on the newest zip.

Thanks a lot, again!

Enjoy the day.

morrownr, 2022-11-07:
That is great JHM-O.

I had thought a recent patch would catch this issue but I missed this issue on my list so I forgot to say something. There is always work to be done on these drivers and this driver is a special PITA since it is old and not well maintained by Realtek. I'll try to keep it going as long as I can.

FYI: Did you know that there is a new usb wifi chipset that is supported with an in-kernel driver and it is as fast as this chipset in 5 GHz?

It is the mt7921au. Only a couple of adapters are available so far but Alfa should be kicking out a couple of adapters soon. Speed test results:

https://github.com/morrownr/USB-WiFi/issues/139

Of note: The mt7921au is actually a WiFi 6e AX3000 chipset but the speed test was done with WiFi 5 because that is all most of the tested adapters could do.

Site Main Menu:

https://github.com/morrownr/USB-WiFi

Regards