morrownr/8812au-20210629 #72: Fails to compile in opensuse 15.3

mjopensuse, 2022-10-31:
Make results in:
root/test/8812au-20210629/os_dep/linux/ioctl_cfg80211.c:10519:3: error: ‘struct cfg80211_ops’ has no member named ‘mgmt_frame_register’
  .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
   ^~~~~~~~~~~~~~~~~~~
/root/test/8812au-20210629/os_dep/linux/ioctl_cfg80211.c:10519:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/test/8812au-20210629/os_dep/linux/ioctl_cfg80211.c:10519:25: note: (near initialization for ‘rtw_cfg80211_ops.mgmt_tx_cancel_wait’)
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-5.3.18-150300.59.98/scripts/Makefile.build:288: /root/test/8812au-20210629/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** [/usr/src/linux-5.3.18-150300.59.98/Makefile:1682: _module_/root/test/8812au-20210629] Error 2
make[1]: *** [../../../linux-5.3.18-150300.59.98/Makefile:179: sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-5.3.18-150300.59.98-obj/x86_64/preempt'
make: *** [Makefile:2513: modules] Error 2
linux-jai1:~/test/8812au-20210629 #

morrownr, 2022-10-31:
Hi @mjopensuse 

There is a pinned message at the top of Issues called Problem Reporting Guidelines. Please read it and provide the requested information.

mjopensuse, 2022-10-31:
Note: I found the following patch for the same error, different project. Tried it, but it did not work for me
https://github.com/aircrack-ng/rtl8812au/compare/v5.6.4.2...CGarces:centos

Then hacked my way through it by changing this fix as follows:
linux-jai1:~/test/rtl8812au # diff ioctl_cfg80211.c /root/test/rtl8812au/os_dep/linux/ioctl_cfg80211.c
7756c7756
< #if (LINUX_VERSION_CODE >= KERNEL_VERSION(5,8,0)) || RHEL_RELEASE_CODE >= RHEL_RELEASE_VERSION(8,0)
---
> #if (LINUX_VERSION_CODE >= KERNEL_VERSION(5,8,0)) || 1
10210c10210
< #if (LINUX_VERSION_CODE >= KERNEL_VERSION(5,8,0)) || RHEL_RELEASE_CODE >= RHEL_RELEASE_VERSION(8,0)
---
> #if (LINUX_VERSION_CODE >= KERNEL_VERSION(5,8,0)) || 1

Hope this helps to pinpoint a proper fix

mjopensuse, 2022-10-31:
Note: I found a patch for the same error, different project. Tried it, but it did not work for me (link below). Then hacked my way through it by changing this fix as follows:
linux-jai1:~/test/rtl8812au # diff ioctl_cfg80211.c /root/test/rtl8812au/os_dep/linux/ioctl_cfg80211.c
7756c7756
< #if (LINUX_VERSION_CODE >= KERNEL_VERSION(5,8,0)) || RHEL_RELEASE_CODE >= RHEL_RELEASE_VERSION(8,0)
---
> #if (LINUX_VERSION_CODE >= KERNEL_VERSION(5,8,0)) || 1
10210c10210
< #if (LINUX_VERSION_CODE >= KERNEL_VERSION(5,8,0)) || RHEL_RELEASE_CODE >= RHEL_RELEASE_VERSION(8,0)
---
> #if (LINUX_VERSION_CODE >= KERNEL_VERSION(5,8,0)) || 1

Hope this helps to pinpoint a proper fix

https://github.com/CGarces/rtl8812au/commit/e44f852d40be662251ae13415f2c1e19488e1f4a

morrownr, 2022-10-31:
Hi again @mjopensuse 

I think I am getting the idea of what the situation is:

Kernel = 3.18-150300.59.98-preempt
Arch = x86_64
Leaf 15.3 appears to be based on Centos or RHEL (this is a problem)

I took a quick look at the patch you mentioned above. That appears to be a little fix to a much larger patch as I know for certain patching code to handle Centos or RHEL takes a lot of changes. The [88x2bu](https://github.com/morrownr/88x2bu-20210702) driver does support RHEL 8.4 and 8.6. The way I handle RHEL support is I allow those who have the time to submit patches that I then test to make sure things still work for those not using RHEL based distros. I simply do not have the time to deal with the mess that Red Hat makes but I will support those that do.

Nick

mjopensuse, 2022-10-31:
Ok, got it. anyways, thanks for the quick response!

closed 2022-10-31 by mjopensuse