morrownr/8812au-20210629 #73: Cannot switch channel nor freq with 8812au in concurrent mode. i use Tenda U12

Stroichik, 2022-11-04:
Maybe i`m doing something wrong, but only way to switch channel on Kali Linux for me is to run hostapd with wlan0 and required channel. This is the only way for me to make interfaces switch channels.

I tried
sudo ip link set <interface> down
sudo iw dev <interface> set channel <#>
sudo op link set <interface> up

But it doesn't work somehow. Is there something special i need to do, when using concurrent mode?

morrownr, 2022-11-07:
Hi @Stroichik 

I have a separate repo set aside for monitor mode issues:

https://github.com/morrownr/USB-WiFi

Select Main Menu item 10. There is a document with some information and a script that helps with entering monitor mode. Study this and try the script. See if this helps. If it does not help, post an issue in the Monitor Mode repo and I will check to see if i can determine the problem.

Regards

Stroichik, 2022-11-07:
Ok, i cant switch channel with script from link too. I`ll post issue in Mon Mode repo

closed 2022-11-07 by Stroichik