morrownr/8812au-20210629 #74: D-Link DWA-182 H/W D1

robinwkurtz, 2022-11-09:
Hello!

Thank you for your util here. I have it used for an older wifi card (hardware version C), but a newer one does not work.

```
Bus 001 Device 016: ID 2001:331c D-Link Corp. 802.11ac NIC # NOT WORKING H/W Ver.: C1
Bus 001 Device 023: ID 2001:3315 D-Link Corp. D-Link Wireless Adapter # WORKING H/W Ver.: D1
```

https://ca.dlink.com/en/products/dwa-182-ac1200-mu-mimo-wi-fi-usb-adapter

Thank you so much!

robinwkurtz, 2022-11-09:
Sorry, it is supported here: https://github.com/morrownr/88x2bu-20210702

My bad!

Thanks again

closed 2022-11-09 by robinwkurtz

morrownr, 2022-11-09:
Glad you found the right driver. The Main Menu is:

https://github.com/morrownr/USB-WiFi

Let me know if you have any problems.

Nick