morrownr/8812au-20210629 #75: wifi disconnects every few minute

biswajit-rajaguru-homepc, 2022-11-22:
hi,
i am on a fedora 37 workstation. running 6.0.8-300 kernel. my machine is core i5 10400+8gb ram. 
my wifi device is wusb6300.

on my machine, the wifi randomly disconnects and reconnects every few minutes. can you please help or if the issue is with my setup, can you pleas give some pointers about what can i do.

morrownr, 2022-11-23:
Please run the following command and paste all output in a reply:

sudo uname -mr; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev; iw reg get

biswajit-rajaguru-homepc, 2022-11-27:
hi,
this is the output on my machine :

6.0.9-300.fc37.x86_64 x86_64
SecureBoot disabled
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 10c4:8108 Silicon Labs USB OPTICAL MOUSE
Bus 001 Device 007: ID c0f4:01b0 USB usb keyboard
Bus 001 Device 006: ID 0d8c:000e C-Media Electronics, Inc. Audio Adapter (Planet UP-100, Genius G-Talk)
Bus 001 Device 005: ID 05e3:0606 Genesys Logic, Inc. USB 2.0 Hub / D-Link DUB-H4 USB 2.0 Hub
Bus 001 Device 004: ID 0424:2504 Microchip Technology, Inc. (formerly SMSC) Hub
Bus 001 Device 003: ID 046d:c016 Logitech, Inc. Optical Wheel Mouse
Bus 001 Device 002: ID 13b1:003f Linksys WUSB6300 802.11a/b/g/n/ac Wireless Adapter [Realtek RTL8812AU]
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
rtl8812au/5.13.6, 6.0.5-200.fc36.x86_64, x86_64: installed
rtl8812au/5.13.6, 6.0.8-200.fc36.x86_64, x86_64: installed
rtl8812au/5.13.6, 6.0.8-300.fc37.x86_64, x86_64: installed
rtl8812au/5.13.6, 6.0.9-300.fc37.x86_64, x86_64: installed
phy#0
	Interface wlp0s20f0u3
		ifindex 3
		wdev 0x1
		addr c8:d7:19:bd:eb:ba
		ssid Gita
		type managed
		channel 157 (5785 MHz), width: 80 MHz, center1: 5775 MHz
		txpower 16.00 dBm
global
country IN: DFS-UNSET
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 30), (N/A)
	(5250 - 5350 @ 80), (N/A, 24), (N/A)
	(5470 - 5725 @ 160), (N/A, 24), (N/A)
	(5725 - 5875 @ 80), (N/A, 30), (N/A)

thank you,
biswajit

morrownr, 2022-11-28:
Everything seems to be in order based on what you posted. I think we may need to look into other possible causes.

I'd like to see what you signal level and link quality are. Would you install and run `wavemon`?

Does the router/AP have 2 net works names? One for 2.4 Ghz and one for 5 GHz? If so, do they have different names?

I have a guide regarding router/AP settings in the README. You might want to take a look?

castillofrancodamian, 2022-12-22:
I can confirm that the same thing has happened to me along with [this](https://github.com/morrownr/8812au-20210629/issues/55) error as well. I don't have it configured to reconnect.

morrownr, 2022-12-31:
@castillofrancodamian @biswajit-rajaguru-homepc 

I merged a really big update to this driver today. Give it a try and let me know.

castillofrancodamian, 2023-01-01:
@morrownr I haven't seen any improvement. The same "symptoms".

morrownr, 2023-01-02:
@castillofrancodamian 

I did some testing on the driver today and it was rock solid. I don't think this is a driver per se. The README has a section of Router/AP settings and on USB issues. Maybe you should take a look. If I had to guess, it might be an issue with heavy congestion or a router setting.

alvergallano, 2024-02-03:
hi i have TP-Link Archer T4U v2 Realtek RTL8812AU
have same problem with this 
i try multiple distro and [Aircrack version](https://github.com/aircrack-ng/rtl8812au) still  same problem with these latency spikes that @castillofrancodamian,.
but when i try this [OpenHD version](https://github.com/OpenHD/rtl8812au) the latency spikes are gone

morrownr, 2024-02-04:
Hi @alvergallano 

> but when i try this OpenHD version the latency spikes are gone

I took a look at the OpenHD driver. Whew!. That is a very old version of the driver. This information could possibly help but the differences in the code between the drivers is very large so we would need to have an idea of what is the cause first.

We are currently working this issue in the private repo for an updated version of this driver. I have not been able to duplicate the issue and I am trying very hard to do so. Would you be interested in helping with the new version? If so, I can add you to the repo.

@morrownr

alvergallano, 2024-02-04:
> hat is a very old version of the driver. This information could possibly help but

of course I'm happy to help

morrownr, 2024-02-04:
@alvergallano 

I have added you to the repo. You will get an email with instructions. Please read Issue 1 first and then post a message in Issue 2 letting us know you are there.

There is an issue already posted about you issue, Please look at it. We need to gather information about systems that are showing this issue so we can determine what might be causing the problem. I have worm my fingers out trying to duplicate the problem and have nothing at this point.

@morrownr