morrownr/8812au-20210629 #78: unable to compile 6.1.1 kernel

dardares, 2022-12-28:
Hi,
Since the release of 6.1.x kernel, i was unable to install the driver. I've downloaded the latest version as of today (28 Dec 2022)
There is the error i got:
Running install-driver.sh version 20221126
Kernel=6.1.1
Architecture=x86_64
Installing 8812au.conf to: /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to: /usr/src/rtl8812au-5.13.6
Creating symlink /var/lib/dkms/rtl8812au/5.13.6/source -> /usr/src/rtl8812au-5.13.6
The driver was added successfully.
Sign command: /lib/modules/6.1.1/build/scripts/sign-file
Binary /lib/modules/6.1.1/build/scripts/sign-file not found, modules won't be signed

Building module:
Cleaning build area...
'make' -j6 KVER=6.1.1 KSRC=/lib/modules/6.1.1/build........(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.1 (x86_64)
Consult /var/lib/dkms/rtl8812au/5.13.6/build/make.log for more information.
An error occurred. dkms build error = 10
Please report this error.
Please copy all screen output and paste it into the report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh

If i try to compile it on the Kali 6.0 kernel (not 6.10 or 6.11) this working fine without error. As soon i boot to a newer kernel, i was unable to get my adapter working. 

Thank!

morrownr, 2022-12-28:
Hi @dardares 

I just tested with kernel 6.1.1, gcc 12.1 on my Xubuntu 22.10 box. Clean compile. 

This is a problem somewhere in the dev environment. As silly as it may sound, try this:

$ sudo update
$ sudo upgrade
$ sudo reboot
$ sudo ./remove-driver.sh
$ sudo ./install-driver.sh
$ sudo reboot

dardares, 2022-12-28:
Hi,
Thank for the answer. I'm using kali linux 2022.3.
I've just tryied the command you ask me to try. First thing, the sudo apt update command and sudo upgrade command didn't work in kali, so i used: sudo apt-get update and sudo apt-get upgrade
Then i uninstalled the driver whit sudo ./remove-driver.sh
The uninstall script recommend to reboot after uninstall, so i rebooted my computer, then i do sudo ./install-driver.sh. Got the exact same error.
If i boot on the Kali 6.0 Kernel, and do the exact same step, didn't get any error.
Thank

morrownr, 2022-12-28:
Well, how did you install 6.1.1?

I've been working on some improvements to the scripts that will help catch more issues. I haven't moved those changes into this driver yet so let me take a look at doing that.

dardares, 2022-12-28:
Hi,
I've installed the kernel using those step:
1. Cloned the kernel into a empty directory
2. Copied the config files for my current kernel to the new kernel config (sudo cp -v /boot/config-$(uname -r) .config
3. Use of the make -j 6 deb-pkg
4. Installed the generated .deb for linux-images, linux-headers, linux-libc
5. Rebooted onto the new kernel then tryied installation

Thank

morrownr, 2022-12-28:
I basically did the same thing with minor variation...

make -j4
sudo make install

Did you follow installation step 3 for Debian, Kali?

sudo apt install -y linux-headers-$(uname -r) build-essential bc dkms git libelf-dev

Change apt to apt-get if needed.

dardares, 2022-12-29:
Hi,

I've tryied to uninstall all of the manually compiled kernel, then reinstall it whit your instruction whit make and make install. The make command work well, but the make install command fail. I will need to troubleshoot furthermore why my kernel is not compiling. 

Thank for your help

morrownr, 2022-12-31:
Hi @dardares 

If the make command is completing, the driver is compiling the driver. It seems the problem is with installation.

Do you already have a driver of the same name installed? How about we try:

$ lsmod

post the results

$ dkms status

post the results

$ make clean
$ make
$ sudo make install

post the results of sudo make install

Nick

dardares, 2023-01-03:
I'm closing the issue. Since i was unable to compile the newer kernel after uninstall it for testing purposes, and since you confirmed me you was able to install it on newer kernel whit another distros, i can now say the issue is not whit the driver. It located somewhere else in my kernel configuration and i will need to troubleshoot it furthermore at my side. Thank for your time.

closed 2023-01-03 by dardares