morrownr/8812au-20210629 #79: EAPOL time out with "last" commit (b106da7_?

jmrcpn, 2023-01-10:
With hostapd (2.10)
After a proper authentication, once disassociated, the link CAN NOT be restablished.
You have sequence  associated/disassociated triggered by EAPOL timeout message (in -d mode).
The only way to have a succesful link is to restart hostapd

I reinstalled driver with downgrading to commit: b85d5dc036f2fc233fd60af0d2558fea3d9b8737
support kernel 6.1
and reinstall this "old" driver.

Then everything equal otherwise, the WIFI link can be close/open multiple time with no trouble anymore
Kernel used: 6.1.4
AP wifi device: ID 13b1:003f Linksys WUSB6300 802.11a/b/g/n/ac Wireless Adapter [Realtek RTL8812AU]
linbnl: 3.7

I am guessing...
commit "376dbde", major updates to various components 
is the trouble maker.

Once again, the first wifi link request is OK, after an "unlink", the link can't be established without
restarting hostapd daemon.

Hoping that help.

morrownr, 2023-01-11:
Well, rats! I'll start working on this but for the next 10 days or so I do not know how much spare I will have so if you don't hear back from by 10 days from now, gimme a shout.

Nick

morrownr, 2023-02-11:
@jmrcpn 

I merged some major changes today. Test again and let me know.