morrownr/8812au-20210629 #80: Support for TP-Link T3U Plus

fs30000, 2023-01-13:
Device id 2357-0138, https://linux-hardware.org/?id=usb:2357-0138.

I think it has a 8812au chip. I already tried https://github.com/morrownr/88x2bu-20210702 but compilation failed.

This one compiles, but no  listing in ./os_dep/linux/usb_intf.c. I added it manually, compiled, but nothing. Dmesg and /var/log/messages show nothing. ip a, nothing new, neither lspci.

morrownr, 2023-01-13:
Hi @fs30000 

The evidence I am finding indicates a rtl8812cu chipset so the 88x2bu is the right driver. You can remove 8812au as such from the driver directory:

sudo ./remove-driver.sh

and then delete the directory.

Try 88x2bu again and report the error in that repo. I checked and the device ID, `ID 2357:0138` is already in that driver.

I've been doing some work lately on the 88x2bu driver and maybe I messed something up.

Nick

morrownr, 2023-01-13:
Correction: The evidence I am finding indicates a rtl8812bu chipset...

Typing without enough coffee.

fs30000, 2023-01-14:
Ok, i found the device on the 88x2bu repo, but failed:
Does ac 80MHz work with this driver?

```
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.4.228-1.el8.elrepo.x86_64/build M=/tmp/88x2bu-20210702  modules
make[1]: Entering directory '/usr/src/kernels/5.4.228-1.el8.elrepo.x86_64'
  CC [M]  /tmp/88x2bu-20210702/core/rtw_cmd.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_security.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_debug.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_io.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_ioctl_query.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_ioctl_set.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_ieee80211.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_mlme.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_mlme_ext.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_mi.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_wlan_util.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_vht.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_pwrctrl.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_rf.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_chplan.o
  CC [M]  /tmp/88x2bu-20210702/core/monitor/rtw_radiotap.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_recv.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_sta_mgt.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_ap.o
  CC [M]  /tmp/88x2bu-20210702/core/wds/rtw_wds.o
  CC [M]  /tmp/88x2bu-20210702/core/mesh/rtw_mesh.o
  CC [M]  /tmp/88x2bu-20210702/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /tmp/88x2bu-20210702/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_xmit.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_p2p.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_rson.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_tdls.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_br_ext.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_iol.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_sreset.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_btcoex_wifionly.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_btcoex.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_beamforming.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_odm.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_rm.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_rm_fsm.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_ft.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_wnm.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_mbo.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_rm_util.o
  CC [M]  /tmp/88x2bu-20210702/core/efuse/rtw_efuse.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_roch.o
  CC [M]  /tmp/88x2bu-20210702/core/crypto/aes-internal.o
  CC [M]  /tmp/88x2bu-20210702/core/crypto/aes-internal-enc.o
  CC [M]  /tmp/88x2bu-20210702/core/crypto/aes-gcm.o
  CC [M]  /tmp/88x2bu-20210702/core/crypto/aes-ccm.o
  CC [M]  /tmp/88x2bu-20210702/core/crypto/aes-omac1.o
  CC [M]  /tmp/88x2bu-20210702/core/crypto/ccmp.o
  CC [M]  /tmp/88x2bu-20210702/core/crypto/gcmp.o
  CC [M]  /tmp/88x2bu-20210702/core/crypto/aes-siv.o
  CC [M]  /tmp/88x2bu-20210702/core/crypto/aes-ctr.o
  CC [M]  /tmp/88x2bu-20210702/core/crypto/sha256-internal.o
  CC [M]  /tmp/88x2bu-20210702/core/crypto/sha256.o
  CC [M]  /tmp/88x2bu-20210702/core/crypto/sha256-prf.o
  CC [M]  /tmp/88x2bu-20210702/core/crypto/rtw_crypto_wrap.o
  CC [M]  /tmp/88x2bu-20210702/core/rtw_swcrypto.o
  CC [M]  /tmp/88x2bu-20210702/os_dep/osdep_service.o
  CC [M]  /tmp/88x2bu-20210702/os_dep/linux/os_intfs.o
  CC [M]  /tmp/88x2bu-20210702/os_dep/linux/usb_intf.o
  CC [M]  /tmp/88x2bu-20210702/os_dep/linux/usb_ops_linux.o
  CC [M]  /tmp/88x2bu-20210702/os_dep/linux/ioctl_linux.o
  CC [M]  /tmp/88x2bu-20210702/os_dep/linux/xmit_linux.o
  CC [M]  /tmp/88x2bu-20210702/os_dep/linux/mlme_linux.o
  CC [M]  /tmp/88x2bu-20210702/os_dep/linux/recv_linux.o
[... 10 lines cut ...]
 void cfg80211_ch_switch_started_notify(struct net_device *dev,
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/tmp/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c: At top level:
/tmp/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c:8398:9: warning: ‘struct mgmt_frame_regs’ declared inside parameter list will not be visible out                                                                side of this definition or declaration
  struct mgmt_frame_regs *upd)
         ^~~~~~~~~~~~~~~
/tmp/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_update_mgmt_frame_register’:
/tmp/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c:8426:8: error: dereferencing pointer to incomplete type ‘struct mgmt_frame_regs’
    (upd->interface_stypes & rtw_stypes_mask)) {
        ^~
/tmp/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c: At top level:
/tmp/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c:10521:3: error: ‘struct cfg80211_ops’ has no member named ‘update_mgmt_frame_registrations’; did                                                                 you mean ‘mgmt_frame_register’?
  .update_mgmt_frame_registrations = cfg80211_rtw_update_mgmt_frame_register,
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   mgmt_frame_register
/tmp/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c:10521:37: error: initialization of ‘int (*)(struct wiphy *, struct wireless_dev *, u64)’ {aka ‘i                                                                nt (*)(struct wiphy *, struct wireless_dev *, long long unsigned int)’} from incompatible pointer type ‘void (*)(struct wiphy *, struct wireless_de                                                                v *, struct mgmt_frame_regs *)’ [-Werror=incompatible-pointer-types]
  .update_mgmt_frame_registrations = cfg80211_rtw_update_mgmt_frame_register,
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/tmp/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c:10521:37: note: (near initialization for ‘rtw_cfg80211_ops.mgmt_tx_cancel_wait’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:262: /tmp/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:1739: /tmp/88x2bu-20210702] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.4.228-1.el8.elrepo.x86_64'
make: *** [Makefile:2506: modules] Error 2
```

morrownr, 2023-01-14:
> Does ac 80MHz work with this driver?

Yes

> Rocky 8.7 kernel from elrepo 5.4.228-1.el8.elrepo.x86_64.

This is where we need to concentrate on. This is not a documented distro and I know nothing about it. The problem is in your dev environment. What distro is Rocky based on?

fs30000, 2023-01-14:
Rocky is a clone of Red Hat Enterprise, the new Centos. Same team actually.

fs30000, 2023-01-14:
> > Does ac 80MHz work with this driver?
> 
> Yes
> 

AP mode? (sorry forgot to say it is AP mode)

morrownr, 2023-01-14:
> AP mode? (sorry forgot to say it is AP mode)

This driver works well in AP mode, 80 Mhz channel width and even supports DFS channels.

> Rocky is a clone of Red Hat Enterprise, the new Centos. Same team actually.

Ah, this is a show stopper. The best advice I can give is to find another non-RHEL based distro if you want to run your adapter. Red Hat does things with their kernels that make it take more time than I have available to support it.

That is the best advice I can give you. Unless you have a specific reason to run a distro based on RHEL, run away from it as fast you can.

Nick

fs30000, 2023-01-14:
> > AP mode? (sorry forgot to say it is AP mode)
> 
> This driver works well in AP mode, 80 Mhz channel width and even supports DFS channels.

Nice! Thanks for the work!

> Ah, this is a show stopper. The best advice I can give is to find another non-RHEL based distro if you want to run your adapter. Red Hat does things with their kernels that make it take more time [...]

lol, that's the second time i read that today about Red Hat. The first was about wireguard. Wireguard also stopped supporting RHEL 8.

And yeah, that's why i'm running a kernel from elrepo. elrepo has more recent kernels than Red Hat's.

Red Hat 8 current kernel: 4.18.0-425.10.1.el8_7
kernel on the error above:  5.4.228-1.el8.elrepo.x86_64

There's a newer kernel 6.x for rhel8, but it doesn't boot properly here. So i think i'll upgrade to rhel9 and try there for better luck.

morrownr, 2023-01-15:
Hmmm... let me put this another way: Red Hat, with RHEL, is their own little world. If it is RHEL kernel, it is highly unlikely it will work with this driver or with any out-of-kernel drivers for anything unless a Red Hat dev has done some mods on the driver.

If you want to be close to Red Hat maintain compatibility, try fedora.

There is another possible solution but you would have to do the research: use an adapter that works with in-kernel drivers. My site Main Menu is:

https://github.com/morrownr/USB-WiFi

Read the first 2 menu items. The 2nd menu item contains list of adapters that work with in-kernel drivers. You would have to do the research as to what RHEL supports in that regard.

Nick

fs30000, 2023-01-16:
Ok, so here are the results for Rocky/RHEL 9:

standard kernel 5.14.0-162.6.1.el9_1.0.1.x86_64: does not compile. Error at the end, error: implicit declaration of function 'PDE_DATA'; did you mean 'NODE_DATA'?

Elrepo kernel: kernel-ml-core-6.1.6-1.el9.elrepo.x86_64 it compiles. But i am finding difficulties configuring hostapd. 

With ieee80211n=1, it does  a fantastic 11MB/s  on iperf.
With ieee80211ac=1, only does 3.75MB/s.

morrownr, can you please share a working config for higher speeds?

morrownr, 2023-01-17:
Good day @fs30000 

Go to:

https://github.com/morrownr/USB-WiFi

Read menu item 9. It contains a lot of information about using hostapd and it has sample hostapd.conf files.

fs30000, 2023-01-17:
> Good day @fs30000
> 
> Go to:
> 
> https://github.com/morrownr/USB-WiFi
> 
> Read menu item 9. It contains a lot of information about using hostapd and it has sample hostapd.conf files.

wow, thanks for the configs and write up! Finally, decent speeds! I can get 35 sometimes even 40MB/s from my phone. Sometimes just 20MB/s. Anyway, i'm happy with this. Maybe later a good 11ax adapter will come to linux. Intel modules suck because they don't support 5Ghz in AP mode.

I'll post my config  here so others on RHEL9 can use it.

###########
RHEL 9.1

##################
packages from elrepo:
kernel-ml-core-6.1.6-1.el9.elrepo.x86_64
kernel-ml-modules-6.1.6-1.el9.elrepo.x86_64
kernel-ml-6.1.6-1.el9.elrepo.x86_64

###################
Just make and make install on this repo's dir.

###########################
/etc/modprobe.d/88x2bu.conf
options 88x2bu rtw_vht_enable=2 rtw_switch_usb_mode=1

#####################
hostapd.conf:
logger_syslog=-1
logger_syslog_level=2

bridge=br0
interface=wlp0s21f0u2
driver=nl80211

macaddr_acl=0

ssid=myap
wpa_passphrase=mypass
hw_mode=a
channel=36
vht_oper_chwidth=1

vht_oper_centr_freq_seg0_idx=42
country_code=PT
ieee80211d=1
ieee80211h=1

macaddr_acl=0

auth_algs=1
ignore_broadcast_ssid=0
wpa=2
rsn_pairwise=CCMP
wpa_key_mgmt=WPA-PSK

ieee80211n=1
wmm_enabled=1
ht_capab=[LDPC][HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935][DSSS_CCK-40]

ieee80211ac=1
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7][RXLDPC][HTC-VHT]

#end

morrownr, 2023-01-18:
Hi @fs30000 

You are getting good at this. There is no need to buy a wifi router/AP when you can make your own.

Something you might consider if traffic is congested on channel 36 is using a DFS channel if DFS channels are available in your country and if the devices that you will be using support connecting to DFS channels. To activate, you would need to determine what channel to try and add a parameter to the options line:

```
# DFS Options ( rtw_dfs_region_domain )
#
# 0 = NONE (default)
# 1 = FCC
# 2 = MKK
# 3 = ETSI
#
# Notes:
# - Activates DFS channels in AP mode.
# - DFS FCC 80 MHz channels for hostapd: 52(58), 100(106), 116(122) and 132(138)
# - For more information: https://en.wikipedia.org/wiki/List_of_WLAN_channels
```

I can put you on a link to the regdb if you need to look up and info.

Nick

morrownr, 2023-01-18:
> Can this USB dongle use these higher channels (100-140)?

With the driver in this repo, yes. My 8812au adapter is in AP mode on a RasPi4B on channel 100. Nothing else is on this channel here so I can run full speed.

> Which m.2 or mPCIE wifi module do you recommend for 11ax other than intel (no support for 5Ghz AP mode)?

Yeah, Intel has kinda messed up on 5 GHz AP mode. I am going to recommend some things here based on conversations and not actual use so keep that in mind.

I think you would find cards based on the mt7921k or mt7922 chipsets would serve you well. The mt7921k is tri-band and supports up to 80 MHz channel width. The mt7922 is tri-band and supports up to 160 MHz channel width. Both work with the in-kernel mt7921u driver. You need at least a 5.19 kernel to get the driver with AP mode.

This might help. The Main Menu for this site is (I think I posted this already):

https://github.com/morrownr/USB-WiFi

We use `Issues` to have discussions, ask questions and post info. You might post your question there and see what answers you get. You can also read menu items like 2 but it is only about usb adapters, however, there is info that might come in handy.

Nick

fs30000, 2023-01-18:
> Yeah, Intel has kinda messed up on 5 GHz AP mode. I am going to recommend some things here based on conversations and not actual use so keep that in mind.
>
> I think you would find cards based on the [...]

Will get one of of those then.

> This might help. The Main Menu for this site is (I think I posted this already):
>
> https://github.com/morrownr/USB-WiFi
>
> We use `Issues` to have discussions, ask questions and post info. You might [...]

Will ask a few more pointers there then, many thanks so far!

fs30000, 2023-02-02:
So now, on windows the link speed for receive/transmit is 650/86 (Mbps), that means iperf3 only does 6MB/s! Same thing on my phone! I'm puzzled because  in the beginning iperf did 30MB/s on windows and 20 MB/s on the phone. Am i missing something?

If i do iperf -R, it does 30MB/s as usual tough.

closed 2023-03-17 by morrownr

fs30000, 2023-03-18:
Nick, please allow me to share my config for AP so someone can benefict. I quit using this adapter. But this was working good lately.

On Makefile before compiling:
````CONFIG_POWER_SAVING = n````

/etc/modprobe.d/88xx2bu.conf for module options and blocking the kernel's own module, here it sucked:

````
blacklist rtw88_8822bu
blacklist rtw88_usb
blacklist rtw88_8822b
blacklist rtw88_core

options 88x2bu rtw_country_code=( 2 letter country code here)   rtw_btcoex_enable=0   rtw_drv_log_level=2   rtw_led_ctrl=0   rtw_vht_enable=2   rtw_switch_usb_mode=1   rtw_dfs_region_domain=1   rtw_beamform_cap=1   rtw_power_mgnt=0 rtw_enusbss=0 rtw_ips_mode=0
````

hostapd.conf

````

interface=wlp0s20f0u3 (this might vary, check ip a for interface's name) 
bridge=br0
driver=nl80211

ssid=ap1
wpa_passphrase=12345678
country_code=(two letter country code)

auth_algs=3
macaddr_acl=0
ignore_broadcast_ssid=0
wpa=2
rsn_pairwise=CCMP
wpa_key_mgmt=WPA-PSK

ieee80211d=1
ieee80211h=1

hw_mode=a
channel=36

ieee80211n=1
wmm_enabled=1
ht_capab=[LDPC][HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935][DSSS_CCK-40]

ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7][RXLDPC][MAX-AMSDU-7935][SU-BEAMFORMER]
````

Case closed :)

morrownr, 2023-03-20:
Got it. Thanks.