morrownr/8812au-20210629 #82: Running install-driver.sh gives me error.

ducksteam33, 2023-01-29:
Running install-driver.sh gives me error.

Hello. I'im having issues installing the driver.
Running install-driver.sh gives me following output:

> Script: install-driver.sh v20230109
> Arch:x86_64
> MemTotal:       16184508 kB
> Kernel:  6.1.8-arch1-1
> gcc: gcc (GCC) 12.2.1 20230111
> Installing 8812au.conf to /etc/modprobe.d
> The non-dkms installation routines are in use.
> /bin/sh: line 1: bc: command not found
> /bin/sh: line 1: echo: write error: Broken pipe
> make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.1.8-arch1-1/build M=/home/ducksteam/Documents/Tools/8812au-20210629  modules
> make[1]: Entering directory '/usr/lib/modules/6.1.8-arch1-1/build'
> make[1]: *** No rule to make target 'modules'.  Stop.
> make[1]: Leaving directory '/usr/lib/modules/6.1.8-arch1-1/build'
> make: *** [Makefile:2504: modules] Error 2
> An error occurred:  2

 If someone could answer me what the problem could be, because i really don't understand whats going on. 
Thanks!

morrownr, 2023-01-29:
Run `sudo ./remove-driver.sh` to clean up then... install the following package:

`bc`

It looks like you may be running Arch Linux and I don't know how to install packages on Arch so you need to do that.

Then run `sudo ./install-driver.sh` again.

In installation Step 3, there are instuctions for setting up your dev environment for Arch. Did you follow them?

ducksteam33, 2023-01-30:
uffff. Yeah i ovelooked step 3. Thanks for the help! Sorry for bothering you guys!

closed 2023-01-30 by ducksteam33

morrownr, 2023-01-30:
> uffff. Yeah i ovelooked step 3. Thanks for the help! Sorry for bothering you guys!

No problem. Glad to are going.