morrownr/8812au-20210629 #84: Latest Linux-LTS Kernel Breaks Compatibility?

auditor3d, 2023-02-20:
Updating my ArchLinux to the latest Linux-LTS kernel 6.1.12-1 and this package would no longer compile. Tried to manually recompile and getting exit errors.

Had to downgrade to 5.15.94-1 LTS in order to get Wifi Back.  Can anyone else confirm this?

morrownr, 2023-02-21:
Clean here...

```
$ sudo ./install-driver.sh
: ---------------------------
: install-driver.sh v20230126
: x86_64 (architecture)
: 4/4 (in-use/total processing units)
: 12216352 (total system memory)
: 6.1.12-060112-generic (kernel version)
: gcc (Ubuntu 12.1.0-2ubuntu1~22.04) 12.1.0
: dkms-2.8.7
: SecureBoot disabled
: ---------------------------
Installing 8812au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8812au-5.13.6
Creating symlink /var/lib/dkms/rtl8812au/5.13.6/source -> /usr/src/rtl8812au-5.13.6
The driver was added to dkms successfully.

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
./dkms-make.sh.................
Signing module:
 - /var/lib/dkms/rtl8812au/5.13.6/6.1.12-060112-generic/x86_64/module/8812au.ko
Secure Boot not enabled on this system.
cleaning build area...
Compile time: 149.03 seconds
The driver was built by dkms successfully.

8812au.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.1.12-060112-generic/updates/dkms/

depmod...
The driver was installed by dkms successfully.
Do you want to edit the driver options file now? [y/N] y
Do you want to apply the new options by rebooting now? (recommended) [y/N]
```

Did you upgrade this driver on your system before upgrading? The way to upgrade is:

$ git pull
$ sudo ,.install-driver.sh

This driver gets regular upgrades and I can't reach out and put the updated code on your system. The regular updates have to do with Linux being updated almost constantly.

If upgrading and trying again doesn't help, I need to start posting the problems you are seeing.

@morrownr

auditor3d, 2023-02-21:
Upgrading in this way did fix it.  thank you.

closed 2023-02-21 by auditor3d

morrownr, 2023-02-21:
You are welcome.