morrownr/8812au-20210629 #87: Losing internet but connection stays

xenicx, 2023-03-16:
I managed to manually install the driver on Pop!_OS with 6.20-76060200-generic kernel  but sometimes it loses the internet. The connection to my router stays but there isn't internet. If I want to get the internet back I have to reconnect to my router and then it works again.
Here is the output of the command:
```
6.2.0-76060200-generic x86_64
SecureBoot disabled
Platform is in Setup Mode
Bus 002 Device 003: ID 0b05:17d2 ASUSTek Computer, Inc. USB-AC56 802.11a/b/g/n/ac Wireless Adapter [Realtek RTL8812AU]
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
bcmwl/6.30.223.271+bdcom, 6.0.12-76060006-generic, x86_64: installed
bcmwl/6.30.223.271+bdcom, 6.1.11-76060111-generic, x86_64: installed
bcmwl/6.30.223.271+bdcom, 6.2.0-76060200-generic, x86_64: installed
nvidia/525.89.02, 6.2.0-76060200-generic, x86_64: installed
system76/1.0.14~1643391291~22.04~78ede46, 5.18.10-76051810-generic, x86_64: installed
system76/1.0.14~1643391291~22.04~78ede46, 6.0.12-76060006-generic, x86_64: installed
system76/1.0.14~1643391291~22.04~78ede46, 6.1.11-76060111-generic, x86_64: installed
system76/1.0.14~1643391291~22.04~78ede46, 6.2.0-76060200-generic, x86_64: installed
system76_acpi/1.0.2~1678132042~22.04~ed01124, 6.1.11-76060111-generic, x86_64: installed (original_module exists)
system76_acpi/1.0.2~1678132042~22.04~ed01124, 6.2.0-76060200-generic, x86_64: installed (original_module exists)
system76-io/1.0.2~1655490480~22.04~0217576, 5.18.10-76051810-generic, x86_64: installed
system76-io/1.0.2~1655490480~22.04~0217576, 6.0.12-76060006-generic, x86_64: installed
system76-io/1.0.2~1655490480~22.04~0217576, 6.1.11-76060111-generic, x86_64: installed
system76-io/1.0.2~1655490480~22.04~0217576, 6.2.0-76060200-generic, x86_64: installed
phy#0
	Interface wlxfc34972ed5de
		ifindex 3
		wdev 0x1
		addr fc:34:97:2e:d5:de
		ssid dlink
		type managed
		channel 40 (5200 MHz), width: 80 MHz, center1: 5210 MHz
		txpower 15.00 dBm
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

```

xenicx, 2023-03-16:
Sorry for the poor choice of words. I meant I manually built the driver because when I tried to install the driver using the script it didn't work out.

morrownr, 2023-03-16:
It would be helpful if I had the onscreen log that install-driver.sh produces.

I'm not seeing anything so far that is helpful.

xenicx, 2023-03-21:
I now tried installing it using the script and it worked. Here is the log what was on terminal for it.
[terminal_log.txt](https://github.com/morrownr/8812au-20210629/files/11029607/terminal_log.txt)