morrownr/8812au-20210629 #89: Error during manual install

AvariCe-git, 2023-03-20:
Distro : Manjaro  22.0.5 Sikaris
Kernel : 6.1.19
Architecture : x86_64

Hello, I have the following problem either when I try to install it from AUR or manually:
 
[image]

I tried to 

make clean
make
sudo make install

and the problem persists.

What can I do?

AvariCe-git, 2023-03-20:
Now it's installed, I don't understand what I did differently.  Thank you.
If you happen to know, now the receiver doesn't work. 
It is installed in DKMS, but it isn't picked up by Manjaro. lsusb is like this, 
`Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 0b05:18f3 ASUSTek Computer, Inc. AURA LED Controller
Bus 003 Device 010: ID 0bda:8813 Realtek Semiconductor Corp. RTL8814AU 802.11a/b/g/n/ac Wireless Adapter
Bus 003 Device 007: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 003 Device 006: ID 045e:0779 Microsoft Corp. LifeCam HD-3000
Bus 003 Device 004: ID 046d:c08b Logitech, Inc. G502 SE HERO Gaming Mouse
Bus 003 Device 002: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 2357:0604 TP-Link TP-Link UB500 Adapter
Bus 001 Device 003: ID 05e3:0745 Genesys Logic, Inc. Logilink CR0012
Bus 001 Device 002: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

If it's something I missed tell me.

closed 2023-03-20 by AvariCe-git

morrownr, 2023-03-20:
> Edit: I'm an idiot.

You are not alone. I am in the same club.